Emotiv/cortex-example #6: Why do I get error -32018 "Missing required parameters." when using Anonymous authorize?

Frooxius, 2019-06-02:
Hello!

I am trying to use the C# wrapper to get streams from the EPOC+ device, however I am having trouble running the authorize method.

I open the connection CortexClient.Instance.Open(), run GetUserLogin() (this returns correct user) and after that I run the Authorize method, which I've modified to pass empty params as specified in the documentation (https://emotiv.github.io/cortex-docs/#authorize).

However this results in "Error: Missing required parameters." code: -32018

Here's full sequence of the sent and received JSON messages:

`Send getUserLogin
{
  "jsonrpc": "2.0",
  "id": 1,
  "method": "getUserLogin"
}

Received: {"id":1,"jsonrpc":"2.0","result":[{"currentOSUId":"1001","currentOSUsername":"Tomas","loggedInOSUId":"1001","loggedInOSUsername":"Tomas","username":"frooxius"}]}

handleResponse: getUserLogin

GetUserLogin: frooxius

Send authorize
{
  "jsonrpc": "2.0",
  "id": 2,
  "method": "authorize",
  "params": {}
}

Received: {"error":{"code":-32018,"message":"Missing required parameters."},"id":2,"jsonrpc":"2.0"}`

Is there a step that I am missing?

bchaperon, 2019-06-03:
Hello @Frooxius 
I am sorry, the API of Cortex 2.0 is not the same with the Cortex 1.x API.
In Cortex 2.0, you cannot do an anonymous authorize anymore. You must provide a client id and client secret.
The doc you mentioned (https://emotiv.github.io/cortex-docs) is only valid for Cortex 1.x, it is not valid for Cortex 2.0.
Emotiv will publish a brand new documentation for 2.0 very soon, probably by the end of this week.

Frooxius, 2019-06-04:
Thanks @bchaperon for the quick response! Now it makes sense, I assumed that since the JSON is the same, only the C# wrapper structure was changed. I'll try with the client id and secret then, thank you!

closed 2019-06-05 by nguoithichkhampha

bchaperon, 2019-06-11:
Hi @Frooxius
FYI the new API documentation is now available at https://emotiv.gitbook.io/cortex-api/
Hope this will be helpful :)