Emotiv/cortex-example #7: License ID needed for the free version, why ? C#

khaleelasyraaf, 2019-06-03:
Hi there, 

I'm currently working on Emotiv Insight device using C# and I've installed the SDK provided as well as using Cortex V2 examples for my reference. My plan is to use the free version which I don't need to provide the license. 

So I did all the necessary steps to achieve that, I received my client id and client secret from my account then included that in the Config.cs.
However, when I ran the application from project FacialExpressionTraining (which don't require license key), I received an error stating that I need one to run this. The error is stated as below:

-32002 message Invalid license key

[image]

Is there a way to fix this ? Please help me with this. I'm relatively new to Emotiv devices.

nguoithichkhampha, 2019-06-03:
@khaleelasyraaf , which license id you set in Config.cs ?

bchaperon, 2019-06-04:
Hi @khaleelasyraaf 
In theory, when you call "authorize", you don't need to set the license id. Cortex should be able to figure it out by itself.
Can you try to call "authorize" with only 2 parameters, client id and client secret?

bchaperon, 2019-06-04:
I just had a look at the source code, and it seems that in Config.cs, you should set LicenseId to an empty string, or leave it null.

khaleelasyraaf, 2019-06-04:
Hi @bchaperon ,
I've set it into empty string and it works now. Thank you so much for your help! 

Have a nice day ahead!

closed 2019-06-04 by khaleelasyraaf