Emotiv/cortex-example #9: Why do I get MessageErrorRecieved :code -32019 message Session limit on this device has been reached. ?

LucasBuuck, 2019-06-10:
Hi everybody, 
I wanted to try the Cortex V2.0 with the examples presented on C# but it didn't work and I always get this error "MessageErrorRecieved :code -32019 message Session limit on this device has been reached." 
I have a free license and it does this error on all examples.
I didn't have any problem with the former version and according to emotiv website it is noted "Unlimited sessions on 3 devices". So how can I have a session limit ? I only use an Epoc+ headset with Bluetooth.

If you have any clue about this please tell me,
Thanks in advance

nguoithichkhampha, 2019-06-10:
@LucasBuuck it relates to your debit number in method "authorize". 
Ex: you debit 5 session, after use all you will get this error. in that case, you should call "authorize" again.

nguoithichkhampha, 2019-06-10:
you should take a look in the document for more details
https://emotiv.gitbook.io/cortex-api/authentication/authorize

LucasBuuck, 2019-06-10:
Oh indeed, i just noticed that if i don't put a licenseID the default debit will be 0, thanks you it's working now.

closed 2019-06-10 by LucasBuuck

GeorgeSbyrakis, 2019-08-03:
Hello can you please help me as well?  I am having exactly the same issue i dont know how to resolve it.... I also have the free licence.