Emotiv/cortex-example #19: MessageErrorRecieved :code -32027

AdityaSher, 2019-06-24:
message: The application do not have permission to use the license.

nguoithichkhampha, 2019-06-24:
@AdityaSher , some licenses only allow use in specific application. Ex: pro license only use in Emotiv Pro.

AdityaSher, 2019-06-24:
> @AdityaSher , some licenses only allow use in specific application. Ex: pro license only use in Emotiv Pro.

I have access to pro liscense (please check my updated comment as below)

> Note: I have already added a licence key for pro version as well as client id and secret key

nguoithichkhampha, 2019-06-24:
you can no use pro license for your app.

bchaperon, 2019-06-24:
Hello @AdityaSher 
You have some links to apply for a SDK license on this page https://www.emotiv.com/developer/
So if you already have a PRO license, you should fill this form: 
https://www.emotiv.com/non-commercial-sdk-application-form/?preview=true

AdityaSher, 2019-06-24:
How Can I find out whether I already have the api access or not, I think we have registered for the SDK as well.

AdityaSher, 2019-06-24:
> You have some links to apply for an SDK license on this page https://www.emotiv.com/developer/
> So if you already have a PRO license, you should fill this form:
> https://www.emotiv.com/non-commercial-sdk-application-form/?preview=true

Hello @bchaperon 
How Can I find out whether I already have the API access or not, I think we have registered for the SDK as well.

bchaperon, 2019-06-24:
@AdityaSher You should contact the [EMOTIV customer support.](https://www.emotiv.com/contact-support/)
Give them your Emotiv ID and they should be able to sort this out.

AdityaSher, 2019-06-24:
I have contacted Emotiv Care using the link provided by you thankyou very much @bchaperon

closed 2019-06-24 by AdityaSher