Emotiv/cortex-example #21: Request: Unity C# examples

AdityaSher, 2019-06-26:
Can we get examples for Unity similar to the ones that we had for the community SDK, just the connection part at least and generic commands like connecting to the headset on the using buttons?

hoangphamemotiv, 2019-06-28:
we had a example : https://github.com/Emotiv/community-sdk/tree/master/examples_basic/Unity/MentalCommand
i think it has connection part, right?

hoangphamemotiv, 2019-06-28:
@AdityaSher 
we haven't supported examples for Unity yet. at the moment, you can take a look on C# examples

JorgeVelez, 2019-07-05:
> Can we get examples for Unity similar to the ones that we had for the community SDK, just the connection part at least and generic commands like connecting to the headset on the using buttons?

I tested it with unity and it works fine, just copy the dlls and use the program.cs files in each demo. You just have to change the "main" method to "start".

hoangphamemotiv, 2019-07-08:
hi @JorgeVelez . 
did you talk about Cortex example or old SDK Unity example?

JorgeVelez, 2019-07-08:
Im talking about the new cortex v2 sdk. The csharp code almost works as is in Unity.

hoangphamemotiv, 2019-07-09:
wow, this is good news. i haven't tried it with Unity yet.  your information very useful. 
thank @JorgeVelez !

closed 2019-07-09 by hoangphamemotiv

hoangphamemotiv, 2020-02-06:
@JorgeVelez can you share detail the way you make the Cortex working on Unity. it is help other user working on it before we support it.