Emotiv/cortex-example #25: Session limit on this device has been reached

xaviervasques, 2019-07-25:
Hello, 
I tried to run example.py and I have the error below (Session limit on this device has been reached). I have a EPOC+ and a EmotivPro (Academic) license. Can you please help ?
Thanks

air-de-xav:python xav$ python3 example.py 
** USER LOGIN **
Sending request:
{"jsonrpc": "2.0", "method": "getUserLogin", "params": {}, "id": 1}
sent; awaiting response
lib.cortex resp:
{'currentOSUId': '501', 'currentOSUsername': 'xav', 'loggedInOSUId': '501', 'loggedInOSUsername': 'xav', 'username': 'lrenc'}
** GET CORTEX INFO **
Sending request:
{"jsonrpc": "2.0", "method": "getCortexInfo", "params": {}, "id": 2}
sent; awaiting response
lib.cortex resp:
{'id': 2, 'jsonrpc': '2.0', 'result': {'buidDate': '2019-06-24T15:05:21', 'buildNumber': '4eeba8e2', 'version': '2.0.2'}}
** HAS ACCESS RIGHT **
Sending request:
{"jsonrpc": "2.0", "method": "requestAccess", "params": {"clientId": "wW844zQFhtnzHZNdJbfWyE3x5M7qbXxrE0xL7Je7", "clientSecret": "KxcrcJa0MohokLA6pQSfHG1bob0M7FSRtQlpNzjBgCQfRSTdvikXlwLujgbGCLSEf0AdoveHwh2mObe5RytGgXwT2lElj4biL7pTTyzvTnI3u4RyFqwagbWPjGkpdSZl"}, "id": 3}
sent; awaiting response
lib.cortex resp:
{'id': 3, 'jsonrpc': '2.0', 'result': {'accessGranted': True, 'message': 'The User has access right to this application.'}}
** REQUEST ACCESS **
Sending request:
{"jsonrpc": "2.0", "method": "requestAccess", "params": {"clientId": "wW844zQFhtnzHZNdJbfWyE3x5M7qbXxrE0xL7Je7", "clientSecret": "KxcrcJa0MohokLA6pQSfHG1bob0M7FSRtQlpNzjBgCQfRSTdvikXlwLujgbGCLSEf0AdoveHwh2mObe5RytGgXwT2lElj4biL7pTTyzvTnI3u4RyFqwagbWPjGkpdSZl"}, "id": 4}
sent; awaiting response
lib.cortex resp:
{'id': 4, 'jsonrpc': '2.0', 'result': {'accessGranted': True, 'message': 'The User has access right to this application.'}}
** AUTHORIZE **
Sending request:
{"jsonrpc": "2.0", "method": "authorize", "params": {"clientId": "wW844zQFhtnzHZNdJbfWyE3x5M7qbXxrE0xL7Je7", "clientSecret": "KxcrcJa0MohokLA6pQSfHG1bob0M7FSRtQlpNzjBgCQfRSTdvikXlwLujgbGCLSEf0AdoveHwh2mObe5RytGgXwT2lElj4biL7pTTyzvTnI3u4RyFqwagbWPjGkpdSZl"}, "id": 5}
sent; awaiting response
lib.cortex resp:
{'id': 5, 'jsonrpc': '2.0', 'result': {'cortexToken': 'eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhcHBJZCI6ImNvbS5scmVuYy50ZXN0XzEiLCJhcHBWZXJzaW9uIjoiMS4wIiwiZXhwIjoxNTY0MjQwMTE5LCJsaWNlbnNlSWQiOiI5YzU4NTA4NS01ZTNhLTQyZmEtYTE2NS0xMzUxY2ExOGFhN2EiLCJuYmYiOjE1NjM5ODA5MTksInVzZXJJZCI6IjQ2Zjc2NzQwLTFlYWYtNDAzOS04Yzk1LTdhMzFkZGFkOWYwZSIsInVzZXJuYW1lIjoibHJlbmMiLCJ2ZXJzaW9uIjoiMi4wIn0=.DgDomts+xgC7Gqg/xwojSSHD7omrrX8FXE02OiGXRFI='}}
** GET LICENSE INFO **
Sending request:
{"jsonrpc": "2.0", "method": "getLicenseInfo", "params": {"cortexToken": "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhcHBJZCI6ImNvbS5scmVuYy50ZXN0XzEiLCJhcHBWZXJzaW9uIjoiMS4wIiwiZXhwIjoxNTY0MjQwMTE5LCJsaWNlbnNlSWQiOiI5YzU4NTA4NS01ZTNhLTQyZmEtYTE2NS0xMzUxY2ExOGFhN2EiLCJuYmYiOjE1NjM5ODA5MTksInVzZXJJZCI6IjQ2Zjc2NzQwLTFlYWYtNDAzOS04Yzk1LTdhMzFkZGFkOWYwZSIsInVzZXJuYW1lIjoibHJlbmMiLCJ2ZXJzaW9uIjoiMi4wIn0=.DgDomts+xgC7Gqg/xwojSSHD7omrrX8FXE02OiGXRFI="}, "id": 6}
sent; awaiting response
lib.cortex resp:
{'id': 6, 'jsonrpc': '2.0', 'result': {'isOnline': True, 'license': {'applications': ['com.emotiv.sdk.basic'], 'billingFrom': '2019-04-18T02:00:00.000+02:00', 'billingTo': '2029-04-15T02:00:00.000+02:00', 'deviceInfo': {'deviceLimit': 3, 'devicesPerSeat': 3, 'sessionLimit': {'day': None, 'month': None, 'year': None}}, 'expired': False, 'extenderLimit': 6, 'hardLimitTime': '2019-08-26T01:59:59.999+02:00', 'isCommercial': False, 'licenseId': '9c585085-5e3a-42fa-a165-1351ca18aa7a', 'licenseName': 'Basic License', 'localQuota': 0, 'maxDebit': None, 'scopes': [], 'seatCount': 1, 'sessionCount': 0, 'softLimitTime': '2019-08-19T01:59:59.999+02:00', 'totalDebit': 2, 'totalRegisteredDevices': 1, 'validFrom': '2019-04-18T02:00:00.000+02:00', 'validTo': '2029-04-16T01:59:59.999+02:00'}}}
** QUERY HEADSETS **
Sending request:
{"jsonrpc": "2.0", "method": "queryHeadsets", "params": {}, "id": 7}
sent; awaiting response
lib.cortex found headsets ['EPOCPLUS-4A2C06E7']
lib.cortex resp:
{'id': 7, 'jsonrpc': '2.0', 'result': [{'connectedBy': 'bluetooth', 'customName': '', 'dongle': '0', 'firmware': '633', 'id': 'EPOCPLUS-4A2C06E7', 'motionSensors': ['Q0', 'Q1', 'Q2', 'Q3', 'ACCX', 'ACCY', 'ACCZ', 'MAGX', 'MAGY', 'MAGZ'], 'sensors': ['AF3', 'F7', 'F3', 'FC5', 'T7', 'P7', 'O1', 'O2', 'P8', 'T8', 'FC6', 'F4', 'F8', 'AF4'], 'settings': {'eegRate': 128, 'eegRes': 16, 'memsRate': 64, 'memsRes': 16, 'mode': 'EPOCPLUS'}, 'status': 'connected'}]}
** CREATE SESSION **
Sending request:
{"jsonrpc": "2.0", "method": "createSession", "params": {"cortexToken": "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhcHBJZCI6ImNvbS5scmVuYy50ZXN0XzEiLCJhcHBWZXJzaW9uIjoiMS4wIiwiZXhwIjoxNTY0MjQwMTE5LCJsaWNlbnNlSWQiOiI5YzU4NTA4NS01ZTNhLTQyZmEtYTE2NS0xMzUxY2ExOGFhN2EiLCJuYmYiOjE1NjM5ODA5MTksInVzZXJJZCI6IjQ2Zjc2NzQwLTFlYWYtNDAzOS04Yzk1LTdhMzFkZGFkOWYwZSIsInVzZXJuYW1lIjoibHJlbmMiLCJ2ZXJzaW9uIjoiMi4wIn0=.DgDomts+xgC7Gqg/xwojSSHD7omrrX8FXE02OiGXRFI=", "headset": "EPOCPLUS-4A2C06E7", "status": "active"}, "id": 8}
sent; awaiting response
Got error in createSession with params {'cortexToken': 'eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhcHBJZCI6ImNvbS5scmVuYy50ZXN0XzEiLCJhcHBWZXJzaW9uIjoiMS4wIiwiZXhwIjoxNTY0MjQwMTE5LCJsaWNlbnNlSWQiOiI5YzU4NTA4NS01ZTNhLTQyZmEtYTE2NS0xMzUxY2ExOGFhN2EiLCJuYmYiOjE1NjM5ODA5MTksInVzZXJJZCI6IjQ2Zjc2NzQwLTFlYWYtNDAzOS04Yzk1LTdhMzFkZGFkOWYwZSIsInVzZXJuYW1lIjoibHJlbmMiLCJ2ZXJzaW9uIjoiMi4wIn0=.DgDomts+xgC7Gqg/xwojSSHD7omrrX8FXE02OiGXRFI=', 'headset': 'EPOCPLUS-4A2C06E7', 'status': 'active'}:
{"error":{"code":-32019,"message":"Session limit on this device has been reached."},"id":8,"jsonrpc":"2.0"}
Traceback (most recent call last):
  File "example.py", line 45, in <module>
    test()
  File "example.py", line 40, in test
    asyncio.run(do_stuff(cortex))
  File "/usr/local/Cellar/python/3.7.4/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "/usr/local/Cellar/python/3.7.4/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/base_events.py", line 579, in run_until_complete
    return future.result()
  File "example.py", line 24, in do_stuff
    headset_id=cortex.headsets[0])
  File "/Users/xav/Desktop/GitHub/cortex-v2-example/python/lib/cortex.py", line 271, in create_session
    resp = await self.send_command('createSession', **params)
  File "/Users/xav/Desktop/GitHub/cortex-v2-example/python/lib/cortex.py", line 154, in send_command
    raise CortexApiException(resp)
lib.cortex.CortexApiException: {"error":{"code":-32019,"message":"Session limit on this device has been reached."},"id":8,"jsonrpc":"2.0"}
air-de-xav:python xav$

nguoithichkhampha, 2019-07-26:
@xaviervasques , from the response of api "getLicenseInfo". the field "localQuota" = 0. 
So, you get the error while call "createSession".
You should call "authorize" again with "debit" param is big number enough.
https://emotiv.gitbook.io/cortex-api/authentication/authorize

xaviervasques, 2019-07-26:
Hello Nguoithichkhampha  and thank you for your replay. 
What I did to resolve the issue is indeed to modify "debit" param directly into the authorize function. Thanks

closed 2019-07-26 by xaviervasques

GeorgeSbyrakis, 2019-08-03:
Hello i am having the same trouble as well but i have the free licence! Could you please navigate me to the authorize function? Thanks

nguoithichkhampha, 2019-08-05:
@GeorgeSbyrakis take a took in the function 
async def authorize(self, license_id=None, debit=None)
in class 
https://github.com/Emotiv/cortex-v2-example/blob/master/python/lib/cortex.py

dalozanoc, 2019-08-09:
Greetings

 I would like to help me with a problem that I have for now, in python 3.7 after successfully making the streams of "com" with the following lines of code:

{
   "id": 1,
   "jsonrpc": "2.0",
   "method": "subscribe",
   "params": {
       "cortexToken": "xxx",
       "session": "f8cb7289-9a92-438b-8281-e5fdffe8166e",
       "streams": ["com"]
   }
}

I get this answer:

{
   "id": 8,
   "jsonrpc": "2.0",
   "result": {
       "failure": [],
       "success": [{
           "cols": ["act", "pow"],
           "sid": "7f899d66-442b-4bf4-9752-ed06d57b72c3",
           "streamName": "com"
       }]
   }
}

same that allows me to obtain data of some specific activity but I wanted to please help me with the line of code that allows me to obtain this result for example:

{
   "com": ["push", 0.376],
   "sid": "7f899d66-442b-4bf4-9752-ed06d57b72c3",
   "time": 1559900743.3318
}

How can I print on screen the action that is currently being done with the Emotiv ?, I thank you very much in advance.

P.S. I am based on the use of its website: https://emotiv.gitbook.io/cortex-api/data-subscription/data-sample-object
but I can't find a way to show the specific action.

Rick100k, 2019-08-09:
Greetings I have the same problem I subscribe to mot data for the gyroscope and receive the following response
[image]
On the cortex page I find the following example that gives the values ​​of the gyroscope

[
    "COUNTER_MEMS","INTERPOLATED_MEMS",
    "GYROX","GYROY","GYROZ",
    "ACCX","ACCY","ACCZ",
    "MAGX","MAGY","MAGZ"
]
{
    "mot":[
        14,0,
        8206,8187,8181,
        4235,8668,8128,
        8294,8237,7938
    ],
    "sid":"462c4d75-113f-4664-a443-3aaa02c178d0",
    "time":1559902927.7428
}

Saludos estoy utilizando python