Emotiv/cortex-example #26: You must login on CortexUI before request for grant access then rerun

xaviervasques, 2019-07-26:
Hello, 
Even if I am connected through CORTEX App, I have the following message.
How can I solve this issue ?
Thanks,
Xavier

air-de-xav:nodejs xav$ sudo node cortex_code_example.js 
Password:
(node:3628) Warning: Setting the NODE_TLS_REJECT_UNAUTHORIZED environment variable to '0' makes TLS connections and HTTPS requests insecure by disabling certificate verification.
You must login on CortexUI before request for grant access then rerun
(node:3628) UnhandledPromiseRejectionWarning: Error: You must login on CortexUI before request for grant access
    at Cortex.checkGrantAccessAndQuerySessionInfo (/Users/xav/Desktop/GitHub/cortex-v2-example/nodejs/cortex_code_example.js:400:19)
    at processTicksAndRejections (internal/process/task_queues.js:85:5)
    at async WebSocket.<anonymous> (/Users/xav/Desktop/GitHub/cortex-v2-example/nodejs/cortex_code_example.js:423:13)
(node:3628) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:3628) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

nguoithichkhampha, 2019-07-30:
@xaviervasques , plas make sure you are running cortex v2. note that example v2 only can work with cortex v2.

ertot, 2019-07-31:
If you want to run the example you need to first fill the part at bottom socketurl with websocket and user param for clientid, license and client secret (license is not essential since emotiv can get it from clientid) and after that approve it from your app and if it gets stuck try to delete the license section at authorize function since it takes it as unaccepted parameter for me.

dalozanoc, 2019-08-09:
@ertot I work with python 3.7 to access the services of cortex v2 with JSON as I understand you have problems accessing the cortex v2 for this version the websocket is wss://localhost:6868 because for v1 it was wss://emotivcortex.com:54321 and to find the ID and Secret you can place them in cortexApp in the official Emotiv page enter your profile and find that information, try to save the ClientSecret will only show you only once, then for the authorization once generated the ID and Secret uses JSON to generate an authorization to the cortex service from your PC with these parameters.
{
     "id": 1,
     "jsonrpc": "2.0",
     "method": "requestAccess",
     "params": {
         "clientId": "xxx", # replaces the aforementioned data
         "clientSecret": "xxx"
     }
}
  then check your Emotiv App and manually you will have to approve your request.

xaviervasques, 2019-08-12:
Hello and thank you for your help. 
I was not running the right version of cortex ...
Thank you again

closed 2019-08-12 by xaviervasques