Emotiv/cortex-example #27: Cortex-V2 / Node-red : "Please ensure that you have Cortex installed and running in the background"

Paulhb7, 2019-07-28:
Hello, 

I am actually testing EmotivBCI Node-RED Toolbox. Although EmotivApp and Emotiv BCI are open and running, Node-Red displays the following message "Please ensure that you have Cortex installed and running in the background". 

(I followed all the emotiv gitbook's instructions (Nodejs, Node-red and Node-red-contrib-emotiv-bci are correctly installed on the computer, my emotiv Insight is connected and I have a training profile with mental commands)

How can I solve this issue ?

Thanks,

Paul

ertot, 2019-07-28:
Hello i am having the same issue and couldnt solve it myself. I imagine it os a recent problem since everyone was okay before

wed3nsday, 2019-07-29:
Maybe it's possible to switch back to cortex V1 until it works ?

nguoithichkhampha, 2019-07-30:
did you run same version of node-red and Cortex ?

ertot, 2019-07-30:
I run the latest versions of both of them. It seems to be something about websockets maybe when i run example with a manual cresential and secret i get from emotiv andchange the path to my local host example works

bchaperon, 2019-07-30:
Hi @Paulhb7
First, you must check if you are using Cortex version 1.x or version 2.0
Just look at the version displayed by EmotivBCI, on the top of the window
Then check this page to know how to install the correct version of the node red toolbox 
https://www.npmjs.com/package/node-red-contrib-emotiv-bci

If it still doesn't work, then you should contact the [EMOTIV customer support.](https://www.emotiv.com/contact-support/)

ertot, 2019-07-30:
Example stops right before createsession function if you dont run it manually with the latest versions .i dont know if it was meant to be.

ertot, 2019-07-31:
Hello again i fixed the node red problem for myself what you need to do is go to node red contrib emotiv file in your node modules and change the clientid and clientsecret parameters from single quote  to double quote (from '  ' to "  ") and change websocket path to wss://localhost:6868/com.emotiv.noderedtoolbox in single quote and fire node red again

Paulhb7, 2019-08-01:
Working !

closed 2019-08-01 by Paulhb7