Emotiv/cortex-example #30: Cortex 2.1 Python 3.7 Data subscription correct but not displayed in python

dalozanoc, 2019-08-09:
Greetings Emotiv

A pleasure to greet you, I would like to help me with a problem that I have for now, in python 3.7 after successfully making the streams of "com" with the following lines of code:

{
   "id": 1,
   "jsonrpc": "2.0",
   "method": "subscribe",
   "params": {
       "cortexToken": "xxx",
       "session": "f8cb7289-9a92-438b-8281-e5fdffe8166e",
       "streams": ["com"]
   }
}

I get this answer:

{
   "id": 8,
   "jsonrpc": "2.0",
   "result": {
       "failure": [],
       "success": [{
           "cols": ["act", "pow"],
           "sid": "7f899d66-442b-4bf4-9752-ed06d57b72c3",
           "streamName": "com"
       }]
   }
}

same that allows me to obtain data of some specific activity but I wanted to please help me with the line of code that allows me to obtain this result for example:

{
   "com": ["push", 0.376],
   "sid": "7f899d66-442b-4bf4-9752-ed06d57b72c3",
   "time": 1559900743.3318
}

How can I print on screen the action that is currently being done with the Emotiv ?, I thank you very much in advance.

P.S. I am based on the use of its website: https://emotiv.gitbook.io/cortex-api/data-subscription/data-sample-object
but I can't find a way to show the specific action.

nguoithichkhampha, 2019-08-12:
@dalozanoc , I'm not clear about your question, please describe more details. 
in the theory, after you subscribe the data successful from Cortex. your app should able to receive the data stream through websocket.

dalozanoc, 2019-08-12:
Greettings @nguoithichkhampha  I am doing a job with the 5-channel Emotiv Insight device I am using cortex to extract the data such as mental commands and gyroscope, I have no problems at the time of creating cortex session and data subscription apparently but I do not throw the values ​​in In the case of the gyroscope I am using the following code in python

import json
from websocket import create_connection
import ssl
import time
import pyautogui

thought = ""
token = "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhcHBJZCI6ImNvbS5yaWNrc2Fuc29tLmRhdG9zZW1vdGl2IiwiYXBwVmVyc2lvbiI6IjEuMCIsImV4cCI6MTU2NTU1MDYxMSwibGljZW5zZUlkIjoiYTQxZDJkYzAtZWE1OS00ZDFhLWIwNTctODNkYWU1ZmJmZmM4IiwibmJmIjoxNTY1MjkxNDExLCJ1c2VySWQiOiI5Y2FiYTc5NS0yYTM2LTRlNWItOTJmNC1lMjYwZWNiYWUwMmQiLCJ1c2VybmFtZSI6InJpY2tzYW5zb20iLCJ2ZXJzaW9uIjoiMi4wIn0=.2fVCZY0aKOkKVwgHf7E4REQMPXDccRDhln3kUbPvAus="

print("Connecting a websocket...")
receivedData = create_connection("wss://localhost:6868", sslopt={"cert_reqs": ssl.CERT_NONE})

#------------------------------------------------------------------------
print("Revisa la conectividad con el auricular")
receivedData.send(json.dumps({
"jsonrpc": "2.0",
"method": "queryHeadsets",
"params": {},
"id": 1
}))

print(receivedData.recv())

#------------------------------------------------------------------------
print("Crear sesion")

receivedData.send(json.dumps({
"id": 1,
"jsonrpc": "2.0",
"method": "createSession",
"params": {
"cortexToken": token,
"headset": "INSIGHT-A1D201D2",
"status": "open"
}}))

#print(receivedData.recv())

idses=json.loads(receivedData.recv())["result"]
print(idses['id'])

#------------------------------------------------------------------------
print("\nSubscripcion de datos")

receivedData.send(json.dumps({
"id": 1,
"jsonrpc": "2.0",
"method": "subscribe",
"params": {
"cortexToken": token,
"session": idses['id'],
"streams": ["mot"]
}
}))

print(receivedData.recv())

I get the following results

image

But in case I do not throw the read values ​​of the device as shown in guides such as the following
https://emotiv.gitbook.io/cortex-api/data-subscription/data-sample-object

I want to get the following results of the gyroscope or mental commands in advance thank you very much

{
"mot":[
14,0,
8206,8187,8181,
4235,8668,8128,
8294,8237,7938
],
"sid":"462c4d75-113f-4664-a443-3aaa02c178d0",
"time":1559902927.7428
}

Rakesh420, 2019-09-05:
Change the steamName from "mot" to "com"   
"streamName":"com".
You need to go through these - 
https://emotiv.gitbook.io/cortex-api/bci/setupprofile
https://emotiv.gitbook.io/cortex-api/bci/getdetectioninfo
---------------------------
I am working on an eeg controlled rover using Emotiv Insight and this is my first ever project.

nguoithichkhampha, 2019-09-16:
@dalozanoc , you are using new insight headset with new motion chip. The data for this headset does not return gyroscope . it return quaternion data (Q0, Q1, Q2, Q3). you can search the formula convert quaternion to gyroscope.

closed 2019-09-16 by nguoithichkhampha