Emotiv/cortex-example #31: cortext.export_record: TypeError: Object of type set is not JSON serializable

xaviervasques, 2019-08-12:
Hi, 
I am trying to export data on a CSV file using python and for this I added the following lines to cortex.py

```
async def export_record(self, recordIds, folder, format, stream_types):
        params = {'cortexToken': self.auth_token,
                  'recordIds': recordIds,
                  'folder': folder,
                  'format': format,
                  'streamTypes': stream_types}
        resp = await self.send_command('exportRecord', **params)
        logger.debug(f"{__name__} resp:\n{resp}")
        return resp
```
I am calling the function in the main file (example.py) :

```
        while cortex.packet_count < 100:
            await cortex.get_data()
            await cortex.export_record(recordIds="test_record_1" ,folder="/Users/xav/Desktop/",format="CSV",stream_types={'pow','met'})
        await cortex.close_session()
```
My issue is to get the recordIds variable. Tried several things but get errors such as TypeError: Object of type set is not JSON serializable

My question is how can I get recordlds ? 

Thank you,

Xavier

nguoithichkhampha, 2019-08-12:
@xaviervasques seems your error because syntax error of python.
on another question to get recordId you can call "queryRecord" api 
https://emotiv.gitbook.io/cortex-api/records/queryrecords

xaviervasques, 2019-08-12:
Thank you @nguoithichkhampha for your help.  
How can I call the queryRecord api with python ?

I created the following lines in cortex.py:

```
    async def query_records(self, query, orderBy):
        params = {'cortexToken': self.auth_token,
                  'query': query,
                  'orderBy': orderBy}
        resp = await self.send_command('queryRecords', **params)
        logger.debug(f"{__name__} resp:\n{resp}")
        return resp
```

And I am trying to call the api in example.py by:

`await cortex.query_records("startDatetime","DESC")`

But it is the wrong way ... I have the following error:

Got error in queryRecords with params {'cortexToken': 'eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhcHBJZCI6ImNvbS5scmVuYy50ZXN0XzIiLCJhcHBWZXJzaW9uIjoiMS4wIiwiZXhwIjoxNTY1Nzk4OTg4LCJsaWNlbnNlSWQiOiI5YzU4NTA4NS01ZTNhLTQyZmEtYTE2NS0xMzUxY2ExOGFhN2EiLCJuYmYiOjE1NjU1Mzk3ODgsInVzZXJJZCI6IjQ2Zjc2NzQwLTFlYWYtNDAzOS04Yzk1LTdhMzFkZGFkOWYwZSIsInVzZXJuYW1lIjoibHJlbmMiLCJ2ZXJzaW9uIjoiMi4wIn0=.gAuqYh15+dBmNv5rV4V/qfbAg8v44LvGutUdaVarXPQ=', 'query': 'startDatetime', 'orderBy': 'DESC'}:
{"error":{"code":-32602,"message":"Invalid parameters"},"id":11,"jsonrpc":"2.0"}
Traceback (most recent call last):
  File "raw_data.py", line 54, in <module>
    test()
  File "raw_data.py", line 50, in test
    asyncio.run(do_stuff(cortex))
  File "/usr/local/Cellar/python/3.7.4/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "/usr/local/Cellar/python/3.7.4/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/base_events.py", line 579, in run_until_complete
    return future.result()
  File "raw_data.py", line 44, in do_stuff
    await cortex.query_records("startDatetime","DESC")
  File "/Users/xav/Desktop/Juju/raw_data/lib/cortex.py", line 380, in query_records
    resp = await self.send_command('queryRecords', **params)
  File "/Users/xav/Desktop/Juju/raw_data/lib/cortex.py", line 154, in send_command
    raise CortexApiException(resp)
lib.cortex.CortexApiException: {"error":{"code":-32602,"message":"Invalid parameters"},"id":11,"jsonrpc":"2.0"}

I tried different way but I didn't get it :)

await cortex.query_records(query={"applicationId": "com.emotiv.emotivpro"},orderBy={"startDatetime": "DESC"})
await cortex.query_records(query="applicationId": "com.emotiv.emotivpro","startDatetime": "DESC")
etc ...
Thank you in advance for your help,
Xavier

xaviervasques, 2019-08-12:
For exemple, when I try this way:

`await cortex.query_records({applicationId:"com.emotiv.emotivpro"},{startDatetime:"DESC"})`

I have the following:

Traceback (most recent call last):
  File "raw_data.py", line 64, in <module>
    test()
  File "raw_data.py", line 60, in test
    asyncio.run(do_stuff(cortex))
  File "/usr/local/Cellar/python/3.7.4/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "/usr/local/Cellar/python/3.7.4/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/base_events.py", line 579, in run_until_complete
    return future.result()
  File "raw_data.py", line 54, in do_stuff
    await cortex.query_records({startDatetime:"DESC"},{applicationId:"com.emotiv.emotivpro"})
NameError: name 'startDatetime' is not defined

xaviervasques, 2019-08-12:
What I do not understand is when I invoke export_record without recordIds:

`await cortex.export_record(folder="/Users/xav/Desktop/",format="CSV",stream_types={'pow','met'})`

and adding in cortex.py

```
async def export_record(self, folder, format, stream_types):
        params = {'cortexToken': self.auth_token,
            #'recordIds': recordIds,
                  'folder': folder,
                  'format': format,
                  'streamTypes': stream_types}
        resp = await self.send_command('exportRecord', **params)
        logger.debug(f"{__name__} resp:\n{resp}")
        return resp
```
I got the same as mentionned before:

Traceback (most recent call last):
  File "raw_data.py", line 64, in <module>
    test()
  File "raw_data.py", line 60, in test
    asyncio.run(do_stuff(cortex))
  File "/usr/local/Cellar/python/3.7.4/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "/usr/local/Cellar/python/3.7.4/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/base_events.py", line 579, in run_until_complete
    return future.result()
  File "raw_data.py", line 55, in do_stuff
    await cortex.export_record(folder="/Users/xav/Desktop/",format="CSV",stream_types={'pow','met'})
  File "/Users/xav/Desktop/Juju/raw_data/lib/cortex.py", line 373, in export_record
    resp = await self.send_command('exportRecord', **params)
  File "/Users/xav/Desktop/Juju/raw_data/lib/cortex.py", line 148, in send_command
    msg = self.gen_request(method, auth, **kwargs)
  File "/Users/xav/Desktop/Juju/raw_data/lib/cortex.py", line 113, in gen_request
    'id': self.id_sequence
  File "/usr/local/Cellar/python/3.7.4/Frameworks/Python.framework/Versions/3.7/lib/python3.7/json/__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "/usr/local/Cellar/python/3.7.4/Frameworks/Python.framework/Versions/3.7/lib/python3.7/json/encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/usr/local/Cellar/python/3.7.4/Frameworks/Python.framework/Versions/3.7/lib/python3.7/json/encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "/usr/local/Cellar/python/3.7.4/Frameworks/Python.framework/Versions/3.7/lib/python3.7/json/encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
TypeError: Object of type set is not JSON serializable

NguyenDuyToan, 2019-08-13:
@xaviervasques there're few problems with your source code:
- The error 
> TypeError: Object of type set is not JSON serializable

caused by your `stream_types` parameter, you should pass a list, not a set

- `await cortex.query_records({applicationId:"com.emotiv.emotivpro"},{startDatetime:"DESC"})` is  invalid python syntax, the parameter that you're passing to the method call is not a valid python dict.

xaviervasques, 2019-08-13:
Thanks for your reply. I found an another way with subscribe to save on CSV file. But for the others who are interested on querying data, can you write the correct syntax ? I tried various ones and did not work for me.

NguyenDuyToan, 2019-08-14:
@xaviervasques 
- For the stream_types, the example module use build-in python json library, and the library does not support serialize a set, so you should pass it as a list. More information can be found [here](https://docs.python.org/3/library/json.html).
So, instead of
`await cortex.export_record(...,stream_types={'pow','met'})`, 
the function call as `await cortex.export_record(...,stream_types=['pow','met'])` should work.

-  For the 2nd issue
> For exemple, when I try this way:
> 
> `await cortex.query_records({applicationId:"com.emotiv.emotivpro"},{startDatetime:"DESC"})`
> 
> I have the following:
> 
> Traceback (most recent call last):
> File "raw_data.py", line 64, in 
> test()
> File "raw_data.py", line 60, in test
> asyncio.run(do_stuff(cortex))
> File "/usr/local/Cellar/python/3.7.4/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/runners.py", line 43, in run
> return loop.run_until_complete(main)
> File "/usr/local/Cellar/python/3.7.4/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/base_events.py", line 579, in run_until_complete
> return future.result()
> File "raw_data.py", line 53, in do_stuff
> await cortex.query_records({applicationId:"com.emotiv.emotivpro"},{startDatetime:"DESC"})
> NameError: name 'applicationId' is not defined

as I mentioned above, `{applicationId:"com.emotiv.emotivpro"}` is not a valid python dict syntax, you must quote the `applicationId` as the key of the dict. The correct syntax for a python dict should be `{"applicationId": "com.emotiv.emotivpro"}`

I hope that helps!

closed 2019-08-27 by bchaperon