Emotiv/cortex-example #32: Export data to CSV with Python

xaviervasques, 2019-08-12:
Hello, 
If someone has a simple way to export data to a csv file using python it would be very helpful (few lines of code). In addition, getting the EEG data for exemple on a float array would be great. 
Thank you,
Xavier

nguoithichkhampha, 2019-08-13:
Actually, you can only export data via Emotiv Pro app. another way to save data to csv file is subscribe data from cortex api and write csv file by yourself.

xaviervasques, 2019-08-13:
Thank you very much for your reply. I know for Emotiv Pro. You can export data on a CSV and analyse it after the export. I would like to code a stream of data that I can analyze in real time. If subscribe data from cortex api is working, can you please give me an example on how to use it with python ?

xaviervasques, 2019-08-13:
Thanks. Indeed with subscribe data is the best way. You can do json loads and after put them the way you want.

closed 2019-08-13 by xaviervasques