Emotiv/cortex-example #36: error "The stream data is not supported"

paz1, 2019-08-27:
Hello,  I would like you to help me with a problem that I have.
I'm trying to run the cortex_example_export_record.py to get the csv file, but the export_record method doesn't work well.
What can cause the problem?
Thank you!

export record request  {
    "jsonrpc": "2.0",
    "id": 113,
    "method": "exportRecord",
    "params": {
        "cortexToken": "***",
        "folder": "C:\\export",
        "format": "CSV",
        "streamTypes": [
            "EEG"
        ],
        "recordIds": [
            "***"
        ],
        "version": "V2"
    }
}

export record result {
    "jsonrpc": "2.0",
    "warning": {
        "code": 0,
        "message": "All subscriptions of session *** was stopped by Cortex"
    }
}

export record result {
    "id": 113,
    "jsonrpc": "2.0",
    "result": {
        "failure": [
            {
                "code": -32138,
                "message": "The stream data is not supported.",
                "recordId": "***"
            }
        ],
        "success": []
    }
}

bchaperon, 2019-08-28:
Hello @paz1 
The most likely reason is that the license you used to create this record doesn't include the access to the EEG data.
Please use the method [getRecordInfos](https://emotiv.gitbook.io/cortex-api/records/getrecordinfos) to get the record, and check the field "licenseScope" of your record.
If the licenseScope doesn't include the scope "eeg" then you cannot export the EEG data.
See https://emotiv.gitbook.io/cortex-api/records/exportrecord for more details.

paz1, 2019-08-29:
I have fixed it and get permission for the EEG data but now I get this error-

export record request  {
    "jsonrpc": "2.0",
    "id": 113,
    "method": "exportRecord",
    "params": {
        "cortexToken": "***",
        "folder": "C:\\export",
        "format": "CSV",
        "recordIds": [
            "***"
        ],
        "streamTypes": [
            "EEG"
        ],
        "version": "V2"
    }
}
export record result {
    "jsonrpc": "2.0",
    "warning": {
        "code": 0,
        "message": "All subscriptions of session *** was stopped by Cortex"
    }
}
export record result {
    "id": 113,
    "jsonrpc": "2.0",
    "result": {
        "failure": [
            {
                "code": -32137,
                "message": "Cortex can not open record data file.",
                "recordId": "***"
            }
        ],
        "success": []
    }
}

Thank you!

bchaperon, 2019-08-29:
Cortex cannot create the CSV file.
Did you create the folder "C:\export"? If not then you must create it before you call the export API.

closed 2019-08-29 by paz1