Emotiv/cortex-example #37: Cortex-v2 seems to have massively increased latency vs. Cortex-v1

shohamj, 2019-08-30:
Consider the following sample code:

[https://gist.github.com/shohamj/ee93f5b86f226e84c98f11e8b8356404](https://gist.github.com/shohamj/ee93f5b86f226e84c98f11e8b8356404)

On my system:

```
Example Output (No headsets are connected):
** Cortex V1 **
Connection took 0.0929114818572998 seconds
Query took 0.0009982585906982422 seconds
Headsets: []

** Cortex V2 **
Connection took 0.006994962692260742 seconds
Query took 1.0240464210510254 seconds
Headsets: []
```

nguoithichkhampha, 2019-08-30:
@shohamj , thanks for your feedback. we will check again our system.

tungntEmotiv, 2019-10-31:
Hi @shohamj ,

Thanks for your feedback. In our checking, the queryHeadsets time is around ~7 milliseconds. Please update the latest code of Cortex and re-check again. Thanks

closed 2019-11-15 by nguoithichkhampha