Emotiv/cortex-example #42: {"error":{"code":-32019,"message":"Session limit on this device has been reached."},"id":8,"jsonrpc":"2.0"}

chx2017, 2019-09-16:
We have created the cortex_creds using Notepad, including the below information. And the EMOTIV App has been connected with the equipment. Then the example.py has been run and the obtained information is shown as below. We have tried many ways and can not solve it. Please you help us!

client_id JwoNlMmZ7LqXAafsbr3ldnGIoF2F0W5DdUQ27mN9
client_secret D9LHZtzlgYvBEJegWLCMNXhyXCEIimga0W3PSTLJNaOMJDjCXzC8ULmZQ1bXOqaCCrsVbBTOec1bbPhXXCskfk2DbqMrRiDFgA8Ia2pAQhDXjnsAkJZPqcmeRo5bNkE8

D:\Python\Python37\python.exe E:/myTFileschx/EmotivTesting/cortex-v2-example-master/python/example.py
** USER LOGIN **
Sending request:
{"jsonrpc": "2.0", "method": "getUserLogin", "params": {}, "id": 1}
sent; awaiting response
lib.cortex resp:
{'currentOSUId': '1001', 'currentOSUsername': 'lenovo', 'loggedInOSUId': '1001', 'loggedInOSUsername': 'lenovo', 'username': '2259b2ef9eeebb5'}
Sending request:
{"jsonrpc": "2.0", "method": "getCortexInfo", "params": {}, "id": 2}
sent; awaiting response
lib.cortex resp:
{'id': 2, 'jsonrpc': '2.0', 'result': {'buildDate': '2019-09-03T09:40:00', 'buildNumber': 'v2.0.2-417-g3845976', 'version': '2.2.0'}}
Sending request:
{"jsonrpc": "2.0", "method": "requestAccess", "params": {"clientId": "JwoNlMmZ7LqXAafsbr3ldnGIoF2F0W5DdUQ27mN9", "clientSecret": "D9LHZtzlgYvBEJegWLCMNXhyXCEIimga0W3PSTLJNaOMJDjCXzC8ULmZQ1bXOqaCCrsVbBTOec1bbPhXXCskfk2DbqMrRiDFgA8Ia2pAQhDXjnsAkJZPqcmeRo5bNkE8"}, "id": 3}
sent; awaiting response
** GET CORTEX INFO **
** HAS ACCESS RIGHT **
lib.cortex resp:
{'id': 3, 'jsonrpc': '2.0', 'result': {'accessGranted': True, 'message': 'The User has access right to this application.'}}
Sending request:
** REQUEST ACCESS **
{"jsonrpc": "2.0", "method": "requestAccess", "params": {"clientId": "JwoNlMmZ7LqXAafsbr3ldnGIoF2F0W5DdUQ27mN9", "clientSecret": "D9LHZtzlgYvBEJegWLCMNXhyXCEIimga0W3PSTLJNaOMJDjCXzC8ULmZQ1bXOqaCCrsVbBTOec1bbPhXXCskfk2DbqMrRiDFgA8Ia2pAQhDXjnsAkJZPqcmeRo5bNkE8"}, "id": 4}
sent; awaiting response
** AUTHORIZE **
lib.cortex resp:
{'id': 4, 'jsonrpc': '2.0', 'result': {'accessGranted': True, 'message': 'The User has access right to this application.'}}
Sending request:
{"jsonrpc": "2.0", "method": "authorize", "params": {"clientId": "JwoNlMmZ7LqXAafsbr3ldnGIoF2F0W5DdUQ27mN9", "clientSecret": "D9LHZtzlgYvBEJegWLCMNXhyXCEIimga0W3PSTLJNaOMJDjCXzC8ULmZQ1bXOqaCCrsVbBTOec1bbPhXXCskfk2DbqMrRiDFgA8Ia2pAQhDXjnsAkJZPqcmeRo5bNkE8"}, "id": 5}
sent; awaiting response
lib.cortex resp:
{'id': 5, 'jsonrpc': '2.0', 'result': {'cortexToken': 'eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhcHBJZCI6ImNvbS4yMjU5YjJlZjllZWViYjUuamFja2Nhb18yMDE5ODMwIiwiYXBwVmVyc2lvbiI6IjEuMCIsImV4cCI6MTU2ODc2NjU1MSwibGljZW5zZUlkIjoiYWZiMzU5NGMtNTQ2OS00YTFmLWE3MjctNDQ5OTE2NjU4NGEyIiwibmJmIjoxNTY4NTA3MzUxLCJ1c2VySWQiOiIyOTkwZDlmNy1jMWYyLTQ1YWUtYmNmMy0zMjMzODExZDM5MzciLCJ1c2VybmFtZSI6IjIyNTliMmVmOWVlZWJiNSIsInZlcnNpb24iOiIyLjAifQ==.09oJ1w6mDPucPxlSFzHUp6A+o7GeigZxTzacHpbaSaE='}}
Sending request:
{"jsonrpc": "2.0", "method": "getLicenseInfo", "params": {"cortexToken": "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhcHBJZCI6ImNvbS4yMjU5YjJlZjllZWViYjUuamFja2Nhb18yMDE5ODMwIiwiYXBwVmVyc2lvbiI6IjEuMCIsImV4cCI6MTU2ODc2NjU1MSwibGljZW5zZUlkIjoiYWZiMzU5NGMtNTQ2OS00YTFmLWE3MjctNDQ5OTE2NjU4NGEyIiwibmJmIjoxNTY4NTA3MzUxLCJ1c2VySWQiOiIyOTkwZDlmNy1jMWYyLTQ1YWUtYmNmMy0zMjMzODExZDM5MzciLCJ1c2VybmFtZSI6IjIyNTliMmVmOWVlZWJiNSIsInZlcnNpb24iOiIyLjAifQ==.09oJ1w6mDPucPxlSFzHUp6A+o7GeigZxTzacHpbaSaE="}, "id": 6}
sent; awaiting response
** GET LICENSE INFO **
** QUERY HEADSETS **
** CREATE SESSION **
lib.cortex resp:
{'id': 6, 'jsonrpc': '2.0', 'result': {'isOnline': True, 'license': {'applications': ['com.emotiv.sdk.basic'], 'billingFrom': '2019-08-27T08:00:00.000+08:00', 'billingTo': '2029-08-24T08:00:00.000+08:00', 'deviceInfo': {'deviceLimit': 3, 'devicesPerSeat': 3, 'sessionLimit': {'day': None, 'month': None, 'year': None}}, 'expired': False, 'extenderLimit': 6, 'hardLimitTime': '2019-10-05T07:59:59.999+08:00', 'isCommercial': False, 'licenseId': 'afb3594c-5469-4a1f-a727-4499166584a2', 'licenseName': 'Basic License', 'localQuota': 0, 'maxDebit': None, 'scopes': [''], 'seatCount': 1, 'sessionCount': 0, 'softLimitTime': '2019-09-28T07:59:59.999+08:00', 'totalDebit': 0, 'totalRegisteredDevices': 1, 'validFrom': '2019-08-27T08:00:00.000+08:00', 'validTo': '2029-08-25T07:59:59.999+08:00'}}}
Sending request:
{"jsonrpc": "2.0", "method": "queryHeadsets", "params": {}, "id": 7}
sent; awaiting response
lib.cortex found headsets ['EPOCPLUS-3B9AEBCD']
lib.cortex resp:
{'id': 7, 'jsonrpc': '2.0', 'result': [{'connectedBy': 'dongle', 'customName': '', 'dongle': '6ff', 'firmware': '625', 'id': 'EPOCPLUS-3B9AEBCD', 'motionSensors': ['GYROX', 'GYROY', 'GYROZ', 'ACCX', 'ACCY', 'ACCZ', 'MAGX', 'MAGY', 'MAGZ'], 'sensors': ['AF3', 'F7', 'F3', 'FC5', 'T7', 'P7', 'O1', 'O2', 'P8', 'T8', 'FC6', 'F4', 'F8', 'AF4'], 'settings': {'eegRate': 256, 'eegRes': 16, 'memsRate': 0, 'memsRes': 16, 'mode': 'EPOCPLUS'}, 'status': 'connected'}]}
Sending request:
{"jsonrpc": "2.0", "method": "createSession", "params": {"cortexToken": "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhcHBJZCI6ImNvbS4yMjU5YjJlZjllZWViYjUuamFja2Nhb18yMDE5ODMwIiwiYXBwVmVyc2lvbiI6IjEuMCIsImV4cCI6MTU2ODc2NjU1MSwibGljZW5zZUlkIjoiYWZiMzU5NGMtNTQ2OS00YTFmLWE3MjctNDQ5OTE2NjU4NGEyIiwibmJmIjoxNTY4NTA3MzUxLCJ1c2VySWQiOiIyOTkwZDlmNy1jMWYyLTQ1YWUtYmNmMy0zMjMzODExZDM5MzciLCJ1c2VybmFtZSI6IjIyNTliMmVmOWVlZWJiNSIsInZlcnNpb24iOiIyLjAifQ==.09oJ1w6mDPucPxlSFzHUp6A+o7GeigZxTzacHpbaSaE=", "headset": "EPOCPLUS-3B9AEBCD", "status": "active"}, "id": 8}
sent; awaiting response
Got error in createSession with params {'cortexToken': 'eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhcHBJZCI6ImNvbS4yMjU5YjJlZjllZWViYjUuamFja2Nhb18yMDE5ODMwIiwiYXBwVmVyc2lvbiI6IjEuMCIsImV4cCI6MTU2ODc2NjU1MSwibGljZW5zZUlkIjoiYWZiMzU5NGMtNTQ2OS00YTFmLWE3MjctNDQ5OTE2NjU4NGEyIiwibmJmIjoxNTY4NTA3MzUxLCJ1c2VySWQiOiIyOTkwZDlmNy1jMWYyLTQ1YWUtYmNmMy0zMjMzODExZDM5MzciLCJ1c2VybmFtZSI6IjIyNTliMmVmOWVlZWJiNSIsInZlcnNpb24iOiIyLjAifQ==.09oJ1w6mDPucPxlSFzHUp6A+o7GeigZxTzacHpbaSaE=', 'headset': 'EPOCPLUS-3B9AEBCD', 'status': 'active'}:
{"error":{"code":-32019,"message":"Session limit on this device has been reached."},"id":8,"jsonrpc":"2.0"}
Traceback (most recent call last):
File "E:/myTFileschx/EmotivTesting/cortex-v2-example-master/python/example.py", line 44, in <module>
test()
File "E:/myTFileschx/EmotivTesting/cortex-v2-example-master/python/example.py", line 39, in test
asyncio.run(do_stuff(cortex))
File "D:\Python\Python37\lib\asyncio\runners.py", line 43, in run
return loop.run_until_complete(main)
File "D:\Python\Python37\lib\asyncio\base_events.py", line 579, in run_until_complete
return future.result()
File "E:/myTFileschx/EmotivTesting/cortex-v2-example-master/python/example.py", line 23, in do_stuff
await cortex.create_session(activate=True, headset_id=cortex.headsets[0])
File "E:\myTFileschx\EmotivTesting\cortex-v2-example-master\python\lib\cortex.py", line 271, in create_session
resp = await self.send_command('createSession', **params)
File "E:\myTFileschx\EmotivTesting\cortex-v2-example-master\python\lib\cortex.py", line 154, in send_command
raise CortexApiException(resp)
lib.cortex.CortexApiException: {"error":{"code":-32019,"message":"Session limit on this device has been reached."},"id":8,"jsonrpc":"2.0"}

Process finished with exit code 1

nguoithichkhampha, 2019-09-16:
by the default python example does not set debit number in "authorize" request. You should set it with a big number enough (may be 100)
take a look in function
async def authorize(self, license_id=None, debit=None) of file cortex.py

chx2017, 2019-09-16:
Dear Cuong Trinh Manh，

We have run the example.py with the value (debit = 500) and obtained the below information.
(1) Is the number of get_data got {"pow":[.......]}  the EEG data (red color)?  How to define the corresponding items for the different EEG channels？
(2) The example.py can not run continuously，and broken as shown with the blue information after several cycles. what is problem with it?
async def authorize(self, license_id=None, debit=500):
D:\Python\Python37\python.exe E:/myTFileschx/EmotivTesting/cortex-v2-example-master/python/example.py
** USER LOGIN **
Sending request:
{"jsonrpc": "2.0", "method": "getUserLogin", "params": {}, "id": 1}
sent; awaiting response
** GET CORTEX INFO **
** HAS ACCESS RIGHT **
lib.cortex resp:
{'currentOSUId': '1001', 'currentOSUsername': 'lenovo', 'loggedInOSUId': '1001', 'loggedInOSUsername': 'lenovo', 'username': '2259b2ef9eeebb5'}
Sending request:
{"jsonrpc": "2.0", "method": "getCortexInfo", "params": {}, "id": 2}
sent; awaiting response
lib.cortex resp:
{'id': 2, 'jsonrpc': '2.0', 'result': {'buildDate': '2019-09-03T09:40:00', 'buildNumber': 'v2.0.2-417-g3845976', 'version': '2.2.0'}}
Sending request:
{"jsonrpc": "2.0", "method": "requestAccess", "params": {"clientId": "JwoNlMmZ7LqXAafsbr3ldnGIoF2F0W5DdUQ27mN9", "clientSecret": "D9LHZtzlgYvBEJegWLCMNXhyXCEIimga0W3PSTLJNaOMJDjCXzC8ULmZQ1bXOqaCCrsVbBTOec1bbPhXXCskfk2DbqMrRiDFgA8Ia2pAQhDXjnsAkJZPqcmeRo5bNkE8"}, "id": 3}
sent; awaiting response
** REQUEST ACCESS **
lib.cortex resp:
{'id': 3, 'jsonrpc': '2.0', 'result': {'accessGranted': True, 'message': 'The User has access right to this application.'}}
Sending request:
{"jsonrpc": "2.0", "method": "requestAccess", "params": {"clientId": "JwoNlMmZ7LqXAafsbr3ldnGIoF2F0W5DdUQ27mN9", "clientSecret": "D9LHZtzlgYvBEJegWLCMNXhyXCEIimga0W3PSTLJNaOMJDjCXzC8ULmZQ1bXOqaCCrsVbBTOec1bbPhXXCskfk2DbqMrRiDFgA8Ia2pAQhDXjnsAkJZPqcmeRo5bNkE8"}, "id": 4}
sent; awaiting response
lib.cortex resp:
** AUTHORIZE **
{'id': 4, 'jsonrpc': '2.0', 'result': {'accessGranted': True, 'message': 'The User has access right to this application.'}}
Sending request:
{"jsonrpc": "2.0", "method": "authorize", "params": {"clientId": "JwoNlMmZ7LqXAafsbr3ldnGIoF2F0W5DdUQ27mN9", "clientSecret": "D9LHZtzlgYvBEJegWLCMNXhyXCEIimga0W3PSTLJNaOMJDjCXzC8ULmZQ1bXOqaCCrsVbBTOec1bbPhXXCskfk2DbqMrRiDFgA8Ia2pAQhDXjnsAkJZPqcmeRo5bNkE8", "debit": 500}, "id": 5}
sent; awaiting response
lib.cortex resp:
** GET LICENSE INFO **
{'id': 5, 'jsonrpc': '2.0', 'result': {'cortexToken': 'eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhcHBJZCI6ImNvbS4yMjU5YjJlZjllZWViYjUuamFja2Nhb18yMDE5ODMwIiwiYXBwVmVyc2lvbiI6IjEuMCIsImV4cCI6MTU2ODc5Nzg5NSwibGljZW5zZUlkIjoiYWZiMzU5NGMtNTQ2OS00YTFmLWE3MjctNDQ5OTE2NjU4NGEyIiwibmJmIjoxNTY4NTM4Njk1LCJ1c2VySWQiOiIyOTkwZDlmNy1jMWYyLTQ1YWUtYmNmMy0zMjMzODExZDM5MzciLCJ1c2VybmFtZSI6IjIyNTliMmVmOWVlZWJiNSIsInZlcnNpb24iOiIyLjAifQ==.Q345u0EiJbaC51zx++O47EhZI9Nw5Bm0jG+wYMa5FPg='}}
Sending request:
{"jsonrpc": "2.0", "method": "getLicenseInfo", "params": {"cortexToken": "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhcHBJZCI6ImNvbS4yMjU5YjJlZjllZWViYjUuamFja2Nhb18yMDE5ODMwIiwiYXBwVmVyc2lvbiI6IjEuMCIsImV4cCI6MTU2ODc5Nzg5NSwibGljZW5zZUlkIjoiYWZiMzU5NGMtNTQ2OS00YTFmLWE3MjctNDQ5OTE2NjU4NGEyIiwibmJmIjoxNTY4NTM4Njk1LCJ1c2VySWQiOiIyOTkwZDlmNy1jMWYyLTQ1YWUtYmNmMy0zMjMzODExZDM5MzciLCJ1c2VybmFtZSI6IjIyNTliMmVmOWVlZWJiNSIsInZlcnNpb24iOiIyLjAifQ==.Q345u0EiJbaC51zx++O47EhZI9Nw5Bm0jG+wYMa5FPg="}, "id": 6}
sent; awaiting response
** QUERY HEADSETS **
lib.cortex resp:
{'id': 6, 'jsonrpc': '2.0', 'result': {'isOnline': True, 'license': {'applications': ['com.emotiv.sdk.basic'], 'billingFrom': '2019-08-27T08:00:00.000+08:00', 'billingTo': '2029-08-24T08:00:00.000+08:00', 'deviceInfo': {'deviceLimit': 3, 'devicesPerSeat': 3, 'sessionLimit': {'day': None, 'month': None, 'year': None}}, 'expired': False, 'extenderLimit': 6, 'hardLimitTime': '2019-10-05T07:59:59.999+08:00', 'isCommercial': False, 'licenseId': 'afb3594c-5469-4a1f-a727-4499166584a2', 'licenseName': 'Basic License', 'localQuota': 1199, 'maxDebit': None, 'scopes': [''], 'seatCount': 1, 'sessionCount': 0, 'softLimitTime': '2019-09-28T07:59:59.999+08:00', 'totalDebit': 1200, 'totalRegisteredDevices': 1, 'validFrom': '2019-08-27T08:00:00.000+08:00', 'validTo': '2029-08-25T07:59:59.999+08:00'}}}
Sending request:
{"jsonrpc": "2.0", "method": "queryHeadsets", "params": {}, "id": 7}
sent; awaiting response
** CREATE SESSION **
lib.cortex found headsets ['EPOCPLUS-3B9AEBCD']
lib.cortex resp:
{'id': 7, 'jsonrpc': '2.0', 'result': [{'connectedBy': 'dongle', 'customName': '', 'dongle': '6ff', 'firmware': '625', 'id': 'EPOCPLUS-3B9AEBCD', 'motionSensors': ['GYROX', 'GYROY', 'GYROZ', 'ACCX', 'ACCY', 'ACCZ', 'MAGX', 'MAGY', 'MAGZ'], 'sensors': ['AF3', 'F7', 'F3', 'FC5', 'T7', 'P7', 'O1', 'O2', 'P8', 'T8', 'FC6', 'F4', 'F8', 'AF4'], 'settings': {'eegRate': 256, 'eegRes': 16, 'memsRate': 0, 'memsRes': 16, 'mode': 'EPOCPLUS'}, 'status': 'connected'}]}
Sending request:
{"jsonrpc": "2.0", "method": "createSession", "params": {"cortexToken": "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhcHBJZCI6ImNvbS4yMjU5YjJlZjllZWViYjUuamFja2Nhb18yMDE5ODMwIiwiYXBwVmVyc2lvbiI6IjEuMCIsImV4cCI6MTU2ODc5Nzg5NSwibGljZW5zZUlkIjoiYWZiMzU5NGMtNTQ2OS00YTFmLWE3MjctNDQ5OTE2NjU4NGEyIiwibmJmIjoxNTY4NTM4Njk1LCJ1c2VySWQiOiIyOTkwZDlmNy1jMWYyLTQ1YWUtYmNmMy0zMjMzODExZDM5MzciLCJ1c2VybmFtZSI6IjIyNTliMmVmOWVlZWJiNSIsInZlcnNpb24iOiIyLjAifQ==.Q345u0EiJbaC51zx++O47EhZI9Nw5Bm0jG+wYMa5FPg=", "headset": "EPOCPLUS-3B9AEBCD", "status":  [... cut ...] ef9eeebb5', 'recordIds': ['1f40e380-845c-4051-aca6-a2aed85c3596'], 'recording': False, 'started': '2019-09-16T17:11:35.765+08:00', 'status': 'closed', 'stopped': '2019-09-16T17:11:38.171+08:00', 'streams': []}}
E:\myTFileschx\EmotivTesting\cortex-v2-example-master\python\lib\cortex.py:173: RuntimeWarning: coroutine 'WebSocketCommonProtocol.close' was never awaited
  self.websocket.close()
RuntimeWarning: Enable tracemalloc to get the object allocation traceback

Process finished with exit code 0

chx2017, 2019-09-16:
Dear Cuong Trinh Manh，

You can see the attached files. The EEG data are obtained from EmotivPRO V2.2.0 and example.py.
There are 14 channels of data in EmotivPRO V2.20, However, there are only 10 channels of data from example.py.

Best regards
Jack Cao

nguoithichkhampha, 2019-09-16:
the data you send above is not eeg data. it's band power data. and I see band power data have value for all 14 channels.

chx2017, 2019-09-17:
Thank you for your help!

bchaperon, 2019-09-17:
Hi @chx2017 
Please note that you should never share the client secret of your application with anyone, and so you should never post it on a forum like this one.
As the name implies, your client secret must remain a secret :)

chx2017, 2019-09-18:
I have set the debit value in cortex.py [async def authorize(self, license_id=None, debit=100)].
The example.py has been run and the obtained results are shown as below. I do not understand the problem clearly. 
Can you tell me clearly how to obtain the EEG data from the equipment using this python SDK?
____________________________________________________________________________________________________________

** SUBSCRIBE POW & MET **
lib.cortex resp:
{'id': 9, 'jsonrpc': '2.0', 'result': {'record': {'applicationId': 'com.2259b2ef9eeebb5.jackcao_2019830', 'applicationVersion': '1.0', 'description': '', 'endDatetime': '', 'experimentId': 0, 'licenseId': 'afb3594c-5469-4a1f-a727-4499166584a2', 'licenseScope': [''], 'ownerId': '2990d9f7-c1f2-45ae-bcf3-3233811d3937', 'startDatetime': '2019-09-18T14:20:58.275678+08:00', 'tags': [], 'title': 'test record 1', 'uuid': '0e8325a3-8abd-4c89-a32e-29be8bb46334'}, 'sessionId': '2f814d65-38fe-40fe-b25c-1b3d12d3d057'}}
Sending request:
{"jsonrpc": "2.0", "method": "subscribe", "params": {"cortexToken": "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhcHBJZCI6ImNvbS4yMjU5YjJlZjllZWViYjUuamFja2Nhb18yMDE5ODMwIiwiYXBwVmVyc2lvbiI6IjEuMCIsImV4cCI6MTU2ODk2MDQ1OCwibGljZW5zZUlkIjoiYWZiMzU5NGMtNTQ2OS00YTFmLWE3MjctNDQ5OTE2NjU4NGEyIiwibmJmIjoxNTY4NzAxMjU4LCJ1c2VySWQiOiIyOTkwZDlmNy1jMWYyLTQ1YWUtYmNmMy0zMjMzODExZDM5MzciLCJ1c2VybmFtZSI6IjIyNTliMmVmOWVlZWJiNSIsInZlcnNpb24iOiIyLjAifQ==.dVxcV3pGJ4GWkgtBSGygoS8c0EapLsL8A/gmvk4u0ZE=", "session": "2f814d65-38fe-40fe-b25c-1b3d12d3d057", "streams": ["pow", "met"]}, "id": 10}
sent; awaiting response
lib.cortex resp:
{'id': 10, 'jsonrpc': '2.0', 'result': {'failure': [], 'success': [{'cols': ['AF3/theta', 'AF3/alpha', 'AF3/betaL', 'AF3/betaH', 'AF3/gamma', 'F7/theta', 'F7/alpha', 'F7/betaL', 'F7/betaH', 'F7/gamma', 'F3/theta', 'F3/alpha', 'F3/betaL', 'F3/betaH', 'F3/gamma', 'FC5/theta', 'FC5/alpha', 'FC5/betaL', 'FC5/betaH', 'FC5/gamma', 'T7/theta', 'T7/alpha', 'T7/betaL', 'T7/betaH', 'T7/gamma', 'P7/theta', 'P7/alpha', 'P7/betaL', 'P7/betaH', 'P7/gamma', 'O1/theta', 'O1/alpha', 'O1/betaL', 'O1/betaH', 'O1/gamma', 'O2/theta', 'O2/alpha', 'O2/betaL', 'O2/betaH', 'O2/gamma', 'P8/theta', 'P8/alpha', 'P8/betaL', 'P8/betaH', 'P8/gamma', 'T8/theta', 'T8/alpha', 'T8/betaL', 'T8/betaH', 'T8/gamma', 'FC6/theta', 'FC6/alpha', 'FC6/betaL', 'FC6/betaH', 'FC6/gamma', 'F4/theta', 'F4/alpha', 'F4/betaL', 'F4/betaH', 'F4/gamma', 'F8/theta', 'F8/alpha', 'F8/betaL', 'F8/betaH', 'F8/gamma', 'AF4/theta', 'AF4/alpha', 'AF4/betaL', 'AF4/betaH', 'AF4/gamma'], 'sid': '2f814d65-38fe-40fe-b25c-1b3d12d3d057', 'streamName': 'pow'}, {'cols': ['eng', 'exc', 'lex', 'str', 'rel', 'int', 'foc'], 'sid': '2f814d65-38fe-40fe-b25c-1b3d12d3d057', 'streamName': 'met'}]}}
get_data got {"pow":[0.036,0.02,0.026,0.032,0.024,0.059,0.016,0.035,0.153,0.053,0.117,0.05,0.09,0.297,0.047,0.08,0.027,0.097,0.166,0.025,0.819,0.937,4.692,7.552,5.018,0.133,0.061,0.159,0.266,0.046,0.086,0.058,0.162,0.213,0.042,0.076,0.085,0.233,1.996,0.098,0.163,0.056,0.1,0.329,0.022,0.027,0.034,0.019,0.011,0.025,0.21,0.059,0.289,0.924,0.809,0.325,0.066,1.704,8.527,12.445,0.035,0.037,0.041,0.043,0.018,0.605,0.58,2.4,3.446,4.739],"sid":"2f814d65-38fe-40fe-b25c-1b3d12d3d057","time":1568787658.3346}
get_data got {"pow":[0.035,0.02,0.025,0.034,0.024,0.048,0.015,0.042,0.154,0.056,0.13,0.05,0.102,0.274,0.05,0.079,0.022,0.095,0.174,0.024,0.691,0.943,4.4,7.311,5,0.144,0.053,0.13,0.264,0.049,0.085,0.048,0.155,0.202,0.039,0.078,0.096,0.264,2.017,0.097,0.179,0.071,0.12,0.33,0.022,0.025,0.032,0.016,0.012,0.024,0.202,0.088,0.298,0.98,0.852,0.343,0.052,1.615,8.491,12.357,0.039,0.042,0.038,0.039,0.017,0.515,0.741,2.601,3.501,4.789],"sid":"2f814d65-38fe-40fe-b25c-1b3d12d3d057","time":1568787658.4606}
get_data got {"pow":[0.033,0.019,0.025,0.035,0.023,0.042,0.016,0.051,0.159,0.057,0.141,0.049,0.11,0.26,0.053,0.078,0.021,0.087,0.178,0.023,0.529,0.923,3.938,6.983,4.965,0.159,0.05,0.108,0.256,0.051,0.087,0.045,0.138,0.189,0.036,0.077,0.101,0.307,2.063,0.092,0.187,0.087,0.147,0.321,0.025,0.022,0.028,0.014,0.014,0.022,0.199,0.123,0.333,1.161,0.965,0.347,0.044,1.53,8.454,12.18,0.045,0.047,0.032,0.034,0.016,0.512,0.948,2.778,3.561,4.734],"sid":"2f814d65-38fe-40fe-b25c-1b3d12d3d057","time":1568787658.5866}
get_data got {"pow":[0.03,0.018,0.026,0.037,0.024,0.041,0.02,0.057,0.177,0.057,0.147,0.046,0.114,0.27,0.056,0.079,0.022,0.077,0.173,0.023,0.378,0.886,3.422,6.625,4.889,0.177,0.059,0.1,0.244,0.05,0.092,0.052,0.119,0.176,0.031,0.073,0.097,0.342,2.121,0.085,0.183,0.1,0.171,0.308,0.031,0.02,0.023,0.012,0.017,0.021,0.199,0.157,0.382,1.468,1.148,0.341,0.045,1.488,8.48,11.945,0.048,0.05,0.027,0.031,0.017,0.553,1.172,2.88,3.558,4.594],"sid":"2f814d65-38fe-40fe-b25c-1b3d12d3d057","time":1568787658.7129}
get_data got {"pow":[0.027,0.018,0.029,0.038,0.025,0.045,0.028,0.06,0.209,0.054,0.145,0.042,0.116,0.313,0.06,0.083,0.027,0.066,0.158,0.023,0.285,0.851,2.962,6.269,4.754,0.191,0.077,0.105,0.237,0.047,0.099,0.068,0.101,0.169,0.028,0.067,0.085,0.353,2.166,0.08,0.169,0.106,0.184,0.304,0.04,0.02,0.019,0.011,0.02,0.02,0.2,0.179,0.428,1.876,1.394,0.331,0.055,1.511,8.61,11.693,0.05,0.05,0.022,0.03,0.018,0.608,1.395,2.877,3.478,4.402],"sid":"2f814d65-38fe-40fe-b25c-1b3d12d3d057","time":1568787658.8387}
get_data got {"pow":[0.024,0.019,0.033,0.041,0.029,0.052,0.039,0.059,0.251,0.05,0.135,0.036,0.118,0.386,0.064,0.091,0.032,0.055,0.139,0.023,0.276,0.817,2.627,5.942,4.56,0.2,0.099,0.116,0.238,0.045,0.107,0.09,0.089,0.17,0.026,0.06,0.07,0.334,2.17,0.079,0.15, [... cut ...] ef9eeebb5', 'recordIds': ['0e8325a3-8abd-4c89-a32e-29be8bb46334'], 'recording': False, 'started': '2019-09-18T14:20:58.341+08:00', 'status': 'closed', 'stopped': '2019-09-18T14:21:00.830+08:00', 'streams': []}}
E:\myTFileschx\EmotivTesting\cortex-v2-example-master\python\lib\cortex.py:173: RuntimeWarning: coroutine 'WebSocketCommonProtocol.close' was never awaited
  self.websocket.close()
RuntimeWarning: Enable tracemalloc to get the object allocation traceback

Process finished with exit code 0

nguoithichkhampha, 2019-09-18:
just subscribe stream "eeg". if your license has the permission. You should able to get eeg data.

chx2017, 2019-09-18:
I shall keep it secretly. Thank you for your kindly reminder.

chx2017, 2019-09-18:
Our license  has the permission till september 27. 2019. 
How to subscribe stream "eeg" in example.py?

nguoithichkhampha, 2019-09-18:
I don't say about expire time. I mean, scope of your license should eeg access. 
to get eeg, just change stream name in example "pow" -> "eeg". you will get eeg instead band power.

chx2017, 2019-09-19:
Thank you for your help!
Our license can access eeg. We have changed the stream name in example  "pow" -> "eeg".
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        print("** SUBSCRIBE eeg & MET **")
        await cortex.subscribe(['eeg', 'met'])
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
There are eighteen small contacting pads on the equipment(Emotiv EPOC+), and the data of fourteen channels(AF3,F7,F3,FC5,T7,P7 and etc) are shown on EMOTIVPRO.  Please you check the below data obtained using example.py, Where are the EEG data corresponding to all the channels ?
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
sent; awaiting response
** SUBSCRIBE eeg & MET **
lib.cortex resp:
{'id': 10, 'jsonrpc': '2.0', 'result': {'failure': [{'code': -32016, 'message': 'The stream is unavailable or unsupported.', 'streamName': 'eeg'}], 'success': [{'cols': ['eng', 'exc', 'lex', 'str', 'rel', 'int', 'foc'], 'sid': 'c93688c4-9b76-4cb3-9d56-96f58948a9ec', 'streamName': 'met'}]}}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855650.5686}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855660.6544}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855670.7489}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855680.8357}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855690.9221}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855701.0092}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855711.0955}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855721.1822}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855731.2691}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855741.3554}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660543,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855751.4425}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855761.5291}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660543,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855771.6158}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855781.7147}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855791.8049}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855801.8715}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855811.9627}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855822.0491}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855832.1515}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660543,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855842.2107}
Sending request:
{"jsonrpc": "2.0", "method": "injectMarker", "params": {"cortexToken": "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhcHBJZCI6ImNvbS4yMjU5YjJlZjllZWViYjUuamFja2Nhb18yMDE5ODMwIiwiYXBwVmVyc2lvbiI6IjEuMCIsImV4cCI6MTU2OTAyODQ0MiwibGljZW5zZUlkIjoiYWZiMzU5NGMtNTQ2OS00YTFmLWE3MjctNDQ5OTE2NjU4NGEyIiwibmJmIjoxNTY4NzY5MjQyLCJ1c2VySWQiOiIyOTkwZDlmNy1jMWYyLTQ1YWUtYmNmMy0zMjMzODExZDM5MzciLCJ1c2VybmFtZSI6IjIyNTliMmVmOWVlZWJiNSIsInZlcnNpb24iOiIyLjAifQ==.NNf0Xc9BhERe6FsL1K8+aimOBdFj3p3YCJz5BCCYhgc=", "session": "c93688c4-9b76-4cb3-9d56-96f58948a9ec", "label": "halfway", "value": 1, "port": "", "time": 1568855842333}, "id": 11}
sent; awaiting response
[... cut ...] ef9eeebb5', 'recordIds': ['2ceba5c1-cf97-4bef-a858-4151d6a7933a'], 'recording': False, 'started': '2019-09-19T09:14:03.360+08:00', 'status': 'closed', 'stopped': '2019-09-19T09:17:22.433+08:00', 'streams': []}}
E:\myTFileschx\EmotivTesting\cortex-v2-example-master\python\lib\cortex.py:173: RuntimeWarning: coroutine 'WebSocketCommonProtocol.close' was never awaited
  self.websocket.close()
RuntimeWarning: Enable tracemalloc to get the object allocation traceback

Process finished with exit code 0

chx2017, 2019-09-19:
can you send us the operating manual of the SDK(python or matlab)?

chx2017, 2019-09-19:
Thank you for your help!

Our license can access eeg. We have changed the stream name in example "pow" -> "eeg".

        print("** SUBSCRIBE eeg & MET **")
        await cortex.subscribe(['eeg', 'met'])

There are eighteen small contacting pads on the equipment(Emotiv EPOC+), and the data of fourteen channels(AF3,F7,F3,FC5,T7,P7 and etc) are shown on EMOTIVPRO. Please you check the below data obtained using example.py, Where are the EEG data corresponding to all the channels ? Can you send us the operating manual of the SDK (python or matlab)?
sent; awaiting response
** SUBSCRIBE eeg & MET **
lib.cortex resp:
{'id': 10, 'jsonrpc': '2.0', 'result': {'failure': [{'code': -32016, 'message': 'The stream is unavailable or unsupported.', 'streamName': 'eeg'}], 'success': [{'cols': ['eng', 'exc', 'lex', 'str', 'rel', 'int', 'foc'], 'sid': 'c93688c4-9b76-4cb3-9d56-96f58948a9ec', 'streamName': 'met'}]}}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855650.5686}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855660.6544}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855670.7489}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855680.8357}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855690.9221}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855701.0092}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855711.0955}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855721.1822}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855731.2691}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855741.3554}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660543,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855751.4425}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855761.5291}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660543,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855771.6158}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855781.7147}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855791.8049}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855801.8715}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855811.9627}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855822.0491}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660544,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855832.1515}
get_data got {"met":[0.560504520614219,0,0,1.1399918530430564e-12,0.30294071603459233,0.6513548646660543,0.08689552951575855],"sid":"c93688c4-9b76-4cb3-9d56-96f58948a9ec","time":1568855842.2107}
Sending request:
{"jsonrpc": "2.0", "method": "injectMarker", "params": {"cortexToken": "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhcHBJZCI6ImNvbS4yMjU5YjJlZjllZWViYjUuamFja2Nhb18yMDE5ODMwIiwiYXBwVmVyc2lvbiI6IjEuMCIsImV4cCI6MTU2OTAyODQ0MiwibGljZW5zZUlkIjoiYWZiMzU5NGMtNTQ2OS00YTFmLWE3MjctNDQ5OTE2NjU4NGEyIiwibmJmIjoxNTY4NzY5MjQyLCJ1c2VySWQiOiIyOTkwZDlmNy1jMWYyLTQ1YWUtYmNmMy0zMjMzODExZDM5MzciLCJ1c2VybmFtZSI6IjIyNTliMmVmOWVlZWJiNSIsInZlcnNpb24iOiIyLjAifQ==.NNf0Xc9BhERe6FsL1K8+aimOBdFj3p3YCJz5BCCYhgc=", "session": "c93688c4-9b76-4cb3-9d56-96f58948a9ec", "label": "halfway", "value": 1, "port": "", "time": 1568855842333}, "id": 11}
sent; awaiting response
lib.cortex resp:
[... cut ...] ecordIds': ['2ceba5c1-cf97-4bef-a858-4151d6a7933a'], 'recording': False, 'started': '2019-09-19T09:14:03.360+08:00', 'status': 'closed', 'stopped': '2019-09-19T09:17:22.433+08:00', 'streams': []}}
E:\myTFileschx\EmotivTesting\cortex-v2-example-master\python\lib\cortex.py:173: RuntimeWarning: coroutine 'WebSocketCommonProtocol.close' was never awaited
  self.websocket.close()
RuntimeWarning: Enable tracemalloc to get the object allocation traceback

Process finished with exit code 0

Best regards

nguoithichkhampha, 2019-09-23:
all the informations we write in the cortex docs.
from your log above. it see that your license does not have permission to get eeg data.

nguoithichkhampha, 2019-09-25:
@chx2017 , in API "authorize" you need to set param "license" to Cortex work with your license pro. Actually, you don't set this param, so our cloud system assign for you a basic license without permission with eeg.
https://emotiv.gitbook.io/cortex-api/authentication/authorize

bchaperon, 2019-09-26:
@chx2017 The Cortex API documentation is available at https://emotiv.gitbook.io/cortex-api/
Please note that there is no manual specific to Python or Matlab.
You access the Cortex API via a web socket, and you can do that with any programming language.

chx2017, 2019-10-09:
**We have tried it again.
In the file(example.py), the 'pow' have been changed to ‘eeg’. Some data have been obtained as shown below(get_data got{"eeg":..........}) . 
_Do you confirm that those data in "get_data got{"eeg":..........}" are the raw eeg data?_
Thank you for your big help!**
—————————————————————
#print("** SUBSCRIBE POW & MET **")
#await cortex.subscribe(['pow', 'met'])

print("** SUBSCRIBE eeg & MET **")
await cortex.subscribe(['eeg', 'met'])
—————————————————————
** SUBSCRIBE eeg & MET **
lib.cortex resp:
{'id': 10, 'jsonrpc': '2.0', 'result': {'failure': [], 'success': [{'cols': ['COUNTER', 'INTERPOLATED', 'AF3', 'F7', 'F3', 'FC5', 'T7', 'P7', 'O1', 'O2', 'P8', 'T8', 'FC6', 'F4', 'F8', 'AF4', 'RAW_CQ', 'MARKER_HARDWARE', 'MARKERS'], 'sid': '3d8ddffc-1c35-465e-aed1-0e341c53c594', 'streamName': 'eeg'}, {'cols': ['eng', 'exc', 'lex', 'str', 'rel', 'int', 'foc'], 'sid': '3d8ddffc-1c35-465e-aed1-0e341c53c594', 'streamName': 'met'}]}}
get_data got {"eeg":[219,0,4176.41,4223.462,4175.128,4182.692,4215.256,4214.872,4204.487,4216.41,4216.41,4223.462,4277.436,5023.333,4247.692,4241.667,0,0,[]],"sid":"3d8ddffc-1c35-465e-aed1-0e341c53c594","time":1570583903.9613}
get_data got {"eeg":[220,0,4177.308,4223.846,4163.59,4183.59,4212.436,4215,4206.667,4217.051,4216.923,4224.359,4280,5032.564,4247.308,4246.667,0,0,[]],"sid":"3d8ddffc-1c35-465e-aed1-0e341c53c594","time":1570583903.9653}
get_data got {"eeg":[221,0,4196.667,4224.103,4164.744,4195.385,4216.538,4218.205,4216.154,4220.385,4219.103,4225.128,4277.179,5031.923,4230.769,4229.872,0,0,[]],"sid":"3d8ddffc-1c35-465e-aed1-0e341c53c594","time":1570583903.9692}
get_data got {"eeg":[222,0,4219.231,4222.436,4166.282,4207.564,4220.513,4218.846,4221.026,4222.821,4220.513,4225.128,4275.256,5011.923,4217.949,4214.872,0,0,[]],"sid":"3d8ddffc-1c35-465e-aed1-0e341c53c594","time":1570583903.9731}
get_data got {"eeg":[223,0,4229.615,4218.59,4164.103,4212.564,4218.974,4213.846,4215.256,4221.923,4219.359,4224.231,4279.487,4972.308,4223.462,4223.077,0,0,[]],"sid":"3d8ddffc-1c35-465e-aed1-0e341c53c594","time":1570583903.977}
get_data got {"eeg":[224,0,4228.462,4214.744,4171.154,4214.744,4214.872,4207.179,4206.41,4219.231,4217.436,4223.333,4286.026,4925.769,4238.462,4244.487,0,0,[]],"sid":"3d8ddffc-1c35-465e-aed1-0e341c53c594","time":1570583903.9809}
get_data got {"eeg":[225,0,4228.333,4213.077,4200.128,4222.692,4214.231,4205.256,4205.769,4218.205,4217.436,4223.077,4286.923,4890.641,4241.282,4250.769,0,0,[]],"sid":"3d8ddffc-1c35-465e-aed1-0e341c53c594","time":1570583903.9849}
get_data got {"eeg":[226,0,4235,4213.59,4240.641,4235.513,4217.692,4209.231,4213.462,4219.872,4220.641,4223.333,4280.897,4878.718,4225.769,4232.949,0,0,[]],"sid":"3d8ddffc-1c35-465e-aed1-0e341c53c594","time":1570583903.9888}
get_data got {"eeg":[227,0,4240.897,4214.744,4265.128,4242.564,4219.359,4213.333,4218.846,4221.923,4225.128,4223.846,4275.513,4888.205,4207.564,4213.333,0,0,[]],"sid":"3d8ddffc-1c35-465e-aed1-0e341c53c594","time":1570583903.9926}
get_data got {"eeg":[228,0,4236.923,4215.256,4258.462,4237.949,4215.641,4212.692,4215,4222.436,4228.462,4224.359,4277.821,4909.615,4204.615,4215.256,0,0,[]],"sid":"3d8ddffc-1c35-465e-aed1-0e341c53c594","time":1570583903.9966}
Sending request:
{"jsonrpc": "2.0", "method": "injectMarker", "params": {"cortexToken": "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhcHBJZCI6ImNvbS4yMjU5YjJlZjllZWViYjUuamFja2Nhb18yMDE5ODMwIiwiYXBwVmVyc2lvbiI6IjEuMCIsImV4cCI6MTU3MDc1NjcwMywibGljZW5zZUlkIjoiMjIwMWM2NmYtMWM1My00N2JlLWJlNDAtYzliOTVjZDM0NjJkIiwibmJmIjoxNTcwNDk3NTAzLCJ1c2VySWQiOiIyOTkwZDlmNy1jMWYyLTQ1YWUtYmNmMy0zMjMzODExZDM5MzciLCJ1c2VybmFtZSI6IjIyNTliMmVmOWVlZWJiNSIsInZlcnNpb24iOiIyLjAifQ==.vGdiBWbuk/foc1T3v/I30Qhp/5g8E6JGTAx+VdUmoVQ=", "session": "3d8ddffc-1c35-465e-aed1-0e341c53c594", "label": "halfway", "value": 1, "port": "", "time": 1570583904189}, "id": 11}
sent; awaiting response
lib.cortex resp:
{'eeg': [229, 0, 4227.179, 4214.872, 4236.795, 4228.974, 4210.128, 4208.974, 4207.436, 4222.179, 4229.487, 4225, 4284.744, 4935, 4214.487, 4233.077, 0, 0, []], 'sid': '3d8ddffc-1c35-465e-aed1-0e341c53c594', 'time': 1570583904.0005}
Sending request:
{"jsonrpc": "2.0", "method": "updateSession", "params": {"cortexToken": "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhcHBJZCI6ImNvbS4yMjU5YjJlZjllZWViYjUuamFja2Nhb18yMDE5ODMwIiwiYXBwVmVyc2lvbiI6IjEuMCIsImV4cCI6MTU3MDc1NjcwMywibGljZW5zZUlkIjoiMjIwMWM2NmYtMWM1My00N2JlLWJlNDAtYzliOTVjZDM0NjJkIiwibmJmIjoxNTcwNDk3NTAzLCJ1c2VySWQiOiIyOTkwZDlmNy1jMWYyLTQ1YWUtYmNmMy0zMjMzODExZDM5MzciLCJ1c2VybmFtZSI6IjIyNTliMmVmOWVlZWJiNSIsInZlcnNpb24iOiIyLjAifQ==.vGdiBWbuk/foc1T3v/I30Qhp/5g8E6JGTAx+VdUmoVQ=", "session": "3d8ddffc-1c35-465e-aed1-0e341c53c594", "status": "close"}, "id": 12}
sent; awaiting response
lib.cortex resp:
{'id': 12, 'jsonrpc': '2.0', 'result': {'appId': 'com.2259b2ef9eeebb5.jackcao_2019830', 'headset': {'connectedBy': 'dongle', 'customName': '', 'dongle': '6ff', 'firmware': '625', 'id': 'EPOCPLUS-3B9AEBCD', 'motionSensors': ['GYROX', 'GYROY', 'GYROZ', 'ACCX', 'ACCY', 'ACCZ', 'MAGX', 'MAGY', 'MAGZ'], 'sensors': ['AF3', 'F7', 'F3', 'FC5', 'T7', 'P7', 'O1', 'O2', 'P8', 'T8', 'FC6', 'F4', 'F8', 'AF4'], 'settings': {'eegRate': 256, 'eegRes': 16, 'memsRate': 0, 'memsRes': 16, 'mode': 'EPOCPLUS'}, 'status': 'connected'}, 'id': '3d8ddffc-1c35-465e-aed1-0e341c53c594', 'license': '2201c66f-1c53-47be-be40-c9b95cd3462d', 'owner': '2259b2ef9eeebb5', 'recordIds': ['1a525a4f-aec6-47bf-b493-e60ce1133536'], 'recording': False, 'started': '2019-10-09T09:18:24.134+08:00', 'status': 'closed', 'stopped': '2019-10-09T09:18:24.216+08:00', 'streams': []}}
E:\myTFileschx\EmotivTesting\cortex-v2-example-master\python\lib\cortex.py:173: RuntimeWarning: coroutine 'WebSocketCommonProtocol.close' was never awaited
self.websocket.close()
RuntimeWarning: Enable tracemalloc to get the object allocation traceback

Process finished with exit code 0

bchaperon, 2019-10-09:
Hi @chx2017 
Yes, it is the raw EEG data.
You can check the doc to see how to interpret these values.
https://emotiv.gitbook.io/cortex-api/data-subscription/data-sample-object

chx2017, 2019-10-09:
Thank you for your big help!

chx2017, 2019-10-09:
How can the example.py only export the raw eeg data continually? 
How to modify the example.py?
Or  other python files can do this?

chx2017, 2019-10-12:
What is the unit of these values?
________________________________________________________
get_data got {"eeg":[219,0,4176.41,4223.462,4175.128,4182.692,4215.256,4214.872,4204.487,4216.41,4216.41,4223.462,4277.436,5023.333,4247.692,4241.667,0,0,[]],"sid":"3d8ddffc-1c35-465e-aed1-0e341c53c594","time":1570583903.9613}
get_data got {"eeg":[220,0,4177.308,4223.846,4163.59,4183.59,4212.436,4215,4206.667,4217.051,4216.923,4224.359,4280,5032.564,4247.308,4246.667,0,0,[]],"sid":"3d8ddffc-1c35-465e-aed1-0e341c53c594","time":1570583903.9653}
get_data got {"eeg":[221,0,4196.667,4224.103,4164.744,4195.385,4216.538,4218.205,4216.154,4220.385,4219.103,4225.128,4277.179,5031.923,4230.769,4229.872,0,0,[]],"sid":"3d8ddffc-1c35-465e-aed1-0e341c53c594","time":1570583903.9692}

nguoithichkhampha, 2019-10-12:
it's μV value, we call it raw eeg.

closed 2019-10-14 by nguoithichkhampha