Emotiv/cortex-example #43: Cortex token and Session Creation

brainsensors, 2019-09-23:
Hi i am working with Emotiv Flex. While calling EEgLogger. exe simultaneously with another modality it starts with authorization everytime in the same application, due to which I am not able to fix both modalities timings. Also whenever I start EEGLogger It's time starts decreasing after ever run and than becomes constant.
How to save the cortex token and change the create session files for solving the above problem.

bchaperon, 2019-09-26:
Hello @brainsensors 
Sorry, I don't understand your issue.
What do you mean by "another modality" and "modalities timings"?

brainsensors, 2019-09-26:
"Another modality" here is other sensor device. I am working with two different sensor devices and combined it into one custom application.  On button click both sensor .exe run at one time but EEGLogger takes time first to authorize and then start with EEG data collection. I want  two sensors should start collecting data at one time ( "minus authorization"). otherwise there is mismatch of timings between two sensor devices data.

bchaperon, 2019-09-27:
So what you are trying to do is to synchronize the EEG samples from 2 different headsets, is that right?
In that case you should use the timestamp of the EEG samples to synchronize them.
Each EEG sample has a number like "1569559928.2512" that is an Unix epoch time.
So you can know the date and time of each sample, with a precision of less than a millisecond.
https://en.wikipedia.org/wiki/Unix_time

brainsensors, 2019-09-27:
No, its a EMG sensor. I want to start both the sensors at the same time. On our interface when I call EEG-logger, it takes some time for authorization due to which we cannot get synchronized data from both EEG and EMG.

bchaperon, 2019-10-10:
I have no experience with EMG device so I cannot help you with this issue.
Please contact the [Emotiv customer support](https://www.emotiv.com/contact-support/), they should be able to help you.
Sorry for the late answer.

closed 2019-10-21 by bchaperon