Emotiv/cortex-example #44: lib.cortex.CortexApiException: {"error":{"code":-32019,"message":"Session limit on this device has been reached."},"id":8,"jsonrpc":"2.0"}

PsychoHJ, 2019-09-25:
Got error in createSession with params {'cortexToken': 'eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhcHBJZCI6ImNvbS5oX2Fsamlib3J5LmhqYXBwIiwiYXBwVmVyc2lvbiI6IjEuMCIsImV4cCI6MTU2OTU3OTMxMywibGljZW5zZUlkIjoiMzEwMDk0ZDItYjUzNi00YzJjLWJhNzYtM2Q3YWI2NTk2YmEyIiwibmJmIjoxNTY5MzIwMTEzLCJ1c2VySWQiOiI4MGE4ZTM1NS1hMWQ1LTRhMGMtYjIxYy0yMjVjN2Q2ZTU0OTciLCJ1c2VybmFtZSI6ImhfYWxqaWJvcnkiLCJ2ZXJzaW9uIjoiMi4wIn0=.DxnhnzUS4R7k+35ar3qlF6/+QJ6NUO+gP1af7uFuX1A=', 'headset': 'EPOCPLUS-4A2C045C', 'status': 'active'}:
{"error":{"code":-32019,"message":"Session limit on this device has been reached."},"id":8,"jsonrpc":"2.0"}
Traceback (most recent call last):
  File "C:\Users\Hammody\Desktop\BCI\pyemotivcortexv2\demo.py", line 53, in <module>
    test()
  File "C:\Users\Hammody\Desktop\BCI\pyemotivcortexv2\demo.py", line 44, in test
    asyncio.run(do_stuff(cortex))
  File "C:\Users\Hammody\AppData\Local\Programs\Python\Python37-32\lib\asyncio\runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "C:\Users\Hammody\AppData\Local\Programs\Python\Python37-32\lib\asyncio\base_events.py", line 579, in run_until_complete
    return future.result()
  File "C:\Users\Hammody\Desktop\BCI\pyemotivcortexv2\demo.py", line 25, in do_stuff
    headset_id=cortex.headsets[0])
  File "C:\Users\Hammody\Desktop\BCI\pyemotivcortexv2\lib\cortex.py", line 258, in create_session
    resp = await self.send_command('createSession', **params)
  File "C:\Users\Hammody\Desktop\BCI\pyemotivcortexv2\lib\cortex.py", line 133, in send_command
    raise CortexApiException(resp)
lib.cortex.CortexApiException: {"error":{"code":-32019,"message":"Session limit on this device has been reached."},"id":8,"jsonrpc":"2.0"}

NguyenDuyToan, 2019-09-27:
Possibly duplicated with  #42

PsychoHJ, 2019-09-27:
#42 sol. didn't work with me

closed 2019-11-19 by nguoithichkhampha