Emotiv/cortex-example #45: Error -32019 message Session limit on this device has been reached.

temolzin, 2019-09-28:
Estoy programando con C#
con los ejemplos que vienen en CORTEXUI2
Pero no he podido resolver el error Received: Session limit on this device has been reached.
MessageErrorRecieved :code -32019 message Session limit on this device has been reached.
En el archivo config no tengo nada en LicenseId,
 en DebitNumber puse el valor 0.

Y no he podido resolver el error.

shung1209, 2019-10-02:
I get  **-32019 message Session limit on this device has been reached** on every examples even i edited the debit to 100 even to 200. This is really getting frustrating cause there is not really a clear document how to deal with this.

nguoithichkhampha, 2019-10-02:
@shung1209 Can you send me request/response to/from Cortex ?

shung1209, 2019-10-02:
this is what is shown in the command.

EEG LOGGER
Please wear Headset with good signal!!!
Send getUserLogin
Press Esc to exit
handleResponse: getUserLogin
Send hasAccessRight
handleResponse: hasAccessRight
Send authorize
handleResponse: authorize
Authorize successfully.
FindHeadset
Send queryHeadsets
handleResponse: queryHeadsets
Send createSession
Received: Session limit on this device has been reached.
MessageErrorRecieved :code -32019 message Session limit on this device has been reached.

nguoithichkhampha, 2019-10-02:
I meant the details of request/response to Cortex. I want to see the input params.

shung1209, 2019-10-02:
May i ask where do i get that? I am so confused.

nguoithichkhampha, 2019-10-02:
https://github.com/Emotiv/cortex-v2-example/blob/master/csharp/CortexAccess/CtxClient.cs
this class uncomment line 164 and 177

temolzin, 2019-10-03:
I uncomment line 164 and 177

This is my screen.

[cid:71051cd5-9212-4d06-86f8-ead4abfc777f]

https://github.com/Emotiv/cortex-v2-example/blob/master/csharp/CortexAccess/CtxClient.cs

shung1209, 2019-10-03:
[image]

Finally found on of the problems, Go to CtxClient.cs line 540, set param.Add("debit", 0); from 0 to 1

closed 2019-10-14 by nguoithichkhampha