Emotiv/cortex-example #47: Can I connect to the Cortex API using Android Studio?

kitsune127, 2019-10-14:
I am interested to create an Android application to interact with a EMOTIV Insight, but I am wondering is it possible to connect to Cortex API using Android Studio?

nguoithichkhampha, 2019-10-14:
@kitsune127 , this is good idea. Actually, cortex only run on windows and mac, you can run client on android device to connect with cortex by url Ex: "wss://192.168.1.30:6868". But you should config ignore ssl error because our certificate does not signed for IP lan.
on other side, we are developing cortex android version. We will publish it next year.

closed 2019-10-24 by nguoithichkhampha

cstayyab, 2020-01-30:
@kitsune127  If you have made any progress can you please share your work?

cstayyab, 2020-01-30:
@nguoithichkhampha Any idea if you are publishing it in near future? first quarter of the year or after that?

nguoithichkhampha, 2020-01-30:
@cstayyab we are in testing step, but I can not confirm with you exactly release time.

KhangSoCute, 2020-05-07:
i don't know how to add Cortex API websocket into dependencies in Android. Do you have any idea ?