Emotiv/cortex-example #51: What could be the reason why my training is failing? I have the request and the MC failed response below.

nka292, 2019-11-08:
______________REQUEST___________________

#neutral
ws.send(json.dumps({
    "id": 1,
    "jsonrpc": "2.0",
    "method": "training",
    "params": {
        "action": "neutral",
        "cortexToken": cortexToken,
        "detection": "mentalCommand",
        "session": sessionId,
        "status": "start"
    }
   
}))

print("_____1____",ws.recv()+"\n")
time.sleep(5)
print(ws.recv())
time.sleep(10)
print(ws.recv())
print("Start training for neutral ---> ",ws.recv()+"\n")

ws.send(json.dumps({
    "id": 1,
    "jsonrpc": "2.0",
    "method": "training",
    "params": {
        "action": "neutral",
        "cortexToken": cortexToken,
        "detection": "mentalCommand",
        "session": sessionId,
        "status": "accept"
    }
}))
print("Accept Training for neutral---> ",ws.recv()+"\n")
print(ws.recv())
time.sleep(2)
print(ws.recv())

________________RESPONSE_____________________

***************** START TRAINING **********************
_____1____ {"id":1,"jsonrpc":"2.0","result":{"action":"neutral","message":"Set up training successfully","status":"start"}}
{"sid":"1b467357-39c8-4c3f-bf8d-d701968414ac","sys":["mentalCommand","MC_Started"],"time":1573177873.6001072}
{"sid":"1b467357-39c8-4c3f-bf8d-d701968414ac","sys":["mentalCommand","MC_Failed"],"time":1573177883.698816}

nguoithichkhampha, 2019-11-08:
@nka292 , probably relate to contact quality, you should make sure you have good contact quality before start training.

nka292, 2019-11-08:
> @nka292 , probably relate to contact quality, you should make sure you have good contact quality before start training.

I had 80% while I ran my script

nguoithichkhampha, 2019-11-19:
what is your headset ? you connect by usb dongle or bluetooth ?

nka292, 2019-11-19:
Usb dongle

Best Regards,
Nana Asubonteng

bchaperon, 2019-11-20:
Hi @nka292 
Can you try to do the training with EmotivBCI first?
If it works in BCI then it should also work in your application.

closed 2019-12-10 by bchaperon