Emotiv/cortex-example #52: Access Sqlite locally

michelpf, 2019-11-14:
The data folder, example, '''ProgramData''', has some Sqlite databases. In order to best get the data we tried to access that database, but it has a password.

It's OK to access this database locally? In case of yes, what password we should use in order to access the data?

Thank you.

nguoithichkhampha, 2019-11-15:
@michelpf , you can not access them.

bchaperon, 2019-11-20:
Hi @michelpf 
Just to clarify, this database is an implementation detail of Cortex. Its data structure can change at any time, or maybe it will even be removed in a future version of Cortex.
The only way to communicate with Cortex is to use the Web Socket API.

May I ask why you believe it would be easier to access to database than to use the API?
Did you encounter any limitation or issue using the API?

michelpf, 2019-11-20:
Hi @bchaperon, that's all right. The API is great, especially the last major update.

I was studying ways to collect the data independently of the internet connection, but, as I can see, it is impossible considering the actual architecture. Regardless the data is out from our Emotives they travel to your datacenters (US) and then come back to us (Brazil).

Do you have plans to release data locally?

bchaperon, 2019-11-21:
There is a misunderstanding, you can actually collect data when you are offline.
Cortex can stay offline for several days or even weeks.
The exact duration depends on what is permitted by your license.

So you can use EmotivPRO when you are offline.
https://emotiv.gitbook.io/emotivpro/subscriptions/offline_use

You can also use your own application. First, you must be online and get a Cortex token.
https://emotiv.gitbook.io/cortex-api/authentication/authorize
Then you can store the token on your computer and use it when you are offline.
If you want to stay offline more then 2 days then you need to periodically renew the token.
https://emotiv.gitbook.io/cortex-api/authentication/generatenewtoken

But in any case, you cannot stay offline forever. At some point, Cortex will need to connect to the Emotiv cloud to check your license again.

michelpf, 2019-11-21:
Great @bchaperon. I'm not aware about it.

Now, will fit perfectly in our use case.

Thank you so much!

closed 2019-11-21 by michelpf