Emotiv/cortex-example #53: Using Emotiv Insight and Arduino offline

pattyreb, 2019-11-15:
I'm currently working on a project that involves the Emotiv insight to communicate data directly to an arduino through bluetooth. So, this would not be using the websocket-client that was used in Cortex API. 
I was wondering if anyone has done this successfully and wouldn't mind explaining the basics of connecting the two devices and send information from the Emotiv device to the arduino through bluetooth,

bchaperon, 2019-11-20:
Hello @pattyreb 
This is not possible. The data sent on the bluetooth connection is encrypted, so you won't be able to read it.

hoangphamemotiv, 2019-11-20:
@pattyreb how about if you get data stream from websocket-client, convert to OSC and send it via a port which can be connected by Adruino-esp8266 via wifi?

bchaperon, 2019-11-21:
@pattyreb Can you give us more information about what you are trying to do?
Maybe we can help you find an alternative solution.

closed 2019-12-04 by bchaperon