Emotiv/cortex-example #54: Creating and Loading the profiles

Aniruddha008, 2019-11-17:
Hello all,
I am having trouble with creating a profile via code in python. I tried creating profile in the EmotivBci and then loading it in the project, but it keeps giving me the error of missing parameters or invalid parameters.
I have followed the documentation of cortex 2.0 completely.
def setup_profile(self,name):
    create_request = {
    "id": 1,
    "jsonrpc": "2.0",
    "method": "setupProfile",
    "params": {
        "cortexToken": self.auth,
        "profile": name,
        'status': 'create'
    }}
    self.ws.send(json.dumps(create_request))
    result = self.ws.recv()
    result_dic = json.loads(result)
    print('setting up profile', json.dumps(result_dic, indent=4))

def load_profile(self):
    load_request = {
    "id": 1,
    "jsonrpc": "2.0",
    "method": "setupProfile",
    "params": {
        "cortexToken": self.auth,
        'headset_id' : self.headset_id,
        "profile": "Aniruddha",
        'status' : 'load'
    }}
    self.ws.send(json.dumps(load_request))
    result = self.ws.recv()
    result_dic = json.loads(result)
    print('loading up profile', json.dumps(result_dic, indent=4))

I am using this in a GUI application. I am able to subscribe to 'com' for mental commands but getting stuck with loading the profile even though I have trained the profile in the EmotivBCI.
I would really appreciate it if someone could pinpoint my mistake here.

nguoithichkhampha, 2019-11-19:
@Aniruddha008 , when call "load profile". you need to make sure your headset is connected and set correct headset serial in the param "headset"

Aniruddha008, 2019-11-20:
> 
> 
> @Aniruddha008 , when call "load profile". you need to make sure your headset is connected and set correct headset serial in the param "headset"

yes, I realized it later, that I was using 'headset_id' instead of 'headset'. Thank you so much.

closed 2019-11-20 by Aniruddha008