Emotiv/cortex-example #55: EEG stream is unavailable or unsupported

mvidaldp, 2019-11-19:
I am trying to access the raw EEG data via Cortex v2 API using the example code from this repo. I made sure I have a pro license and I was granted access to the raw EEG data. Nevertheless I keep getting messages saying that I have no access to the eeg data stream.
I logged out and logged in of my account on the Emotiv App. I even uninstalled the emotiv apps, removed any emotiv data folders left on the my computer and installed it again. Same result.
I also tried to remove the current app, create and authorize a new one with new client_id and client_secret codes, but same result.

I copied and pasted the debugging messages from the python example code here. I tried it with the CSharp code and I get the same result. I removed the credentials, user-related ids, tokens and so on by "foo":

** USER LOGIN **
Sending request:
{"jsonrpc": "2.0", "method": "getUserLogin", "params": {}, "id": 1}
sent; awaiting response
lib.cortex resp:
{'currentOSUId': '1002', 'currentOSUsername': 'foo', 'loggedInOSUId': '1002', 'loggedInOSUsername': 'foo', 'username': 'foo'}
** GET CORTEX INFO **
Sending request:
{"jsonrpc": "2.0", "method": "getCortexInfo", "params": {}, "id": 2}
sent; awaiting response
lib.cortex resp:
{'id': 2, 'jsonrpc': '2.0', 'result': {'buildDate': '2019-11-12T09:47:41', 'buildNumber': 'v2.2.1-10-ge7781ad', 'version': '2.2.3'}}
** HAS ACCESS RIGHT **
Sending request:
{"jsonrpc": "2.0", "method": "requestAccess", "params": {"clientId": "foo", "clientSecret": "foo"}, "id": 3}
sent; awaiting response
lib.cortex resp:
{'id': 3, 'jsonrpc': '2.0', 'result': {'accessGranted': True, 'message': 'The user has granted access right to this application.'}}
** REQUEST ACCESS **
Sending request:
{"jsonrpc": "2.0", "method": "requestAccess", "params": {"clientId": "foo", "clientSecret": "foo"}, "id": 4}
sent; awaiting response
lib.cortex resp:
{'id': 4, 'jsonrpc': '2.0', 'result': {'accessGranted': True, 'message': 'The user has granted access right to this application.'}}
** AUTHORIZE **
Sending request:
{"jsonrpc": "2.0", "method": "authorize", "params": {"clientId": "foo", "clientSecret": "foo", "debit": 100}, "id": 5}
sent; awaiting response
lib.cortex resp:
{'id': 5, 'jsonrpc': '2.0', 'result': {'cortexToken': 'foo'}}
** GET LICENSE INFO **
Sending request:
{"jsonrpc": "2.0", "method": "getLicenseInfo", "params": {"cortexToken": "foo"}, "id": 6}
sent; awaiting response
lib.cortex resp:
{'id': 6, 'jsonrpc': '2.0', 'result': {'isOnline': True, 'license': {'applications': ['com.emotiv.sdk.basic'], 'billingFrom': '2019-10-29T01:00:00.000+01:00', 'billingTo': '2029-10-26T02:00:00.000+02:00', 'deviceInfo': {'deviceLimit': 3, 'devicesPerSeat': 3, 'sessionLimit': {'day': None, 'month': None, 'year': None}}, 'expired': False, 'extenderLimit': 0, 'hardLimitTime': '2019-12-07T00:59:59.999+01:00', 'isCommercial': False, 'licenseId': 'foo', 'licenseName': 'Basic License', 'localQuota': 218, 'maxDebit': None, 'scopes': [''], 'seatCount': 1, 'sessionCount': 10, 'softLimitTime': 
'2019-11-30T00:59:59.999+01:00', 'totalDebit': 3130, 'totalRegisteredDevices': 1, 'validFrom': '2019-10-29T01:00:00.000+01:00', 'validTo': '2029-10-27T01:59:59.999+02:00'}}}
** QUERY HEADSETS **
Sending request:
{"jsonrpc": "2.0", "method": "queryHeadsets", "params": {}, "id": 7}
sent; awaiting response
lib.cortex found headsets ['EPOC-60874887']
lib.cortex resp:
{'id': 7, 'jsonrpc': '2.0', 'result': [{'connectedBy': 'dongle', 'customName': '', 'dongle': '565', 'firmware': '565', 'id': 'EPOC-60874887', 'motionSensors': ['GYROX', 'GYROY'], 'sensors': ['AF3', 'F7', 'F3', 'FC5', 'T7', 'P7', 'O1', 'O2', 'P8', 'T8', 'FC6', 'F4', 'F8', 'AF4'], 'settings': {'eegRate': 128, 'eegRes': 14, 'memsRate': 128, 'memsRes': 14, 'mode': 'STANDARD'}, 'status': 'connected'}]}
** CREATE SESSION **
Sending request:
{"jsonrpc": "2.0", "method": "createSession", "params": {"cortexToken": "foo", "headset": "EPOC-60874887", "status": "active"}, "id": 8}
sent; awaiting response
lib.cortex resp:
{'id': 8, 'jsonrpc': '2.0', 'result': {'appId': 'com.mvidaldepalo.rawdata', 'headset': {'connectedBy': 'dongle', 'customName': '', 'dongle': '565', 'firmware': '565', 'id': 'EPOC-60874887', 'motionSensors': ['GYROX', 'GYROY'], 'sensors': ['AF3', 'F7', 'F3', 'FC5', 'T7', 'P7', 'O1', 'O2', 'P8', 'T8', 'FC6', 'F4', 'F8', 'AF4'], 'settings': {'eegRate': 128, 'eegRes': 14, 'memsRate': 128, 'memsRes': 14, 'mode': 'STANDARD'}, 'status': 'connected'}, 'id': 'd2d02ae5-490a-485c-b1b1-2e48de2ddb86', 'license': '6cba759c-a9fc-48af-9dd8-0b3220d9dfa7', 'owner': 'mvidaldepalo', 'recordIds': [], 'recording': False, 'started': '2019-11-19T12:13:30.136+01:00', 'status': 'activated', 'stopped': '', 'streams': []}}
** CREATE RECORD **
Sending request:
{"jsonrpc": "2.0", "method": "createRecord", "params": {"cortexToken": "foo", "session": "d2d02ae5-490a-485c-b1b1-2e48de2ddb86", "title": "test record 1"}, "id": 9}
sent; awaiting response
lib.cortex resp:
{'id': 9, 'jsonrpc': '2.0', 'result': {'record': {'applicationId': 'com.mvidaldepalo.rawdata', 'applicationVersion': '1.0', 'description': '', 'endDatetime': '', 'experimentId': 0, 'licenseId': 'foo', 'licenseScope': [''], 'localOnly': False, 'ownerId': 'foo', 'startDatetime': '2019-11-19T12:13:30.058710+01:00', 'tags': [], 'title': 'test record 1', 'uuid': '63b7d15c-cbda-4e6e-a834-ebdd8cfdc95e'}, 'sessionId': 'foo'}}
** SUBSCRIBE POW & MET **
Sending request:
{"jsonrpc": "2.0", "method": "subscribe", "params": {"cortexToken": "foo", "session": "d2d02ae5-490a-485c-b1b1-2e48de2ddb86", "streams": ["eeg"]}, "id": 10}
sent; awaiting response
lib.cortex resp:
{'id': 10, 'jsonrpc': '2.0', 'result': {'failure': [{'code': -32016, 'message': 'The stream is unavailable or unsupported.', 'streamName': 'eeg'}], 'success': []}}

nguoithichkhampha, 2019-11-19:
@mvidaldp , in the request "authorize" you don't set param "license". So, by default you are using basic license. you can see in the response of request "getLicenseInfo"
 `'licenseName': 'Basic License'`

mvidaldp, 2019-11-20:
> 
> 
> @mvidaldp , in the request "authorize" you don't set param "license". So, by default you are using basic license. you can see in the response of request "getLicenseInfo"
> `'licenseName': 'Basic License'`

Actually the parameter is optional and expects my license id. By trying I realized that the key 'license_id' on params (line 202) was wrong. By changing it to license I can use this parameter but I am getting the same error as below. I am afraid there is something not properly set internally about my license.

bchaperon, 2019-11-20:
Hi @mvidaldp 
Can you see the EEG data in EmotivPRO? If yes, then your license is probably OK.
Please note that to use your PRO license in your app you need to fill an application form on the Emotiv web site.
https://emotiv.gitbook.io/cortex-api/#license

mvidaldp, 2019-11-20:
> 
> 
> Hi @mvidaldp
> Can you see the EEG data in EmotivPRO? If yes, then your license is probably OK.
> Please note that to use your PRO license in your app you need to fill an application form on the Emotiv web site.
> https://emotiv.gitbook.io/cortex-api/#license

Hi @bchaperon 
Yes, I can see it and also export it into a file once a session is recorded on EmotivPRO. I also filled out the application and my access was confirmed per email. However, I realized that on my EmotivApp my license information says "EmotivPRO Student" and "Basic License". Also, if I check my subscription on the site, on the status it says "pending".

bchaperon, 2019-11-21:
Ok, so maybe there is a problem with your license after all.
Please contact the Emotiv customer support, they can check you subscription.
https://www.emotiv.com/contact-support/

mvidaldp, 2019-11-22:
@bchaperon @nguoithichkhampha 
It is already fixed. I was using the wrong license_id, since when you upgrade your basic license to pro you get a new id. Thanks for your support :)

closed 2019-11-22 by mvidaldp

haidergb, 2022-01-09:
> 

> Hi @mvidaldp Can you see the EEG data in EmotivPRO? If yes, then your license is probably OK. Please note that to use your PRO license in your app you need to fill an application form on the Emotiv web site. https://emotiv.gitbook.io/cortex-api/#license

Hi @bchaperon 
Do I need to fill the application each time I purchase a new PRO license?

bchaperon, 2022-01-11:
Hi @haidergb 
Yes, you do :)