Emotiv/cortex-example #59: Error creating cortex_creds

SiwenW, 2019-11-24:
Hello, I followed the readme file to run the python example, and I finished the "before you start section" (at least I think I did...). But I couldn't create the credentials file, got some error from my command prompt.
Could someone please help me with this?
[image]

bchaperon, 2019-11-25:
What OS are you using?
If the command "cat" doesn't work then you can just use a text editor to create the file "cortex_creds". This file is just a normal text file that contains your client id and client secret. Its content should look like this:

> client_id 123456789abcdefghijklmnopqrstuvwxyzABCDE
> client_secret 123456789abcdefghijklmnopqrstuvwxyzABCDExxxxxxxxxxxxxxx

SiwenW, 2019-11-25:
Hello Benoit, 

thanks for your help. I'm using Windows. 

I created a txt file named "cortex_creds" as you suggested and placed under the same folder as example.py. However, the code does not seem to be able to find the file. I tried read the original code and see if I need to modify anything, but I didn't really understand it.

Is there anything I need to modify in the code or do I need to place the file in a certain location?

Thanks for your time and help.
[image]

[image]

closed 2019-11-26 by SiwenW