Emotiv/cortex-example #62: How to get all data of streams that I subscribed to in case of calling cortex.get_data once?

fosaken, 2019-12-03:
Hello!

I want to get eeg data and dev data at same time so I subscribe to stream_list:['eeg', 'dev'], I got the following logging info after I run example.py.  I called cortex.get_data() once per 0.2s, but I got the eeg and dev data irregularly.
`get_data got {"eeg":[61,0,0.007,0.383,-3.775,0.169,0.671,-2.068,0.643,0.642,0.535,-0.636,0.218,0.412,3.971,0.223,-0.325,-0.021,-0.446,0.024,0.478,0.388,-3.604,-0.375,0.036,0.828,1.65,0.712,0.326,-0.554,-1.979,-4.646,-2.841,0.095,8064.0,0.0,0,[]],"sid":"98bc0ab6-7b56-4235-b336-f1e020a20330","time":1575369399.2567}
get_data got {"eeg":[62,0,0.007,0.383,-4.285,0.169,0.671,-0.015,0.643,0.641,0.535,0.903,0.218,0.412,3.967,0.223,-0.837,-0.021,-0.446,0.024,0.477,0.388,-3.6,-0.375,0.036,0.827,1.648,0.711,0.325,-0.553,-1.977,-5.154,-3.864,0.094,8064.0,0.0,0,[]],"sid":"98bc0ab6-7b56-4235-b336-f1e020a20330","time":1575369399.2645}
get_data got {"dev":[-1,0,[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]],"sid":"98bc0ab6-7b56-4235-b336-f1e020a20330","time":1575369399.3645}
get_data got {"eeg":[63,0,0.007,0.382,-2.229,0.169,0.67,-0.015,0.642,0.64,0.534,0.389,0.218,0.411,3.963,0.223,-0.323,-0.021,-0.446,0.024,0.477,0.387,-3.596,-0.375,0.036,0.826,2.159,0.71,0.325,-0.553,-1.975,-4.636,-3.347,0.094,8064.0,0.0,0,[]],"sid":"98bc0ab6-7b56-4235-b336-f1e020a20330","time":1575369399.2721}
get_data got {"eeg":[64,0,0.007,0.382,-0.175,0.168,0.669,-0.015,0.128,0.64,0.534,0.389,0.218,0.411,3.959,0.222,0.19,-0.021,-0.445,0.024,0.476,0.387,-3.593,0.139,0.036,0.825,1.644,0.71,0.325,-0.552,-1.973,-4.631,-3.344,0.094,0.0,0.0,0,[]],"sid":"98bc0ab6-7b56-4235-b336-f1e020a20330","time":1575369399.2799}
get_data got {"dev":[-1,0,[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]],"sid":"98bc0ab6-7b56-4235-b336-f1e020a20330","time":1575369399.3723}
get_data got {"dev":[-1,0,[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]],"sid":"98bc0ab6-7b56-4235-b336-f1e020a20330","time":1575369399.3801}
get_data got {"eeg":[65,0,0.007,0.381,-0.688,0.168,0.669,-0.015,1.667,0.126,0.533,0.388,0.218,0.41,3.955,0.222,-3.4,-0.021,-0.445,0.024,0.476,0.386,-3.589,-0.887,0.036,0.824,-0.922,0.709,0.324,-0.551,-1.971,-5.14,-3.853,0.094,0.0,0.0,0,[]],"sid":"98bc0ab6-7b56-4235-b336-f1e020a20330","time":1575369399.2875}
get_data got {"eeg":[66,0,0.007,0.381,-0.175,0.168,0.668,-0.015,3.716,0.639,0.533,0.901,0.217,0.41,3.951,0.222,-3.91,-0.021,-0.444,0.024,0.475,0.386,-3.586,-2.937,0.036,0.824,-0.921,0.708,0.324,-0.038,-1.969,-4.622,-3.336,0.094,0.0,0.0,0,[]],"sid":"98bc0ab6-7b56-4235-b336-f1e020a20330","time":1575369399.2953}
get_data got {"dev":[-1,0,[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]],"sid":"98bc0ab6-7b56-4235-b336-f1e020a20330","time":1575369399.3879}`
How can  I get eeg data and dev data at the same time? And what should I do if I want to get different data in different frequency?(The api get_data requires no args, how to get data flexibly?)
Thank u very much.

bchaperon, 2019-12-04:
Hello @fosaken 
Cortex sends the data samples one by one, and there is no guarantee that EEG and DEV samples will arrive at the same time.

To synchronize these 2 streams (and any other stream you subscribed to) you must use the field "time".
For each EEG sample, you will get a DEV sample with the exact same time. So we need to put the samples in a buffer and then match them 2 by 2, using their time.

fosaken, 2019-12-04:
Thank u  for ur answer @bchaperon .

My understanding now is as following:

So I should not call `cortex.get_data()` per 0.2s. if I want to get eeg data at frequencies lower than 128 hz and any other stream I subscribed to at any frequencies , I needn't set my own frequency and I can get them all at their fixed frequencies . I should sample the data I want at a time step I need(e.g. 0.2s), and match data from different streams by the field "time".

I wonder if it is right.

Thanks a lot.

bchaperon, 2019-12-04:
Sorry, I am not a Python developer, so I don't know the best way to get the data from the web socket.
For his kind of question, you should probably ask on stackoverflow.com or a Python forum.

I just know that for a Insight headset, Cortex will send you around 128 EEG samples and 128 CQ samples every seconds. For an Epoc+, it can be 256 EEG and 256 CQ per second.

fosaken, 2019-12-04:
Thank you for your big help and I can verify my ideas by experiments.

closed 2019-12-10 by bchaperon