Emotiv/cortex-example #63: BCI Cortex subscribe API issue

Akshathhm, 2019-12-07:
Hi,

In our profile we have "only one command" trained - that is PUSH
We have coded for the Subscribe API( for mental commands).. While listening to the socket/stream, without thinking anything or being in neutral state -we are getting PUSH command in the socket or API response..

This is not correct.. We should only got "neutral" in the response..

Interestingly, with BCI standalone application's "LIVE testing" mode doesn't behave as above.. That is "the cube doesn't get pushed" and it stays still while being in "neutral" or not thinking about PUSH..

Kindly request you to correct this flaw at the earliest.

Our profile details

UserId: zabibaig
Profile Name: aci12345

Akshathhm, 2019-12-07:
Kindly consider this issue with high severity

Akshathhm, 2019-12-07:
The above comment is because of closing similar issue in cortex v1

nguoithichkhampha, 2019-12-07:
@Akshathhm , I think your training is not good. therefor, detection return push action even you don't do anything.

Akshathhm, 2019-12-07:
Can you suggest on how to train?

If training is not good, then why does the BCI app - LIVE MODE's "cube" doesn't do PUSH?

nguoithichkhampha, 2019-12-09:
> If training is not good, then why does the BCI app - LIVE MODE's "cube" doesn't do PUSH?
did you see the difference from BCI and your project at the same time ? BCI show neutral and your project show push with same profile ? Can you capture a video and show me ?

abaniuszewicz, 2019-12-19:
What is the value associated with received push command? Maybe if it is very low you can filter it out in your code? That’s what I did - every command that has value below certain threshold is considered a noise and ignored.

closed 2019-12-26 by nguoithichkhampha