Emotiv/cortex-example #64: -32019 Session limit on this device has been reached

depauantonio59, 2019-12-11:
We are trying to get the command stream, we have modified the DataStreamExample.
Everything is fine until the _sessionCreator.Create(_cortexToken, headsetId), where we get the error code -32019.

This is our code:

using Newtonsoft.Json.Linq;
using System;
using System.Collections;
using System.Collections.Generic;

namespace CortexAccess
{
    public class DataStreamExample
    {
        private CortexClient _ctxClient;
        private List<string> _streams;
        private string _cortexToken;
        private string _sessionId;

        private HeadsetFinder _headsetFinder;
        private Authorizer _authorizer;
        private SessionCreator _sessionCreator;

        public List<string> Streams
        {
            get
            {
                return _streams;
            }

            set
            {
                _streams = value;
            }
        }

        public string SessionId
        {
            get
            {
                return _sessionId;
            }
        }

        // Event
        public event EventHandler<ArrayList> OnMotionDataReceived; // motion data
        public event EventHandler<ArrayList> OnEEGDataReceived; // eeg data
        //public event EventHandler<ArrayList> OnDevDataReceived; // contact quality data
        public event EventHandler<ArrayList> OnPerfDataReceived; // performance metric
        public event EventHandler<ArrayList> OnBandPowerDataReceived; // band power
        public event EventHandler<ArrayList> OnCommandDataReceived; // mental commands
        public event EventHandler<Dictionary<string, JArray>> OnSubscribed;

        private List<string> _profileLists;

        // Constructor
        public DataStreamExample(string AppClientId, string AppClientSecret) {

            // Event register
            _ctxClient = CortexClient.Instance;
            //
            _ctxClient.AppClientId = AppClientId;
            _ctxClient.AppClientSecret = AppClientSecret;
            _authorizer = new Authorizer();
            _headsetFinder = new HeadsetFinder();
            _sessionCreator = new SessionCreator();
            _cortexToken = "";
            _sessionId = "";
            _streams = new List<string>();
            _profileLists = new List<string>();
            //
            _ctxClient.OnErrorMsgReceived += MessageErrorRecieved;
            _ctxClient.OnStreamDataReceived += StreamDataReceived;
            _ctxClient.OnSubscribeData += SubscribeDataOK;
            _ctxClient.OnUnSubscribeData += UnSubscribeDataOK;
            _ctxClient.OnQueryProfile += QueryProfileOK;

            _authorizer.OnAuthorized += AuthorizedOK;
            _headsetFinder.OnHeadsetConnected += HeadsetConnectedOK;
            _sessionCreator.OnSessionActivated += SessionActivatedOk;
            _sessionCreator.OnSessionClosed += SessionClosedOK;
        }

        private void QueryProfileOK(object sender, JArray profiles)
        {
            //Console.WriteLine("QueryProfileOK" + profiles);
            foreach (JObject ele in profiles)
            {
                string name = (string)ele["name"];
                _profileLists.Add(name);
            }
            // find headset
            _headsetFinder.FindHeadset();
        }

        public void LoadProfile(string profileName)
        {
            if (_profileLists.Contains(profileName))
                _ctxClient.SetupProfile(_cortexToken, profileName, "load");
            //else
                //Console.WriteLine("The profile can not be loaded. The name " + profileName + " has not existed.");
        }

        private void SessionClosedOK(object sender, string sessionId)
        {
            //Console.WriteLine("The Session " + sessionId + " has closed successfully.");
        }

        private void UnSubscribeDataOK(object sender, MultipleResultEventArgs e)
        {
            foreach (JObject ele in e.SuccessList)
            {
                string streamName = (string)ele["streamName"];
                if (_streams.Contains(streamName))
                {
                    _streams.Remove(streamName);
                }
            }
            foreach (JObject ele in e.FailList)
            {
                string streamName = (string)ele["streamName"];
                int code = (int)ele["code"];
                string errorMessage = (string)ele["message"];
                //Console.WriteLine("UnSubscribe stream " + streamName + " unsuccessfully." + " code: " + code + " message: " + errorMessage);
            }
        }

        private void SubscribeDataOK(object sender, MultipleResultEventArgs e)
        {
            foreach (JObject ele in e.FailList)
            {
                string streamName = (string)ele["streamName"];
                int code = (int)ele["code"];
                string errorMessage = (string)ele["message"];
                //Console.WriteLine("Subscribe stream " + streamName + " unsuccessfully." + " code: " + code + " message: " + errorMessage);
                if (_streams.Contains(streamName))
                {
                    _streams.Remove(streamName);
                }
            }
            Dictionary<string, JArray> header = new Dictionary<string, JArray>();
            foreach (JObject ele in e.SuccessList)
            {
                string streamName = (string)ele["streamName"];
                JArray cols = (JArray)ele["cols"];
                header.Add(streamName, cols);
            }
            if (header.Count > 0)
            {
                OnSubscribed(this, header);
            }
            else
            {
                //Console.WriteLine("No Subscribe Stream Available");
            }
        }

        private void SessionActivatedOk(object sender, string sessionId)
        {
            // subscribe
            _sessionId = sessionId;
            _ctxClient.Subscribe(_cortexToken, _sessionId, Streams);
        }

        private void HeadsetConnectedOK(object sender, string headsetId)
        {
            ////Console.WriteLine("HeadsetConnectedOK " + headsetId);
            // CreateSession
            _sessionCreator.Create(_cortexToken, headsetId);
        }

        private void AuthorizedOK(object sender, string cortexToken)
        {
            if (!String.IsNullOrEmpty(cortexToken))
            {
                _cortexToken = cortexToken;
                // find headset
                _headsetFinder.FindHeadset();
            }
        }

        private void StreamDataReceived(object sender, StreamDataEventArgs e)
        {
            //Console.WriteLine(e.StreamName + " data received.");
            ArrayList data = e.Data.ToObject<ArrayList>();
            // insert timestamp to datastream
            data.Insert(0, e.Time);
            if (e.StreamName == "eeg")
            {
                OnEEGDataReceived(this, data);
            }
            else if (e.StreamName == "mot")
            {
                
                OnMotionDataReceived(this, data);
            }
            else if (e.StreamName == "met")
            {
                OnPerfDataReceived(this, data);
            }
            else if (e.StreamName == "pow")
            {
                OnBandPowerDataReceived(this, data);
            }
            else if (e.StreamName == "com")
            {
                OnCommandDataReceived(this, data);
            }
        }
        private void MessageErrorRecieved(object sender, ErrorMsgEventArgs e)
        {
            Console.WriteLine("MessageErrorRecieved :code " + e.Code + " message " + e.MessageError);
        }

        // set Streams
        public void AddStreams(string stream)
        {
            if (!_streams.Contains(stream))
            {
                _streams.Add(stream);
            }
        }
        // start
        public void Start()
        {
            _authorizer.Start();
        }

        // Unsubscribe
        public void UnSubscribe(List<string> streams = null)
        {
            if (streams == null)
            {
                // unsubscribe all data
                _ctxClient.UnSubscribe(_cortexToken, _sessionId, _streams);
            }
            else 
                _ctxClient.UnSubscribe(_cortexToken, _sessionId, streams);
        }
        public void CloseSession()
        {
            _sessionCreator.CloseSession();
        }
    }
}

nguoithichkhampha, 2019-12-16:
@depauantonio59 . could you try with this extension
https://chrome.google.com/webstore/detail/simple-websocket-client/pfdhoblngboilpfeibdedpjgfnlcodoo?hl=en
and call api [get license info](https://emotiv.gitbook.io/cortex-api/authentication/getlicenseinfo) of Cortex
by cortex token you received.

depauantonio59, 2019-12-18:
Hi,
we get the following reply:
{"id":1,"jsonrpc":"2.0","result":{"isOnline":true,"license":{"applications":["com.emotiv.sdk.basic"],"billingFrom":"2019-11-11T01:00:00.000+01:00","billingTo":"2029-11-08T01:00:00.000+01:00","deviceInfo":{"deviceLimit":3,"devicesPerSeat":3,"sessionLimit":{"day":null,"month":null,"year":null}},"expired":false,"extenderLimit":0,"hardLimitTime":"2020-01-19T00:59:59.999+01:00","isCommercial":false,"licenseId":"6d96548c-f5eb-4c6b-bbef-742c187737cb","licenseName":"Basic License","localQuota":0,"maxDebit":null,"scopes":[""],"seatCount":1,"sessionCount":0,"softLimitTime":"2020-01-12T00:59:59.999+01:00","totalDebit":0,"totalRegisteredDevices":1,"validFrom":"2019-11-11T01:00:00.000+01:00","validTo":"2029-11-09T00:59:59.999+01:00"}}}

nguoithichkhampha, 2019-12-18:
hi @depauantonio59 , from the response you can see 
"totalDebit":0 and "localQuota":0 -> mean you did not debit . So, you don't have local session available to create (localQuota = 0).
to fix it, in the request [authorize](https://emotiv.gitbook.io/cortex-api/authentication/authorize) you should set param "debit"

depauantonio59, 2019-12-19:
Hi,
In the Autorizhe() call we are currently passing an empty string as LicenseID, and 5 for debit.

depauantonio59, 2019-12-19:
Hi,
In the Autorize() call we are currently passing an empty string as LicenseID, and 5 for debit.

nguoithichkhampha, 2019-12-20:
oh, really. I wonder why Cortex return totalDebit = 0.
Can you try to make 2 request authorize and get license info, after that send the result for me via email
cuong@emotiv.com ?

depauantonio59, 2019-12-20:
Hi,
I have found out that in your example code, when licenseID was empty it was sending debit=0 in the authorize() call. I fixed it and now we local quota:

{"id":1,"jsonrpc":"2.0","result":{"isOnline":true,"license":{"applications":["com.emotiv.sdk.basic"],"billingFrom":"2019-11-11T01:00:00.000+01:00","billingTo":"2029-11-08T01:00:00.000+01:00","deviceInfo":{"deviceLimit":3,"devicesPerSeat":3,"sessionLimit":{"day":null,"month":null,"year":null}},"expired":false,"extenderLimit":0,"hardLimitTime":"2020-01-19T00:59:59.999+01:00","isCommercial":false,"licenseId":"6d96548c-f5eb-4c6b-bbef-742c187737cb","licenseName":"Basic License","localQuota":5,"maxDebit":null,"scopes":[""],"seatCount":1,"sessionCount":0,"softLimitTime":"2020-01-12T00:59:59.999+01:00","totalDebit":5,"totalRegisteredDevices":1,"validFrom":"2019-11-11T01:00:00.000+01:00","validTo":"2029-11-09T00:59:59.999+01:00"}}}

Thanks for your help!

closed 2019-12-20 by depauantonio59

patdotme, 2020-01-07:
Fixed in PR https://github.com/Emotiv/cortex-v2-example/pull/68