Emotiv/cortex-example #69: Session limit on this device has been reached

michelpf, 2020-01-06:
This error is pretty unclear to me, sorry.

It seems to be related to a limit of API, but how we can check it? I'm not abusing data collect anyway...

Maybe that value could be a hint?

````
 'localQuota': -1474834899,
````

Most cases here were solved but I can't figure out how. Some users changed the  ```debit```parameter to 100 and worked. In my case, not... :(

Since the new version of API, I'm struggling to collect data consistently.

Giving my logs, anyone can help to understand why I can't read the data stream?

Here some logs (removed intentionally user, pass and tokens). I'm using the python demo for this scenario.

````
** REQUEST ACCESS **
Sending request:
{"jsonrpc": "2.0", "method": "requestAccess", "params": {"clientId": "", "clientSecret": ""}, "id": 5}
sent; awaiting response
lib.cortex resp:
{'id': 5, 'jsonrpc': '2.0', 'result': {'accessGranted': True, 'message': 'The user has granted access right to this application.'}}
** AUTHORIZE **
Sending request:
{"jsonrpc": "2.0", "method": "authorize", "params": {"clientId": "", "clientSecret": "", "debit": 1}, "id": 6}
sent; awaiting response
lib.cortex resp:
{'id': 6, 'jsonrpc': '2.0', 'result': {'cortexToken': ''}}
** GET LICENSE INFO **
Sending request:
{"jsonrpc": "2.0", "method": "getLicenseInfo", "params": {"cortexToken": ""}, "id": 7}
sent; awaiting response
lib.cortex resp:
{'id': 7, 'jsonrpc': '2.0', 'result': {'isOnline': True, 'license': {'applications': ['com.emotiv.sdk.basic'], 'billingFrom': '2018-08-28T21:00:00.000-03:00', 'billingTo': '2028-08-25T21:00:00.000-03:00', 'deviceInfo': {'deviceLimit': 3, 'devicesPerSeat': 3, 'sessionLimit': {'day': None, 'month': None, 'year': None}}, 'expired': False, 'extenderLimit': 0, 'hardLimitTime': '2020-02-05T20:59:59.999-03:00', 'isCommercial': False, 'licenseId': 'd0cd774a-0639-4e45-a09f-9a349bf177e8', 'licenseName': 'Basic License', 'localQuota': -1474834899, 'maxDebit': None, 'scopes': [''], 'seatCount': 1, 'sessionCount': 1, 'softLimitTime': '2020-01-29T20:59:59.999-03:00', 'totalDebit': 2820132393, 'totalRegisteredDevices': 1, 'validFrom': '2018-08-28T21:00:00.000-03:00', 'validTo': '2028-08-26T20:59:59.999-03:00'}}}
** QUERY HEADSETS **
Sending request:
{"jsonrpc": "2.0", "method": "queryHeadsets", "params": {}, "id": 8}
sent; awaiting response
lib.cortex found headsets ['INSIGHT-5A68CB86']
lib.cortex resp:
{'id': 8, 'jsonrpc': '2.0', 'result': [{'connectedBy': 'bluetooth', 'customName': '', 'dongle': '0', 'firmware': '925', 'id': 'INSIGHT-5A68CB86', 'motionSensors': ['GYROX', 'GYROY', 'GYROZ', 'ACCX', 'ACCY', 'ACCZ', 'MAGX', 'MAGY', 'MAGZ'], 'sensors': ['AF3', 'T7', 'Pz', 'T8', 'AF4'], 'settings': {'eegRate': 128, 'eegRes': 14, 'memsRate': 128, 'memsRes': 14, 'mode': 'INSIGHT'}, 'status': 'connected'}]}
** CREATE SESSION **
Sending request:
{"jsonrpc": "2.0", "method": "createSession", "params": {"cortexToken": "", "headset": "INSIGHT-5A68CB86", "status": "active"}, "id": 9}
sent; awaiting response
Got error in createSession with params {'cortexToken': '', 'headset': 'INSIGHT-5A68CB86', 'status': 'active'}:
{"error":{"code":-32019,"message":"Session limit on this device has been reached."},"id":9,"jsonrpc":"2.0"}
```

nguoithichkhampha, 2020-01-07:
@michelpf , thanks for your feedback. what is platform you are using ?

michelpf, 2020-01-07:
I'm using MacOS, with Python example.

nguoithichkhampha, 2020-01-10:
@michelpf , can you send me database of cortex at path "/Library/Application Support/Emotiv/Cortex/v2/database" via email cuong@emotiv.com

michelpf, 2020-01-10:
I just sent the files to you.
Yes, I tried with 0, None, 1, 10, 100, 10000000.

nguoithichkhampha, 2020-01-10:
@michelpf I think our system has been overflow. Do you have any data on local want to keep and is not uploaded on cloud ? if not we can try:

1. uninstall cortex
2. delete folder "/Library/Application Support/Emotiv/Cortex"
3. install cortex again.

and pls don't debit with big number. we will fix this issue in next version.

mecowhy, 2020-01-10:
similar problem here but using MotionLogger C# sample app: 

> Motion LOGGER
Please wear Headset with good signal!!!
Send getUserLogin
Press Esc to exit
Received: {"id":1,"jsonrpc":"2.0","result":[{"currentOSUId":"1001","currentOSUsername":"myName","lastLoginTime":"2020-01-10T16:34:47.814+01:00","loggedInOSUId":"1001","loggedInOSUsername":"myName","username":"myLogin"}]}
handleResponse: getUserLogin
Send hasAccessRight
Received: {"id":2,"jsonrpc":"2.0","result":{"accessGranted":true,"message":"The access right to the application has already been granted."}}
handleResponse: hasAccessRight
Send authorize
Received: {"id":3,"jsonrpc":"2.0","result":{"cortexToken":""}}
handleResponse: authorize
Authorize successfully.
FindHeadset
Send queryHeadsets
Received: {"id":4,"jsonrpc":"2.0","result":[{"connectedBy":"bluetooth","customName":"","dongle":"0","firmware":"815","id":"INSIGHT-999999","motionSensors":["GYROX","GYROY","GYROZ","ACCX","ACCY","ACCZ","MAGX","MAGY","MAGZ"],"sensors":["AF3","T7","Pz","T8","AF4"],"settings":{"eegRate":128,"eegRes":14,"memsRate":128,"memsRes":14,"mode":"INSIGHT"},"status":"connected"}]}
handleResponse: queryHeadsets
HeadsetConnectedOK INSIGHT-999999
Send createSession
Received: {"error":{"code":-32019,"message":"Session limit on this device has been reached."},"id":5,"jsonrpc":"2.0"}
Received: Session limit on this device has been reached.
MessageErrorRecieved :code -32019 message Session limit on this device has been reached.

token,password,names,serial number removed.

nguoithichkhampha, 2020-01-10:
@mecowhy , can you show me response of request [getlicenseinfo](https://emotiv.gitbook.io/cortex-api/authentication/getlicenseinfo)

mecowhy, 2020-01-10:
ok now its clear, my pending free licence request ...
any ideas what can be the reason its not approved yet (10 days) ?

`{"id":5,"jsonrpc":"2.0","result":{"isOnline":true,"license":{"applications":["com.emotiv.sdk.basic"],"billingFrom":"2019-10-15T02:00:00.000+02:00","billingTo":"2029-10-12T02:00:00.000+02:00","deviceInfo":{"deviceLimit":3,"devicesPerSeat":3,"sessionLimit":{"day":null,"month":null,"year":null}},"expired":false,"extenderLimit":0,"hardLimitTime":"2020-01-23T00:59:59.999+01:00","isCommercial":false,"licenseId":"5b3e4595-ef25-499e-8868-bacdaebc2420","licenseName":"Basic License","localQuota":0,"maxDebit":null,"scopes":[""],"seatCount":1,"sessionCount":0,"softLimitTime":"2020-01-16T00:59:59.999+01:00","totalDebit":0,"totalRegisteredDevices":1,"validFrom":"2019-10-15T02:00:00.000+02:00","validTo":"2029-10-13T01:59:59.999+02:00"}}}`

nguoithichkhampha, 2020-01-10:
because you did not set "debit" param in the request "authorize".
As I see "localQuota":0 and "totalDebit":0

mecowhy, 2020-01-13:
Hi,
I set DebitNumber to 1 and got this:

`Motion LOGGER
Please wear Headset with good signal!!!
Send getUserLogin
TX: {
  "jsonrpc": "2.0",
  "id": 1,
  "method": "getUserLogin"
}
Press Esc to exit
RX: {
  "id": 1,
  "jsonrpc": "2.0",
  "result": [
    {
      "currentOSUId": "1001",
      "currentOSUsername": "__REMOVED__",
      "lastLoginTime": "2020-01-10T16:34:47.814+01:00",
      "loggedInOSUId": "1001",
      "loggedInOSUsername": "__REMOVED__",
      "username": "__REMOVED__"
    }
  ]
}
handleResponse: getUserLogin
Send hasAccessRight
TX: {
  "jsonrpc": "2.0",
  "id": 2,
  "method": "hasAccessRight",
  "params": {
    "clientId": "__REMOVED__",
    "clientSecret": "__REMOVED__"
  }
}
RX: {
  "id": 2,
  "jsonrpc": "2.0",
  "result": {
    "accessGranted": true,
    "message": "The access right to the application has already been granted."
  }
}
handleResponse: hasAccessRight
Send authorize
TX: {
  "jsonrpc": "2.0",
  "id": 3,
  "method": "authorize",
  "params": {
    "clientId": "__REMOVED__",
    "clientSecret": "__REMOVED__",
    "debit": 0
  }
}
RX: {
  "id": 3,
  "jsonrpc": "2.0",
  "result": {
    "cortexToken": "__REMOVED__"
  }
}
handleResponse: authorize
Authorize successfully.
FindHeadset
Send queryHeadsets
TX: {
  "jsonrpc": "2.0",
  "id": 4,
  "method": "queryHeadsets"
}
Send getLicenseInfo
TX: {
  "jsonrpc": "2.0",
  "id": 5,
  "method": "getLicenseInfo",
  "params": {
    "cortexToken": "__REMOVED__"
  }
}
RX: {
  "id": 4,
  "jsonrpc": "2.0",
  "result": [
    {
      "connectedBy": "dongle",
      "customName": "",
      "dongle": "6ff",
      "firmware": "565",
      "id": "EPOC-__REMOVED__",
      "motionSensors": [
        "GYROX",
        "GYROY"
      ],
      "sensors": [
        "AF3",
        "F7",
        "F3",
        "FC5",
        "T7",
        "P7",
        "O1",
        "O2",
        "P8",
        "T8",
        "FC6",
        "F4",
        "F8",
        "AF4"
      ],
      "settings": {
        "eegRate": 128,
        "eegRes": 14,
        "memsRate": 128,
        "memsRes": 14,
        "mode": "STANDARD"
      },
      "status": "connected"
    },
    {
      "connectedBy": "bluetooth",
      "customName": "",
      "dongle": "0",
      "firmware": "0",
      "id": "INSIGHT-__REMOVED__",
      "motionSensors": [],
      "sensors": [],
      "settings": {
        "eegRate": 128,
        "eegRes": 0,
        "memsRate": 0,
        "memsRes": 0,
        "mode": "UNKNOWN"
      },
      "status": "discovered"
    }
  ]
}
handleResponse: queryHeadsets
HeadsetConnectedOK EPOC-__REMOVED__
Send createSession
TX: {
  "jsonrpc": "2.0",
  "id": 6,
  "method": "createSession",
  "params": {
    "headset": "EPOC-__REMOVED__",
    "cortexToken": "__REMOVED__",
    "status": "active"
  }
}
RX: {
  "error": {
    "code": -32019,
    "message": "Session limit on this device has been reached."
  },
  "id": 6,
  "jsonrpc": "2.0"
}
Received: Session limit on this device has been reached.
MessageErrorRecieved :code -32019 message Session limit on this device has been reached.
RX: {
  "id": 5,
  "jsonrpc": "2.0",
  "result": {
    "isOnline": true,
    "license": {
      "applications": [
        "com.emotiv.sdk.basic"
      ],
      "billingFrom": "2019-10-15T02:00:00+02:00",
      "billingTo": "2029-10-12T02:00:00+02:00",
      "deviceInfo": {
        "deviceLimit": 3,
        "devicesPerSeat": 3,
        "sessionLimit": {
          "day": null,
          "month": null,
          "year": null
        }
      },
      "expired": false,
      "extenderLimit": 0,
      "hardLimitTime": "2020-01-23T00:59:59.999+01:00",
      "isCommercial": false,
      "licenseId": "5b3e4595-ef25-499e-8868-bacdaebc2420",
      "licenseName": "Basic License",
      "localQuota": 0,
      "maxDebit": null,
      "scopes": [
        ""
      ],
      "seatCount": 1,
      "sessionCount": 0,
      "softLimitTime": "2020-01-16T00:59:59.999+01:00",
      "totalDebit": 0,
      "totalRegisteredDevices": 1,
      "validFrom": "2019-10-15T02:00:00+02:00",
      "validTo": "2029-10-13T01:59:59.999+02:00"
    }
  }
}
handleResponse: getLicenseInfo`

nguoithichkhampha, 2020-01-13:
@mecowhy which line code you changed to set debit = 1. I still see "debit" 0 when send request to cortex
`TX: {
  "jsonrpc": "2.0",
  "id": 3,
  "method": "authorize",
  "params": {
    "clientId": "__REMOVED__",
    "clientSecret": "__REMOVED__",
    "debit": 0
  }
}`

mecowhy, 2020-01-13:
ok now its working , previously I had licenceID field empty in Config class, now I used value from getLicenceInfo response and its working.

michelpf, 2020-01-13:
It worked!
Thank you so much!

closed 2020-01-13 by michelpf

catonmat, 2020-09-05:
This issue solved my issue too -- any chance you could update the official documentation on your website to reflect the necessity of including `"debit": 1` in the `"method": authorize` API call?

https://emotiv.gitbook.io/cortex-api/authentication/authorize