Emotiv/cortex-example #71: Using python to export record

liyi9810857, 2020-01-13:
### Using python code to exportRecord
I'm trying to export recording using python code. Since there is an API for exportRecord using websocket and JSON, but there is no such example in[ lib/cortex.py](https://github.com/Emotiv/cortex-v2-example/blob/master/python/lib/cortex.py), so I added some code in this file as:

```
async def export_record(self, recordIds = None, folder = None, streamTypes = None):
        params = {
            'cortexToken': self.auth_token,
            'recordIds': recordIds,
            'folder': folder,
            'streamTypes': streamTypes,
            'format': 'CSV',
            'version': 'V2'
        }

        resp = await self.send_command('exportRecord', **params)
        logger.debug(f"{__name__} resp:\n{resp}")
        return resp
```
hopes to get response just like that in the document, and see exported file:

```
{
    "id": 1,
    "jsonrpc": "2.0",
    "result": {
        "failure": [],
        "success": [
            {
                "recordId": "ec0ac33f-ad4e-48b1-bbc3-8502f5c49b62"
            }
        ]
    }
}
```

I added some calling code in the [example.py ](https://github.com/Emotiv/cortex-v2-example/blob/master/python/example.py)as follows:

```
......
if len(cortex.headsets) > 0:
        print("** CREATE SESSION **")
        await cortex.create_session(activate=True,
                                    headset_id=cortex.headsets[0])
        print("** CREATE RECORD **")
        resp = await cortex.create_record(title="test record 1")
        # for ID test
        id = resp['result']['record']['uuid']
        print('id: ' + id )

        print("** SUBSCRIBE EEG & MOT **")
        await cortex.subscribe(['mot'])
        while cortex.packet_count < 10:
            await cortex.get_data()
        await cortex.inject_marker(label='halfway', value=1,
                                   time=cortex.to_epoch())
        while cortex.packet_count < 20:
            await cortex.get_data()
        print("** STOP THE RECORD")
        await cortex.stop_record()
        print("** DISCONNECT THE HEADSET")
        await cortex.control_device('disconnect', headset_id=cortex.headsets[0])
        # export motion data only.
        streamTypes = ['MOTION']
        folder = './export'
        print('Export')
        resp = await cortex.export_record( recordIds=id, folder=folder, streamTypes=streamTypes)
        # check whether export is success or not
        result = resp['result']['success']
        print('Finish Export')
```

But nothing happened, no exported file.  And the response from stop the record is like:

```
......
** STOP THE RECORD
Sending request:
{"jsonrpc": "2.0", "method": "stopRecord", "params": {"cortexToken": "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhcHBJZCI6ImNvbS50b2hlYXIxMDAxLnNvdW5kdGVzdCIsImFwcFZlcnNpb24iOiIxLjAiLCJleHAiOjE1NzkxMDA4NTUsIm5iZiI6MTU3ODg0MTY1NSwidXNlcklkIjoiODM5MjZhY2UtNGQ3Zi00MTZkLWFlOWEtYTljZmI4NjVhYzg1IiwidXNlcm5hbWUiOiJ0b2hlYXIxMDAxIiwidmVyc2lvbiI6IjIuMCJ9.dbbVMB5fraNuq/0D0KApT5vh6vomEGqkPQ8cjmkFKKQ=", "session": "74554d22-5298-4dbe-b987-c094bada5112"}, "id": 12}
sent; awaiting response
lib.cortex resp:
{'mot': [40, 0, 8150.0, 8146.0, 8226.0, 12283.0, 8194.0, 8541.0, 8270.0, 8310.0, 8070.0], 'sid': '74554d22-5298-4dbe-b987-c094bada5112', 'time': 1578928056.3054}
** DISCONNECT THE HEADSET
Sending request:
{"jsonrpc": "2.0", "method": "controlDevice", "params": {"headset": "EPOCPLUS-3B9ADA34", "command": "disconnect", "cortexToken": "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhcHBJZCI6ImNvbS50b2hlYXIxMDAxLnNvdW5kdGVzdCIsImFwcFZlcnNpb24iOiIxLjAiLCJleHAiOjE1NzkxMDA4NTUsIm5iZiI6MTU3ODg0MTY1NSwidXNlcklkIjoiODM5MjZhY2UtNGQ3Zi00MTZkLWFlOWEtYTljZmI4NjVhYzg1IiwidXNlcm5hbWUiOiJ0b2hlYXIxMDAxIiwidmVyc2lvbiI6IjIuMCJ9.dbbVMB5fraNuq/0D0KApT5vh6vomEGqkPQ8cjmkFKKQ="}, "id": 13}
sent; awaiting response
lib.cortex resp:
{'mot': [41, 0, 8144.0, 8145.0, 8226.0, 12289.0, 8184.0, 8535.0, 8267.0, 8311.0, 8066.0], 'sid': '74554d22-5298-4dbe-b987-c094bada5112', 'time': 1578928056.3212}
Export
Sending request:
{"jsonrpc": "2.0", "method": "exportRecord", "params": {"cortexToken": "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhcHBJZCI6ImNvbS50b2hlYXIxMDAxLnNvdW5kdGVzdCIsImFwcFZlcnNpb24iOiIxLjAiLCJleHAiOjE1NzkxMDA4NTUsIm5iZiI6MTU3ODg0MTY1NSwidXNlcklkIjoiODM5MjZhY2UtNGQ3Zi00MTZkLWFlOWEtYTljZmI4NjVhYzg1IiwidXNlcm5hbWUiOiJ0b2hlYXIxMDAxIiwidmVyc2lvbiI6IjIuMCJ9.dbbVMB5fraNuq/0D0KApT5vh6vomEGqkPQ8cjmkFKKQ=", "recordIds": "e05d038b-6428-4345-939e-afe934f4488f", "folder": "./export", "streamTypes": ["MOTION"], "format": "CSV", "version": "V2"}, "id": 14}
sent; awaiting response
lib.cortex resp:
{'mot': [42, 0, 8147.0, 8143.0, 8223.0, 12283.0, 8186.0, 8526.0, 8267.0, 8312.0, 8069.0], 'sid': '74554d22-5298-4dbe-b987-c094bada5112', 'time': 1578928056.337}
Finish Export
```

I think there is something wrong with these responses? Can anyone help me out? 

### Calling sequence for export record
Another problem is about the calling sequence for exportRecord. The current sequence I use is:
* stop record
* disconnect headset
* export record
* close session
I don't know whether the sequence is right or not?

tungntEmotiv, 2020-01-14:
Hi @liyi9810857 ,

The problem is at disconnect headset, do not put "cortexToken" here. 
For exporting record, the sequence will be: stop record -> close session (optional) -> disconnect headset -> export Record

Please try and let me know about the result. Thanks

liyi9810857, 2020-01-14:
Thank you for your advice. 
I made some change to the code: 

[cortex.py.txt](https://github.com/Emotiv/cortex-v2-example/files/4061679/cortex.py.txt)
[test_vnew.py.txt](https://github.com/Emotiv/cortex-v2-example/files/4061683/test_vnew.py.txt)

I tried again. Still there are problems:
1. to disconnect headset, using the control_device provided in the cortex.py, as cortex. control_device(command='disconnect', headset_id=cortex.headset[0]), I can't find where is the cortexToken.
2. Using the calling sequence, get response output as follows:

** USER LOGIN **
Sending request:
{"jsonrpc": "2.0", "method": "getUserLogin", "params": {}, "id": 1}
sent; awaiting response
lib.cortex resp:
{'currentOSUId': '501', 'currentOSUsername': 'liyi', 'lastLoginTime': '2020-01-10T18:33:41.619-05:00', 'loggedInOSUId': '501', 'loggedInOSUsername': 'liyi', 'username': 'tohear1001'}
** GET CORTEX INFO **
Sending request:
{"jsonrpc": "2.0", "method": "getCortexInfo", "params": {}, "id": 2}
sent; awaiting response
lib.cortex resp:
{'id': 2, 'jsonrpc': '2.0', 'result': {'buildDate': '2019-12-04T15:01:43', 'buildNumber': 'v2.2.3-62-ge0a57e0a', 'version': '2.3.0'}}
** HAS ACCESS RIGHT **
Sending request:
{"jsonrpc": "2.0", "method": "requestAccess", "params": {"clientId": "SIpa7mS4vlKC5WS9NgJBpJjFnTWxpqKWLp6nz5XP", "clientSecret": "k2apbNzV2ZiH1bOTooRiwTTuO7rKaheAW5yF5eLWXO0TtQFZiJn9yHMWogeB8HY32F4R4vyQ0VEOAm6n7XXkkmC5w6758dySHGqIZFVlDdJ96bMFsvzCjejpatXmcfLY"}, "id": 3}
sent; awaiting response
lib.cortex resp:
{'id': 3, 'jsonrpc': '2.0', 'result': {'accessGranted': True, 'message': 'The user has granted access right to this application.'}}
** REQUEST ACCESS **
Sending request:
{"jsonrpc": "2.0", "method": "requestAccess", "params": {"clientId": "SIpa7mS4vlKC5WS9NgJBpJjFnTWxpqKWLp6nz5XP", "clientSecret": "k2apbNzV2ZiH1bOTooRiwTTuO7rKaheAW5yF5eLWXO0TtQFZiJn9yHMWogeB8HY32F4R4vyQ0VEOAm6n7XXkkmC5w6758dySHGqIZFVlDdJ96bMFsvzCjejpatXmcfLY"}, "id": 4}
sent; awaiting response
lib.cortex resp:
{'id': 4, 'jsonrpc': '2.0', 'result': {'accessGranted': True, 'message': 'The user has granted access right to this application.'}}
** AUTHORIZE **
Sending request:
{"jsonrpc": "2.0", "method": "authorize", "params": {"clientId": "SIpa7mS4vlKC5WS9NgJBpJjFnTWxpqKWLp6nz5XP", "clientSecret": "k2apbNzV2ZiH1bOTooRiwTTuO7rKaheAW5yF5eLWXO0TtQFZiJn9yHMWogeB8HY32F4R4vyQ0VEOAm6n7XXkkmC5w6758dySHGqIZFVlDdJ96bMFsvzCjejpatXmcfLY", "debit": 100}, "id": 5}
sent; awaiting response
lib.cortex resp:
{'id': 5, 'jsonrpc': '2.0', 'result': {'cortexToken': 'eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhcHBJZCI6ImNvbS50b2hlYXIxMDAxLnNvdW5kaGVhcmluZyIsImFwcFZlcnNpb24iOiIxLjAiLCJleHAiOjE1NzkyMTExMDQsIm5iZiI6MTU3ODk1MTkwNCwidXNlcklkIjoiODM5MjZhY2UtNGQ3Zi00MTZkLWFlOWEtYTljZmI4NjVhYzg1IiwidXNlcm5hbWUiOiJ0b2hlYXIxMDAxIiwidmVyc2lvbiI6IjIuMCJ9.G6MFpFnh7+IuJT22eYnNTsJ4d+7KL4gYBKf+U77UfZ8='}}
** GET LICENSE INFO **
Sending request:
{"jsonrpc": "2.0", "method": "getLicenseInfo", "params": {"cortexToken": "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhcHBJZCI6ImNvbS50b2hlYXIxMDAxLnNvdW5kaGVhcmluZyIsImFwcFZlcnNpb24iOiIxLjAiLCJleHAiOjE1NzkyMTExMDQsIm5iZiI6MTU3ODk1MTkwNCwidXNlcklkIjoiODM5MjZhY2UtNGQ3Zi00MTZkLWFlOWEtYTljZmI4NjVhYzg1IiwidXNlcm5hbWUiOiJ0b2hlYXIxMDAxIiwidmVyc2lvbiI6IjIuMCJ9.G6MFpFnh7+IuJT22eYnNTsJ4d+7KL4gYBKf+U77UfZ8="}, "id": 6}
sent; awaiting response
lib.cortex resp:
{'id': 6, 'jsonrpc': '2.0', 'result': {'isOnline': True, 'license': {'applications': ['com.emotiv.sdk.basic'], 'billingFrom': '2020-01-10T00:00:00.000+00:00', 'billingTo': '2030-01-07T00:00:00.000+00:00', 'deviceInfo': {'deviceLimit': 3, 'devicesPerSeat': 3, 'sessionLimit': {'day': None, 'month': None, 'year': None}}, 'expired': False, 'extenderLimit': 0, 'hardLimitTime': '2020-02-17T23:59:59.999+00:00', 'isCommercial': False, 'licenseId': 'ee5e51cd-bd71-4bd6-87ae-d8135504e6bd', 'licenseName': 'Basic License', 'localQuota': 4394, 'maxDebit': None, 'scopes': [''], 'seatCount': 1, 'sessionCount': 7, 'softLimitTime': '2020-02-10T23:59:59.999+00:00', 'totalDebit': 4202, 'totalRegisteredDevices': 1, 'validFrom': '2020-01-10T00:00:00.000+00:00', 'validTo': '2030-01-07T23:59:59.999+00:00'}}}
** QUERY HEADSETS **
Sending request:
{"jsonrpc": "2.0", "method": "queryHeadsets", "params": {}, "id": 7}
sent; awaiting response
lib.cortex found headsets ['EPOCPLUS-3B9ADA34']
lib.cortex resp:
{'id': 7, 'jsonrpc': '2.0', 'result': [{'connectedBy': 'bluetooth', 'customName': '', 'dongle': '0', 'firmware': '625', 'id': 'EPOCPLUS-3B9ADA34', 'motionSensors': ['GYROX', 'GYROY', 'GYROZ', 'ACCX', 'ACCY', 'ACCZ', 'MAGX', 'MAGY', 'MAGZ'], 'sensors': ['AF3', 'F7', 'F3', 'FC5', 'T7', 'P7', 'O1', 'O2', 'P8', 'T8', 'FC6', 'F4', 'F8', 'AF4'], 'settings': {'eegRate': 128, 'eegRes': 16, 'memsRate': 64, 'memsRes': 16, 'mode': 'EPOCPLUS'}, 'status': 'connected'}]}
** CREATE SESSION **
Sending request:
{"jsonrpc": "2.0", "method": "createSession", "params": {"cortexToken": "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhcHBJZCI6ImNvbS50b2hlYXIxMDAxLnNvdW5kaGVhcmluZyIsImFwcFZlcnNpb24iOiIxLjAiLCJleHAiOjE1NzkyMTExMDQsIm5iZiI6MTU3ODk1MTkwNCwidXNlcklkIjoiODM5MjZhY2UtNGQ3Zi00MTZkLWFlOWEtYTljZmI4NjVhYzg1IiwidXNlcm5hbWUiOiJ0b2hlYXIxMDAxIiwidmVyc2lvbiI6IjIuMCJ9.G6MFpFnh7+IuJT22eYnNTsJ4d+7KL4gYBKf+U77UfZ8=", "headset": "EPOCPLUS-3B9ADA34", "status": "active"}, "id": 8}
sent; awaiting response
lib.cortex resp:
[... cut ...]
/Users/liyi/MyDocuments/GitHub/cortex-v2-example/python/lib/cortex.py:173: RuntimeWarning: coroutine 'WebSocketCommonProtocol.close' was never awaited
  self.websocket.close()

Still, nothing happens, no file exported. 
I'm wondering whether the export folder I set is OK or not? use relative path or absolute path?
And also, I didn't see the response from exportRecord, which should tell me whether the export is successful or not.
When I add this code just before ' print('Finish Export'):

result = resp['result']['success']

It raises some errors. 

Is this has something to do with license? 
Thanks.

tungntEmotiv, 2020-01-17:
Hi @liyi9810857 ,

I see in your request 
`Sending request:
{"jsonrpc": "2.0", "method": "controlDevice", "params": {"headset": "EPOCPLUS-3B9ADA34", "command": "disconnect", "cortexToken": "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhcHBJZCI6ImNvbS50b2hlYXIxMDAxLnNvdW5kaGVhcmluZyIsImFwcFZlcnNpb24iOiIxLjAiLCJleHAiOjE1NzkyMTExMDQsIm5iZiI6MTU3ODk1MTkwNCwidXNlcklkIjoiODM5MjZhY2UtNGQ3Zi00MTZkLWFlOWEtYTljZmI4NjVhYzg1IiwidXNlcm5hbWUiOiJ0b2hlYXIxMDAxIiwidmVyc2lvbiI6IjIuMCJ9.G6MFpFnh7+IuJT22eYnNTsJ4d+7KL4gYBKf+U77UfZ8="}, "id": 14}`
The API doesn't include "cortexToken" . Please remove the parameter . Reference to https://emotiv.gitbook.io/cortex-api/headset/controldevice
Thanks

closed 2020-07-08 by tungntEmotiv