Emotiv/cortex-example #74: Everything is ok until i subscribe to "com" after that i received the subscription successfully then nothing appears

Hayhat95, 2020-02-15:
Greetings I am doing a job with the 5-channel Emotiv Insight device I am using cortex to extract the data such as mental commands and gyroscope, I have no problems at the time of creating cortex session and data subscription apparently but I do not throw the values ​​in In the case of the gyroscope I am using the following code in python

import json
from websocket import create_connection
import ssl
import time
import pyautogui

thought = ""
token = "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhcHBJZCI6ImNvbS5yaWNrc2Fuc29tLmRhdG9zZW1vdGl2IiwiYXBwVmVyc2lvbiI6IjEuMCIsImV4cCI6MTU2NTU1MDYxMSwibGljZW5zZUlkIjoiYTQxZDJkYzAtZWE1OS00ZDFhLWIwNTctODNkYWU1ZmJmZmM4IiwibmJmIjoxNTY1MjkxNDExLCJ1c2VySWQiOiI5Y2FiYTc5NS0yYTM2LTRlNWItOTJmNC1lMjYwZWNiYWUwMmQiLCJ1c2VybmFtZSI6InJpY2tzYW5zb20iLCJ2ZXJzaW9uIjoiMi4wIn0=.2fVCZY0aKOkKVwgHf7E4REQMPXDccRDhln3kUbPvAus="

print("Connecting a websocket...")
receivedData = create_connection("wss://localhost:6868", sslopt={"cert_reqs": ssl.CERT_NONE})

#------------------------------------------------------------------------
print("Revisa la conectividad con el auricular")
receivedData.send(json.dumps({
"jsonrpc": "2.0",
"method": "queryHeadsets",
"params": {},
"id": 1
}))

print(receivedData.recv())

#------------------------------------------------------------------------
print("Crear sesion")

receivedData.send(json.dumps({
"id": 1,
"jsonrpc": "2.0",
"method": "createSession",
"params": {
"cortexToken": token,
"headset": "INSIGHT-A1D201D2",
"status": "open"
}}))

#print(receivedData.recv())

idses=json.loads(receivedData.recv())["result"]
print(idses['id'])

#------------------------------------------------------------------------
print("\nSubscripcion de datos")

receivedData.send(json.dumps({
"id": 1,
"jsonrpc": "2.0",
"method": "subscribe",
"params": {
"cortexToken": token,
"session": idses['id'],
"streams": ["com"]
}
}))

While(True) :
          print(receivedData.recv())
In last line of my code it just print the subscription is successfully but i didn't get the mental command like online client

bchaperon, 2020-02-18:
Hi @Hayhat95 
I just tried your source on my computer, it works. I can subscribe to the mental command stream and get the data.
Can you try to subscribe to another stream, like "mot" or "dev"?
And do you see the data in EmotivBCI?

Hayhat95, 2020-02-18:
Ok thanks it worked

closed 2020-02-21 by bchaperon