Emotiv/cortex-example #79: C# Example using Microsoft Visual Studio Express

BrainVictory, 2020-03-20:
I unzip the zip file and I navigate to the "csharp" folder and I open up  the CortexExamples.sln file.  Microsoft Visual Studio 2019 opens up with the project.  I press the start button to see what the example does and I get this message: "There were build errors. Would you like to continue and run the last successful build?"  I didn't make any changes in the code.  I just wanted to run it first to see what the example does before I started personalizing the program.  A couple years ago a wrote a program based off of the SDK that was here on GetHub for the emotiv using C#.  Emotiv is saying that values from the older SDK can't be trusted and so I am trying to see if I can do the same thing with this new Cortex, but I can't seem to get be able to get started.  Why do I get an error as soon as I run the example?  Thanks

bchaperon, 2020-03-26:
Hi @BrainVictory 
What error message do you get during the build?
Did you follow the instructions from the readme file? https://github.com/Emotiv/cortex-v2-example/tree/master/csharp
Especially, did you use the Nuget Package Manager to install Newtonsoft.Json, SuperSocket.ClientEngine.Core, WebSocket4Net for CortexAccess project ?

BrainVictory, 2020-04-08:
As you can see in the picture I have installed Newtonsoft.Json,
SuperSocket.ClientEngine.Core, WebSocket4Net for CortexAccess project and I
am still getting build errors in the CortexAccess project.  Something must
not have taken correctly.

[image: image.png]

Here are all my errors:

Severity

Code

Description

Project

File

Line

Suppression State

Suppression State

Error

CS0246

The type or namespace name 'Newtonsoft' could not be found (are you missing
a using directive or an assembly reference?)

CortexAccess

C:\COrtex\cortex-v2-example-master\cortex-v2-example-master\csharp\CortexAccess\CtxClient.cs

4

Active

Error

CS0246

The type or namespace name 'JArray' could not be found (are you missing a
using directive or an assembly reference?)

CortexAccess

C:\COrtex\cortex-v2-example-master\cortex-v2-example-master\csharp\CortexAccess\CtxClient.cs

19

Active

Error

CS0246

The type or namespace name 'JArray' could not be found (are you missing a
using directive or an assembly reference?)

CortexAccess

C:\COrtex\cortex-v2-example-master\cortex-v2-example-master\csharp\CortexAccess\CtxClient.cs

19

Active

[... 543 lines cut ...]
The referenced component 'Newtonsoft.Json' could not be found.

CortexAccess

Warning

Could not resolve this reference. Could not locate the assembly
"Newtonsoft.Json". Check to make sure the assembly exists on disk. If this
reference is required by your code, you may get compilation errors.

CortexAccess

Warning

The referenced component 'Newtonsoft.Json' could not be found.

BandPowerLogger

Warning

Could not resolve this reference. Could not locate the assembly
"Newtonsoft.Json". Check to make sure the assembly exists on disk. If this
reference is required by your code, you may get compilation errors.

BandPowerLogger

tungntEmotiv, 2020-07-08:
Hi @BrainVictory ,

As your log, the reference to "Newtonsoft.json" is broken. Please try to re-install the lib. You can refer to 
https://stackoverflow.com/questions/40416569/newtonsoft-could-not-be-found/40417243.
After reinstalling the library, please send us the result. Thanks

closed 2020-08-03 by nguoithichkhampha