Emotiv/cortex-example #82: Difficulties connecting to the headset (Python)

nkm4, 2020-03-24:
Hi

I am running your example code and I have difficulties connecting to my emotiv epoc+. I keep seeing this repsonse:

```
InspectApi resp:
{'jsonrpc': '2.0', 'warning': {'code': 11, 'message': ' The APIs cannot be registered. Please re-open the connection.'}}
```

to the: 
`await cortex.inspectApi()`

after which it fails with the:
`await cortex.get_user_login()`

Sometimes it is able to connect when I keep on re-running the code but most of the times it fails. I have approved my App in the emotiv App and I do not have any issues finding the headset there as well..

bchaperon, 2020-03-26:
Hi @nkm4 
Which OS and which version or Cortex do you use?
Also, do you run your Python script as a user process, or a root process?
Basically, this warning happens when Cortex cannot get some information related to the client process.

nkm4, 2020-03-26:
I am using Windows 10 and I have installed the emotiv V2.4.1. I just tried to run the script through an administrator cmd and I still see the same behvaior, however I am not entirely sure that doing so gives it the equivalent admin/root privileges :-)

Could something potentially be blogging it? Trend Micro etc.?

nguoithichkhampha, 2020-07-07:
@nkm4 , Do you still get this error with laster Cortex version ?

nkm4, 2020-07-08:
> @nkm4 , Do you still get this error with laster Cortex version ?

My project with the headset is currently on hold for the time beeing. Thanks for the check up :-) I will let you know if I still have issues when I continue on the project.

nguoithichkhampha, 2020-07-09:
@nkm4 , okay. I will temporary close this ticket. You can reopen if it still happens.

closed 2020-07-09 by nguoithichkhampha