Emotiv/cortex-example #83: Create Session not Working , no object returned

ktladi21, 2020-03-26:
Trying to get  an EEG stream from the Emotiv but when function createSession is called no object is returned from the websocket. Please assist.

<!doctype html>
<html>
<header>
  <title>Emotiv Test</title>
</header>

<body>

  <span id='connection'></span>
  <br />
  <span id='socketStatus'></span>
  <br />
  <span id='response'></span>

</body>

<script>

  // Create Socket
  let socket = new WebSocket("wss://localhost:6868");

  // Socket Comms

  const socketStatus = document.getElementById('socketStatus');

  const response = document.getElementById('response');

  const connection = document.getElementById('connection');

  // get Cortex Info 

  function getCortexInfo() {

    const message = {
      "id": 1,
      "jsonrpc": "2.0",
      "method": "getCortexInfo",

      "params": {}

    }

    const data = JSON.stringify(message);

    // on socket open
    socket.onopen = function (e) {

      connection.innerHTML = 'Socket Open and Sending Data';
      // send data
      socket.send(data);
    };

    // on response
    socket.onmessage = function (event) {

        console.log(event.data);
      response.innerHTML = 'Response: ' + event.data;

    };

    socket.onclose = function (event) {
      if (event.wasClean) {
        alert(`[close] Connection closed cleanly, code=${event.code} reason=${event.reason}`);
      } else {
        // e.g. server process killed or network down
        // event.code is usually 1006 in this case
        alert('[close] Connection died');
      }
    };

    socket.onerror = function (error) {
      alert(`[error] ${error.message}`);
    };

  }

  getCortexInfo();

  // get User Login

  function getUserLogin() {

    const message = {
      "id": 1,
      "jsonrpc": "2.0",
      "method": "getUserLogin",

      "params": {}

    }

    const data = JSON.stringify(message);

    // on socket open
    socket.onopen = function (e) {

      connection.innerHTML = 'Socket Open and Sending Data';
      // send data
      socket.send(data);
    };

    socket.onmessage = function (event) {

      console.log(event.data);
    // on response
      response.innerHTML = 'Response: ' + event.data;

    };

    socket.onclose = function (event) {
      if (event.wasClean) {
        alert(`[close] Connection closed cleanly, code=${event.code} reason=${event.reason}`);
      } else {
        // e.g. server process killed or network down
        // event.code is usually 1006 in this case
        alert('[close] Connection died');
      }
    };

    socket.onerror = function (error) {
      alert(`[error] ${error.message}`);
    };

  }

  getUserLogin();

  // Request Access

  
  function requestAccess() {

    const message = {
      "id": 1,
      "jsonrpc": "2.0",
      "method": "requestAccess",

      "params": {

    "clientId": "4GIDK25Z4Ue55EUpyTu4VRf7c1Wmem1GYdSXwpKa",
    "clientSecret": "StdUbo9LVPW1ObDftTY9yDOPrrmWKA4Y0IEKNzIunVtSjtWWtl2yA9f9vS7eEW5FYvwNBjoVhd674gTciZ4nf8mgLhh8SPopTgQ845DunQws0paFLUUPso786dBn7huO",

      }

    }

    const data = JSON.stringify(message);

    // on socket open
    socket.onopen = function (e) {

      connection.innerHTML = 'Socket Open and Sending Data';
      // send data
      socket.send(data);
    };

    // on response
    socket.onmessage = function (event) {

      console.log(event.data);
      response.innerHTML = 'Response: ' + event.data;

    };

    socket.onclose = function (event) {
      if (event.wasClean) {
        alert(`[close] Connection closed cleanly, code=${event.code} reason=${event.reason}`);
      } else {
        // e.g. server process killed or network down
        // event.code is usually 1006 in this case
        alert('[close] Connection died');
      }
    };

    socket.onerror = function (error) {
      alert(`[error] ${error.message}`);
    };

  }

  requestAccess();

// Authorize

function authorize() {
  

const message = {
  "id": 1,
  "jsonrpc": "2.0",
  "method": "authorize",

  "params": {

"clientId": "4GIDK25Z4Ue55EUpyTu4VRf7c1Wmem1GYdSXwpKa",
"clientSecret": "StdUbo9LVPW1ObDftTY9yDOPrrmWKA4Y0IEKNzIunVtSjtWWtl2yA9f9vS7eEW5FYvwNBjoVhd674gTciZ4nf8mgLhh8SPopTgQ845DunQws0paFLUUPso786dBn7huO",
"debit": 500,

  }

}

const data = JSON.stringify(message);

// on socket open
socket.onopen = function (e) {

  connection.innerHTML = 'Soket Open and Sending Data';
  // send data
  socket.send(data);
};

// on response
socket.onmessage = function (event) {

let cortexToken=JSON.parse(event.data)['result']['cortexToken'];

response.innerHTML='Token:'+cortexToken;

getUserInfo(cortexToken)

}

socket.onclose = function (event) {
  if (event.wasClean) {
    alert(`[close] Connection closed cleanly, code=${event.code} reason=${event.reason}`);
  } else {
    // e.g. server process killed or network down
    // event.code is usually 1006 in this case
    alert('[close] Connection died');
  }
};

socket.onerror = function (error) {
  alert(`[error] ${error.message}`);
};

  }

authorize();

//Create Session
  
  function createSession(Token)
  {

const message = {
  "id": 1,
  "jsonrpc": "2.0",
  "method": "updateSession",

  "params": { 
    "cortexToken": Token,
        "headset": "INSIGHT-A1D20617",
        "status": "open"

  }

}

const data = JSON.stringify(message);

// on socket open
socket.onopen = function (e) {

  connection.innerHTML = 'Socket Open and Sending Dat';
  // send data
  socket.send(data);
};

// on response
socket.onmessage = function (event) {

  let sessionId = JSON.parse(event.data)['result']['id']

  response.innerText(sessionId)
  console.log(sessionId)

 
}

socket.onclose = function (event) {
  if (event.wasClean) {
    alert(`[close] Connection closed cleanly, code=${event.code} reason=${event.reason}`);
  } else {
    // e.g. server process killed or network down
    // event.code is usually 1006 in this case
    alert('[close] Connection died');
  }
};

socket.onerror = function (error) {
  alert(`[error] ${error.message}`);
};

}

  

</script>

</html>

nguoithichkhampha, 2020-07-07:
@ktladi21 , Can you send your sample project code ? the message above very hard to read.

closed 2020-08-03 by nguoithichkhampha

ktladi21, 2020-08-04:
It seems the subscription wasn't paid for by my company.