Emotiv/cortex-example #85: problem with work

areejhameed, 2020-04-19:
hello..
i execute the example with my epoc+ and put my information in ( "license " : "EmotivPROStudent",
	"client_id" and "client_secret" , but it does not work . it connected successfully with headset but an error message had appears with license
how can i fix this ?
best regards

nguoithichkhampha, 2020-04-20:
"license " should be your license key.

closed 2020-04-20 by nguoithichkhampha

areejhameed, 2020-04-20:
Thank youمرسل من هاتف Samsung Galaxy الذكي.
-------- الرسالة الأصلية --------من: Cuong Trinh Manh <notifications@github.com> التاريخ: ٢٠‏/٤‏/٢٠٢٠  ٩:٣٩ ص  (GMT+03:00) إلى: Emotiv/cortex-v2-example <cortex-v2-example@noreply.github.com> نسخة: areejhameed <areej.alanbary@gmail.com>, Author <author@noreply.github.com> الموضوع: Re: [Emotiv/cortex-v2-example] problem with work (#85) 
"license " should be your license key.

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/Emotiv/cortex-v2-example/issues/85#issuecomment-616342815",
"url": "https://github.com/Emotiv/cortex-v2-example/issues/85#issuecomment-616342815",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]