Emotiv/cortex-example #86: errors

areejhameed, 2020-04-19:
hello..
pls. the following errors appeared to me and i don't know how to fix them.
could someone help me to solve them?
thanks
[image]

nguoithichkhampha, 2020-04-20:
@areejhameed . did you define "license" in user object ? did you use our example code or write by yourself ? 
`user = {
	"license" : "your license",
	"client_id" : "your client id",
	"client_secret" : "your client ",
	"debit" : 100,
	"number_row_data" : 10
}`

areejhameed, 2020-04-20:
thank you 

> @areejhameed . did you define "license" in user object ? did you use our example code or write by yourself ?
> `user = { "license" : "your license", "client_id" : "your client id", "client_secret" : "your client ", "debit" : 100, "number_row_data" : 10 }`

thank you ,
I used your code and put my personal license information in the user field above

closed 2020-04-20 by nguoithichkhampha