Emotiv/cortex-example #87: "license " should be your license key.

areejhameed, 2020-04-20:
"license " should be your license key.

_Originally posted by @nguoithichkhampha in https://github.com/Emotiv/cortex-v2-example/issues/85#issuecomment-616342815_
I useed my license key but it didn't  work!

nguoithichkhampha, 2020-04-20:
pls show me the error and can you send me the details including your license key via my email cuong@emotiv.com

areejhameed, 2020-04-20:
> pls show me the error and can you send me the details including your license key via my email [cuong@emotiv.com](mailto:cuong@emotiv.com)

[image]
thank you , i will send you the required info.

nguoithichkhampha, 2020-07-07:
@areejhameed , I don't see any email from you. So, I will close this ticket.

closed 2020-07-07 by nguoithichkhampha

areejhameed, 2021-07-25:
hi..
Please, is there an eeg logger example in python to record the eeg signals
to matrix form instead of recording them on an external excel file ?
thanks
best regards

‫في الثلاثاء، 7 يوليو 2020 في 6:37 م تمت كتابة ما يلي بواسطة ‪Cuong Trinh
Manh‬‏ ***@***.***‬‏>:‬