Emotiv/cortex-example #93: createSession throws code: -32004 "The headset is currently not available"

armandokun, 2020-06-08:
Dear Community,

I have been investigating this problem for a while now and I can't understand what's been causing the issue.

When I run queryHeadsets function it returns a normal infographic with the status: 'connected' but after I initialise createSession or controlDevice functions - an error message pops up:
`{"error":{"code":-32004,"message":"The headset is currently not available."},"id":3,"jsonrpc":"2.0"}`

Let me know how can I help to resolve this issue, thanks.

**queryHeadsets response:** 
`{"id":2,"jsonrpc":"2.0","result":[{"connectedBy":"dongle","customName":"","dongle":"6ff","firmware":"634","headbandPosition":null,"id":"EPOCPLUS-4A2xxxx","motionSensors":["Q0","Q1","Q2","Q3","ACCX","ACCY","ACCZ","MAGX","MAGY","MAGZ"],"sensors":["AF3","F7","F3","FC5","T7","P7","O1","O2","P8","T8","FC6","F4","F8","AF4"],"settings":{"eegRate":128,"eegRes":16,"memsRate":0,"memsRes":16,"mode":"EPOCPLUS"},"status":"connected"}]}
`

nguoithichkhampha, 2020-07-07:
@armandokun , seems the signal between your headset and usb dongle is not stable. So, when you call "createSession", Cortex detects headset lost connection and return error. Pls put your headset close with dongle and try again.

closed 2020-07-07 by nguoithichkhampha

armandokun, 2020-07-08:
The headset is as close as possible to the USB dongle, still nothing has changed.