Emotiv/cortex-example #94: cannot find headset from mobile example

KhangSoCute, 2020-06-09:
i am trying the cortex api in android through mobile example. i filled client_id, client_secret and even license_id  in Constant.java. i an connect to emotiv app in android, but when open example band power, it is any things in the list view Headset Name. Do you have any instruction in connect headset to cortex through mobile example.

nguoithichkhampha, 2020-06-09:
Cortex mobile still in beta testing. did you install Emotiv app on Android ?

KhangSoCute, 2020-06-09:
yes, i installed and could find the headset by emotiv app

nguoithichkhampha, 2020-06-09:
can you share the your emotiv id and the email to register testing program ?

KhangSoCute, 2020-06-09:
my emotiv id is tknhatkhang
the email is tknhatkhang@gmail.com

nguoithichkhampha, 2020-06-09:
may you was confused the name of **Emotiv app**.
there is emotiv app you are using ?
https://play.google.com/store/apps/details?id=com.emotiv.insightapp

KhangSoCute, 2020-06-09:
yes, i used it to connect the headset, was it right ?

nguoithichkhampha, 2020-06-09:
this is app use old sdk. it is not designed to work with cortex example. if you want to try Cortex mobile, pls send request to mail  hello@emotiv.com

closed 2020-06-09 by nguoithichkhampha