Emotiv/cortex-example #98: Progress bar does not stop

guneteroglu, 2020-07-09:
Hello,
I have registered atbotizm under hms2017 userid. I have created client id and client secret key
I have downloaded the cortex API from github and and inserted these
into Constant.java file.
I have compiled the code (bandpower) and run app on Android mobile, I can only see progress bar, nothing else.

How can I connect to a headset?
[MainActivity.zip](https://github.com/Emotiv/cortex-v2-example/files/4896153/MainActivity.zip)

nguoithichkhampha, 2020-07-09:
@guneteroglu , did you use our mobile example ?
if yes, you need to be our beta tester to work with mobile example.

guneteroglu, 2020-07-09:
Yes, I have downloaded it from GitHub as Emy suggested.
How can I become a beta tester?

Günet Eroğlu

CEO

HMS Health Mobile Software A,Ş

İçerenköy Mah. Kayışdağı Cad. No:32
Acıbadem Üniversitesi Kuluçka Merkezi 34752
Ataşehir İSTANBUL
+90 532 326 55 82
gunet.eroglu@healthmobilesoftware.com

 https://www.autotrainbrain.com/

nguoithichkhampha, 2020-07-09:
I see your email already in our tester list. Did you install Emotiv App on Android ?

guneteroglu, 2020-07-09:
Yes, I have installed emotiv app on Android, and this app can connect  to EPOC+

emoviet, 2020-07-09:
@guneteroglu I have just checked, it still works fine! . So, please check your CLIENT_ID and CLIENT_SECRET.

guneteroglu, 2020-07-09:
Hi, after a couple of trial, reloading EMOTIV app, and accepting the connection from my app, I have succeeded. There are a couple of questions remained. EMOTIV app requires our user id/pass. This is normally our company's user id. When we adapt our app with cortex api, we will be sharing app with the end users. So, the end users should download the EMOTIV app and have our user id to login. This is not a sustainable approach. 

Second question is that how can I retrieve band power data after I subscribe the data? Do you have a code example for that.
Best regards

emoviet, 2020-07-09:
Hi @guneteroglu 
about #1 question, you can take a look at this file ...cortexv2example/controller/SysController.java with this function:
public void parseSysMessage(String msg)
here you can add a state
` if (sysJson.has("pow")) {
    // log msg (band power data) here
 }`

about #2 question, I think @patemotiv can help to answer :)

guneteroglu, 2020-07-09:
Hi, Thanks a lot for the reply. I have checked SysController.java file, it does not contain an entry for pow.

package com.emotiv.cortexv2example.controller;

import com.emotiv.cortexv2example.interfaces.SysInterface;
import org.json.JSONArray;
import org.json.JSONObject;

public class SysController {
    public static SysController sysController;
    private SysInterface mSysInterface;
    public SysController() {

    }

    public static SysController getInstance() {
        if (sysController == null) {
            sysController = new SysController();
        }
        return sysController;
    }

    public void setSysInterface(SysInterface sysInterface) {
        this.mSysInterface = sysInterface;
    }

    // handle the response to a RPC request
    public void parseSysMessage(String msg) {
        try {
            JSONObject sysJson = new JSONObject(msg);
            if (mSysInterface != null) {
                if (sysJson.has("sys")) {
                    JSONArray sysContentJson = sysJson.getJSONArray("sys");
                    CortexClientController.getInstance().getDataStream().getmTrainingObject().setCurrentDetection(sysContentJson.get(0).toString());
                    CortexClientController.getInstance().getDataStream().getmTrainingObject().setCurrentTrainingEvent(sysContentJson.get(1).toString());

                    if (sysContentJson.get(1).toString().contains("Started")) {
                        mSysInterface.onTrainingStarted(sysContentJson.get(0).toString(), CortexClientController.getInstance().getDataStream().getmTrainingObject().getTrainingAction(), sysContentJson.get(1).toString());
                    } else if (sysContentJson.get(1).toString().contains("Failed")) {
                        mSysInterface.onTrainingFailed();
                    } else if (sysContentJson.get(1).toString().contains("Succeeded")) {
                        mSysInterface.onTrainingSucceeded();
                    } else if (sysContentJson.get(1).toString().contains("Completed")) {
                        mSysInterface.onTrainingCompleted();
                    } else if (sysContentJson.get(1).toString().contains("Rejected")) {
                        mSysInterface.onTrainingRejected();
                    } else if (sysContentJson.get(1).toString().contains("Reset")) {
                        mSysInterface.onTrainingReset();
                    }

                } else if (sysJson.has("eeg")) {
                    mSysInterface.onEegDataRecieved(msg);
                } else if (sysJson.has("mot")) {
                    mSysInterface.onMotDataRecieved(msg);
                } else if (sysJson.has("dev")) {
                    mSysInterface.onDevDataRecieved(msg);
                } else if (sysJson.has("met")) {
                    mSysInterface.onMetDataRecieved(msg);
                } else if (sysJson.has("com")) {
                    mSysInterface.onComDataRecieved(msg);
                } else if (sysJson.has("fac")) {
                    mSysInterface.onFacDataRecieved(msg);
                }
            }
        } catch (Exception e) { e.printStackTrace(); }
    }

}

emoviet, 2020-07-09:
I have updated the answer, please have a look above

guneteroglu, 2020-07-09:
Thank you very much for the response.  I have found that line.
@Override public void onRecievedNewData(String msg) { Log.i(TAG, "new data: " + msg); handleData(msg); }
Could you tell me under which function should I call handleData in MainActivity.java (bandpower app).
Just after subscribe to data, or a trigger is set when I receive any data.

emoviet, 2020-07-09:
you should:
1. add a abstract function on SysInterface.java, such as `void onBandPowerDataRecieved(String bandPowerData);`
2. implement SysInterface in MainActivity.java of band power module.
3. in SysController.java
add statement `if (sysJson.has("pow")) {
                    mSysInterface.onBandPowerDataRecieved(msg);
                }`

guneteroglu, 2020-07-09:
OK. I have implemented what you have requested. Still I didn't understand how can I retrieve json data from MainActivity.java
after I subscribed to Band Power.
The logical place seem to be here:
    @Override
    public void onSubscribeDataOk() {

    }
But how?

emoviet, 2020-07-09:
did you have a look at SysInterface.java?

package com.emotiv.cortexv2example.interfaces;

public abstract interface SysInterface {
    // received data from a data stream
    void onEegDataRecieved(String eEgData);
    void onMotDataRecieved(String motData);
    void onDevDataRecieved(String devData);
    void onMetDataRecieved(String metData);
    void onComDataRecieved(String comData);
    void onFacDataRecieved(String facData);
    void onTrainingStarted(String detection, String action, String event);
    void onTrainingFailed();
    void onTrainingSucceeded();
    void onTrainingCompleted();
    void onTrainingRejected();
    void onTrainingReset();
}

You must add an signal here, such as `void onBandPowerDataRecieved(String bandPowerData);
then implements SysInterface in MainActivity.java of band power module.
and in SysController.java
add statement 
...
else if (sysJson.has("pow")) {
    mSysInterface.onBandPowerDataRecieved(msg);
}

Now you can retrieve message of bandpower data on MainActivity of band power

guneteroglu, 2020-07-09:
new data: {"pow":[0.696,0.119,0.144,0.518,0.666,0.035,0.084,0.07,0.089,0.554,0.023,0.037,0.029,0.035,0.152,0.043,0.012,0.019,0.03,0.202,0.053,0.062,0.057,0.13,0.297,0.066,0.017,0.013,0.061,0.119,0.043,0.055,0.09,0.275,0.223,0.033,0.037,0.046,0.03,0.025,0.015,0.013,0.032,0.044,0.036,0.023,0.013,0.02,0.021,0.051,0.078,0.021,0.023,0.049,0.414,0.069,0.085,0.184,0.427,0.495,0.226,0.103,0.224,1.624,0.504,0.684,0.159,0.067,0.87,1.042],
Yes, I have received the data with success. What is the format of the data? How will I know it from which channel, which frequency band?

emoviet, 2020-07-09:
I believe after subscribe data you will get message with "success" array of stream, plz have a look at logcat and find streamName":"pow"

emoviet, 2020-07-09:
@guneteroglu here are docs for you
https://emotiv.gitbook.io/cortex-api/data-subscription/data-sample-object

patemotiv, 2020-07-09:
hi @guneteroglu 

> EMOTIV app requires our user id/pass. This is normally our company's user id. When we adapt our app with cortex api, we will be sharing app with the end users. So, the end users should download the EMOTIV app and have our user id to login. This is not a sustainable approach.

The end users should be using their own EmotivID instead of your own one. You need to obtain an Open Source Application License (https://www.emotiv.com/contact-free-application) or a Commercial Application License (https://www.emotiv.com/contact-commercial-application) before you can distribute your applications to others.

Please check out more info here https://www.emotiv.com/developer/ or contact our customer support team at hello@emotiv.com.

Thanks!

guneteroglu, 2020-07-09:
Hello,
The data I received at logcat comes from CortexClientController, not from SysInterface I am afraid.

emoviet, 2020-07-10:
hi @guneteroglu about how to get these data, I wrote above. Have you read it? Have you tried to implement yet? In the process of writing code, if there are any difficulties, please ask.

guneteroglu, 2020-07-10:
Yes. I have implemented. I have attached the codes. Please comment on the codes that I have updated. Still, the pow data in the logcat comes from CortexClientController.
Best,
[code.zip](https://github.com/Emotiv/cortex-v2-example/files/4901041/code.zip)

emoviet, 2020-07-10:
Looking at your code, when implementing SysInterface on MainActivity.java you forgot to set SysInterface listener
please add this into onResume() of MainActivity.java
`SysController.getInstance().setSysInterface(this);
`

guneteroglu, 2020-07-10:
Ok. That has solved the issue. Can I combine Create Session and subscribe Band power into one button/action.

emoviet, 2020-07-10:
that depends on your dicision, I think I can just support this ticket until here

guneteroglu, 2020-07-10:
Thank you for the support. All the best.

closed 2020-07-10 by guneteroglu

emoviet, 2020-07-10:
you're welcome!