Emotiv/cortex-example #99: The headset is as close as possible to the USB dongle, still nothing has changed.

armandokun, 2020-07-09:
Dear Community,

I have been investigating this problem for a while now and I can't understand what's been causing the issue.

When I run queryHeadsets function it returns a normal infographic with the status: 'connected' but after I initialise createSession function - an error message pops up:
`{"error":{"code":-32019,"message":"Session limit on this device has been reached."},"id":5,"jsonrpc":"2.0"}`

Let me know how can I help to resolve this issue, thanks.

P.S. There were no sessions created before. querySessions returns an empty array: `{"id":10,"jsonrpc":"2.0","result":[]}`

queryHeadsets response:
{"id":2,"jsonrpc":"2.0","result":[{"connectedBy":"dongle","customName":"","dongle":"6ff","firmware":"634","headbandPosition":null,"id":"EPOCPLUS-4A2xxxx","motionSensors":["Q0","Q1","Q2","Q3","ACCX","ACCY","ACCZ","MAGX","MAGY","MAGZ"],"sensors":["AF3","F7","F3","FC5","T7","P7","O1","O2","P8","T8","FC6","F4","F8","AF4"],"settings":{"eegRate":128,"eegRes":16,"memsRate":0,"memsRes":16,"mode":"EPOCPLUS"},"status":"connected"}]}

**Update:** I have tried to connect the headset via bluetooth as well, still having the same error.

bchaperon, 2020-07-10:
Hello @armandokun 

This error is related to your EMOTIV license. There is nothing wrong with your headset.

The error code -32019 means that you tried to activate a session when the “localQuota” of your license is zero. You can check the “localQuota” by calling the method “getLicenseInfo”. 
[https://emotiv.gitbook.io/cortex-api/authentication/getlicenseinfo](https://emotiv.gitbook.io/cortex-api/authentication/getlicenseinfo)

To solve this error, you must increase this local quota. You do this by setting the parameter “debit” of the method “authorize”. For example, you can call “authorize” with a “debit” of 50, and this will increase your local quota by 50. Then you will be able to create and activate 50 sessions on your computer.
https://emotiv.gitbook.io/cortex-api/authentication/authorize

armandokun, 2020-07-10:
Thanks, it works now!

closed 2020-07-10 by armandokun