Emotiv/cortex-example #100: Combining create session with subscribe

guneteroglu, 2020-07-10:
In the band-power Cortex API interface, I have combined create session and subscribe actions as follows:

          case R.id.btnCreateSession:
                CortexClient.getInstance().createSession(cortexToken, "active", HeadsetObject.getInstance().getHeadsetName());
                break;
            case R.id.btnSubscribeBP:
                CortexClient.getInstance().createSession(cortexToken, "active", HeadsetObject.getInstance().getHeadsetName());
                CortexClient.getInstance().subscribeData(cortexToken, SessionObject.getInstance().getCurrentActivedSession(), streamArray);
                break;
            case R.id.btnUnSubscribeBP:
                CortexClient.getInstance().unSubscribeData(cortexToken, SessionObject.getInstance().getCurrentActivedSession(), streamArray);
                break;
     
The result is that when I click the subscribe button once, it creates session, I have to click once again to subscribe the data.
Can you explain how I can effectively combine Create Session and Subscribe data buttons.

nguoithichkhampha, 2020-07-10:
it depends on your code design, the idea is get response of request `createSession` from Cortex and call `subscribe`
our example only simple code to explain the step work with Cortex.

closed 2020-07-10 by nguoithichkhampha

guneteroglu, 2020-07-10:
The question is that when i call subscribe right after create session, only
create session is executed. i have to click subscribe button once again. If
you run the code you will understand what i mean. I guess there needs to be
a delay between two operations.

şunu yazdı:

bchaperon, 2020-07-13:
Hi @guneteroglu 

In order to call "subscribe", you must have a session id, so you must wait for the result of the method "createSession". 
So it doesn't make sense to combine these 2 methods.