Emotiv/cortex-example #106: Invalid Client Credentials

122671644, 2020-08-02:
[image]
[image]

I have applied for the ID and password on the website. Why is the display invalid? My license will expire in about two years

122671644, 2020-08-02:
> [image]
> [image]
> 
> I have applied for the ID and password on the website. Why is the display invalid? My license will expire in about two years

"D:\python inter\python.exe" D:/EPOC/time/record.py
query headset --------------------------------
EPOCPLUS-4A2C0521
connect headset --------------------------------
connect headset result {
    "id": 2,
    "jsonrpc": "2.0",
    "result": {
        "command": "connect",
        "message": "The headset EPOCPLUS-4A2C0521 has been connected or is connecting."
    }
}
request access --------------------------------
{
    "error": {
        "code": -32021,
        "message": "Invalid Client Credentials."
    },
    "id": 3,
    "jsonrpc": "2.0"
}
authorize --------------------------------
auth request 
 {
    "jsonrpc": "2.0",
    "method": "authorize",
    "params": {
        "clientId": "your client id",
        "clientSecret": "your client secret",
        "license": "your emotivpro license, which could use for third party app",
        "debit": 100
    },
    "id": 4
}
auth result 
 {
    "error": {
        "code": -32102,
        "message": "You have no access rights to the Application."
    },
    "id": 4,
    "jsonrpc": "2.0"
}
Traceback (most recent call last):
  File "D:/EPOC/time/record.py", line 63, in <module>
    r = Record()
  File "D:/EPOC/time/record.py", line 7, in __init__
    self.c.do_prepare_steps()
  File "D:\EPOC\time\cortex.py", line 190, in do_prepare_steps
    self.authorize()
  File "D:\EPOC\time\cortex.py", line 122, in authorize
    self.auth = result_dic['result']['cortexToken']
KeyError: 'result'

Process finished with exit code 1

nguoithichkhampha, 2020-08-03:
`"message": "You have no access rights to the Application."`
you have to accept request from Emotiv app first.
https://emotiv.gitbook.io/cortex-api/authentication/requestaccess

closed 2020-08-09 by nguoithichkhampha