Emotiv/cortex-example #108: Deleting command

skshreyas4, 2020-08-04:
I tried to delete the facial expression by using "erase" status and I even made all the edits suggested by Emotiv support team but it is not working can you please help me by either sending me flowchart or steps or in any way

nguoithichkhampha, 2020-08-04:
are you working on training profile for facial expression ?
which API did you call ?

skshreyas4, 2020-08-05:
yes i am ,everything is working fine except the delete command,
i am using the same method as it was shown in gitbook ,that is training method in which I changed the status to "erase"
code is
this will be in cortex.py
 def delete(self, detection, action, status):
        TRAINING_ID = 9
        train_request_json = {
            "jsonrpc": "2.0",
            "method": "training",
            "params": {
                "cortexToken": self.auth,
                "detection": detection,
                "session": self.session_id,
                "action": action,
                "status": status
            },
            "id": TRAINING_ID
        }
from another file i am calling
def delete_command(self):
          self.cortex = Cortex(url, user, debug_mode=True)
        self.headset_id = self.cortex.query_headset()
        self.cortex.connect_headset()
        self.cortex.request_access()
        auth = self.cortex.authorize()
        self.cortex.create_session(auth, self.headset_id)
        profile_name = 'abc'
        profiles = self.cortex.queryProfile()
        status = 'load'
        self.cortex.setup_profile(profile_name, status)
        self.cortex.delete(detection='facialExpression',action='smile',status='erase')
        status = "save"
        self.cortex.setup_profile(profile_name, status)
        status = 'unload'
        self.cortex.setup_profile(profile_name, status)

nguoithichkhampha, 2020-08-05:
How to you know call api with status "erase" not work ?

skshreyas4, 2020-08-05:
> 
> 
> How to you know call api with status "erase" not work ?
I am using the same profile which I have in EmotivBCI application,so when i delete the command there should be a change in Emotiv BCI application's training count, that is how I thought about it,am I right or wrong about it?

nguoithichkhampha, 2020-08-06:
did you open BCI and your app at the same time ? 
you should load profile again in BCI after delete command and save profile from your app.

skshreyas4, 2020-08-06:
no I did not, Once after I save it to profile and unload my profile,Later I used to see whether it has been updated in the EmotivBCI app or not, you can see my code which I save sent before I have saved and unloaded as well.

nguoithichkhampha, 2020-08-06:
Can you test again and show me BCI screen after reload profile ?

bchaperon, 2020-08-06:
Hi @skshreyas4 
After you call the method "training", do you receive an error message from Cortex, or a result message?
Can you show us the JSON messages that you send and receive from Cortex?

skshreyas4, 2020-08-09:
no i don't receive any error messages and I get JSON messages as training successful
response message is :
`result 
 {
    "id": 7,
    "jsonrpc": "2.0",
    "result": {
        "action": "load",
        "message": "The profile is loaded successfully",
        "name": "skshreyas"
    }
}

command erase initiated

command erase done
 {'id': 9, 'jsonrpc': '2.0', 'result': {'action': 'smile', 'message': 'Set up training successfully', 'status': 'erase'}}
setup profile --------------------------------
setup profile json:
 {
    "jsonrpc": "2.0",
    "method": "setupProfile",
    "params": {
        "cortexToken": "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhcHBJZCI6ImNvbS5za3NocmV5YXM0LmJjb3QiLCJhcHBWZXJzaW9uIjoiMS4wIiwiZXhwIjoxNTk3MTE1MjQ2LCJuYmYiOjE1OTY4NTYwNDYsInVzZXJJZCI6IjMwYWVmYzI5LTE3YTItNGY2Yi1iMTNkLWI4NWY2ODg1ODY0NSIsInVzZXJuYW1lIjoic2tzaHJleWFzNCIsInZlcnNpb24iOiIyLjAifQ==.e0NlQbNLVDXCpX17IDL4xFQf/G1ABRmhfMKsmg1kj/0=",
        "headset": "INSIGHT-A2D20095",
        "profile": "skshreyas",
        "status": "save"
    },
    "id": 7
}

result 
 {
    "id": 7,
    "jsonrpc": "2.0",
    "result": {
        "action": "save",
        "message": "The profile is saved successfully",
        "name": "skshreyas"
    }
}

setup profile --------------------------------
setup profile json:
 {
    "jsonrpc": "2.0",
    "method": "setupProfile",
    "params": {
        "cortexToken": "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhcHBJZCI6ImNvbS5za3NocmV5YXM0LmJjb3QiLCJhcHBWZXJzaW9uIjoiMS4wIiwiZXhwIjoxNTk3MTE1MjQ2LCJuYmYiOjE1OTY4NTYwNDYsInVzZXJJZCI6IjMwYWVmYzI5LTE3YTItNGY2Yi1iMTNkLWI4NWY2ODg1ODY0NSIsInVzZXJuYW1lIjoic2tzaHJleWFzNCIsInZlcnNpb24iOiIyLjAifQ==.e0NlQbNLVDXCpX17IDL4xFQf/G1ABRmhfMKsmg1kj/0=",
        "headset": "INSIGHT-A2D20095",
        "profile": "skshreyas",
        "status": "unload"
    },
    "id": 7
}

result 
 {
    "id": 7,
    "jsonrpc": "2.0",
    "result": {
        "action": "unload",
        "headsetId": "INSIGHT-A2D20095",
        "message": "Profile unloaded successfully for headset INSIGHT-A2D20095"
    }
}
`

bchaperon, 2020-09-08:
Hi @skshreyas4 
In theory, it should work. The only strange thing I see in your JSON messages is that you re-use the same id several times. But that should not be a problem.
Is it possible for you to share the source code of your application with me, so I can try to debug it on my computer?

skshreyas4, 2020-09-10:
hi @bchaperon 
yes it should but I don not know why it is not, you can see source code-[https://github.com/skshreyas4/BCOT-Brain-computer-of-Things-]

nguoithichkhampha, 2020-09-22:
any progress for this issue @bchaperon

bchaperon, 2020-10-14:
Hi @skshreyas4 

I don't see any issue in the Cortex API, so I suspect there is a bug in your application. 
I tried to run your app in my Mac but I get an error. It seems that Apple uses an old version of the Tk library, and it is not compatible with your app.
```
~/git/BCOT-Brain-computer-of-Things-$ /usr/local/Cellar/python\@3.8/3.8.5/bin/python3 MainNavigator.py 
DEPRECATION WARNING: The system version of Tk is deprecated and may be removed in a future release. Please don't rely on it. Set TK_SILENCE_DEPRECATION=1 to suppress this warning.
Traceback (most recent call last):
  File "MainNavigator.py", line 205, in <module>
    app = SampleApp()
  File "MainNavigator.py", line 35, in __init__
    self.show_frame("FacialExpression")
  File "MainNavigator.py", line 46, in show_frame
    frame = frame_class(parent=self.container, controller=self)
  File "/Users/benoit/git/BCOT-Brain-computer-of-Things-/facial_expression_Train.py", line 14, in __init__
    self.back = PhotoImage(file='images/arrow.png')
  File "/usr/local/Cellar/python@3.8/3.8.5/Frameworks/Python.framework/Versions/3.8/lib/python3.8/tkinter/__init__.py", line 4061, in __init__
    Image.__init__(self, 'photo', name, cnf, master, **kw)
  File "/usr/local/Cellar/python@3.8/3.8.5/Frameworks/Python.framework/Versions/3.8/lib/python3.8/tkinter/__init__.py", line 4006, in __init__
    self.tk.call(('image', 'create', imgtype, name,) + options)
_tkinter.TclError: couldn't recognize data in image file "images/arrow.png"
~/git/BCOT-Brain-computer-of-Things-$
``` 

To debug your app, I recommend you to trace ALL the json messages that you send and receive from Cortex.
FYI here are the messages sent and received by Emotiv BCI when deleting a FE action:
```
{"id":6670514,"jsonrpc":"2.0","method":"setupProfile","params":{"cortexToken":"XXX","headset":"INSIGHT-1EBBB030","profile":"Test delete FE","status":"load"}}
{"id":6670514,"jsonrpc":"2.0","result":{"action":"load","message":"The profile is loaded successfully","name":"Test delete FE"}}

{"id":400417,"jsonrpc":"2.0","method":"training","params":{"action":"smile","cortexToken":"XXX","detection":"facialExpression","session":"4d3643e3-8023-425a-8840-cc309675a07b","status":"erase"}}
{"id":400417,"jsonrpc":"2.0","result":{"action":"smile","message":"Set up training successfully","status":"erase"}}

{"id":8130214,"jsonrpc":"2.0","method":"setupProfile","params":{"cortexToken":"XXX","headset":"INSIGHT-1EBBB030","profile":"Test delete FE","status":"save"}}
{"id":8130214,"jsonrpc":"2.0","result":{"action":"save","message":"The profile is saved successfully","name":"Test delete FE"}}

{"id":8790614,"jsonrpc":"2.0","method":"setupProfile","params":{"cortexToken":"XXX","headset":"INSIGHT-1EBBB030","profile":"Test delete FE","status":"unload"}}
{"id":8790614,"jsonrpc":"2.0","result":{"action":"unload","headsetId":"INSIGHT-1EBBB030","message":"Profile unloaded successfully for headset INSIGHT-1EBBB030"}}
```
And your app can call the API `getTrainedSignatureActions` to check the actions of a profile.
You can call it before and after you delete the action and compare the result.
https://emotiv.gitbook.io/cortex-api/advanced-bci/gettrainedsignatureactions

You can also check the Cortex log files. 
On macOS the log folder is `/Users/benoit/Library/Application Support/Emotiv/Cortex/v2/logs/CortexService`
On Windows, it should be `C:/ProgramData/Emotiv/Cortex/v2/logs/CortexService`

The log file will tell you how many FE actions there are in your profile when Cortex saves it to the disk.
Look for lines like these:
```
[2020-10-13T11:19:05.268+07:00][info   ][7000051d8000 ProfileAttWriteThread] [att_store] Profile saved "803c0781-53ed-45ae-bcd0-50954ee5f701" 182
[2020-10-13T11:19:05.268+07:00][info   ][7000051d8000 ProfileAttWriteThread] [att_store] MC actions QJsonValue(object, QJsonObject({"cols":0,"elements":[],"rows":0}))
[2020-10-13T11:19:05.268+07:00][info   ][7000051d8000 ProfileAttWriteThread] [att_store] MC times   QJsonValue(object, QJsonObject({"cols":14,"elements":[0,0,0,0,0,0,0,0,0,0,0,0,0,0],"rows":1}))
[2020-10-13T11:19:05.268+07:00][info   ][7000051d8000 ProfileAttWriteThread] [att_store] FE actions QJsonValue(object, QJsonObject({"cols":2,"elements":[1,128],"rows":1}))
[2020-10-13T11:19:05.268+07:00][info   ][7000051d8000 ProfileAttWriteThread] [att_store] FE times   QJsonValue(object, QJsonObject({"cols":8,"elements":[2,0,2,0,0,0,0,0],"rows":1}))
```
On the 4th line, you can see `"elements":[1,128]`. This means that the profile contains 2 FE actions. The 1 is for neutral and the 128 is for smile.
On the 5th line, you can see `"elements":[2,0,2,0,0,0,0,0]`. This is the number of training for each action . Here I have 2 trainings for neutral, and 2 for smile.

After you delete the training for smile, you should see something like this:
```
[2020-10-13T11:19:56.962+07:00][info   ][7000051d8000 ProfileAttWriteThread] [att_store] Profile saved "803c0781-53ed-45ae-bcd0-50954ee5f701" 182
[2020-10-13T11:19:56.963+07:00][info   ][7000051d8000 ProfileAttWriteThread] [att_store] MC actions QJsonValue(object, QJsonObject({"cols":0,"elements":[],"rows":0}))
[2020-10-13T11:19:56.963+07:00][info   ][7000051d8000 ProfileAttWriteThread] [att_store] MC times   QJsonValue(object, QJsonObject({"cols":14,"elements":[0,0,0,0,0,0,0,0,0,0,0,0,0,0],"rows":1}))
[2020-10-13T11:19:56.963+07:00][info   ][7000051d8000 ProfileAttWriteThread] [att_store] FE actions QJsonValue(object, QJsonObject({"cols":1,"elements":[1],"rows":1}))
[2020-10-13T11:19:56.963+07:00][info   ][7000051d8000 ProfileAttWriteThread] [att_store] FE times   QJsonValue(object, QJsonObject({"cols":8,"elements":[2,0,0,0,0,0,0,0],"rows":1}))
```

skshreyas4, 2020-10-14:
thank you @bchaperon let me check this out

closed 2020-11-05 by nguoithichkhampha