Emotiv/cortex-example #109: extracting eeg from headset to program

areejhameed, 2020-08-05:
in my program I want to extract the eeg signal online (direct to my program to process it  without saving it to file and reading from that file ) , how can I do this step with python spider environment?
thank you very much

nguoithichkhampha, 2020-08-05:
we have example
https://github.com/Emotiv/cortex-v2-example/blob/master/python/sub_data.py
it will print eeg data in the console log.

areejhameed, 2020-08-05:
Thank you very muchمرسل من هاتف Samsung Galaxy الذكي.
-------- الرسالة الأصلية --------من: Cuong Trinh Manh <notifications@github.com> التاريخ: ٥‏/٨‏/٢٠٢٠  ٨:١٧ م  (GMT+03:00) إلى: Emotiv/cortex-v2-example <cortex-v2-example@noreply.github.com> نسخة: areejhameed <areej.alanbary@gmail.com>, Author <author@noreply.github.com> الموضوع: Re: [Emotiv/cortex-v2-example] extracting eeg from headset to program (#109) 
we have example
https://github.com/Emotiv/cortex-v2-example/blob/master/python/sub_data.py
it will print eeg data in the console log.

closed 2020-08-06 by nguoithichkhampha