Emotiv/cortex-example #112: WinError 10061 affecting other Emotiv Applications

blameitonkarly, 2020-08-11:
When running the Python or C# sample code I receive the following error:

ConnectionRefuseError: [WinError 10061] No connection could be made because the target machine actively refused it. 

After researching online forums, I found out this error message means the code is requesting access to the server but the Server is not recognizing the device and is denying access. The information online said this was a problem with the Server code not the senders code. 

After running the example code, all of my Emotiv applications stopped connecting to the Server as well.  The Emotiv Installer, EmoticBCI and EmotivPro app are stuck on the "Connecting to Server" splash screen. I can only assume this is connected to the WinError 10061, I am receiving from my script codes. 

Why is my python/C# code getting this error message, is there a solution?
Why is it effecting my other Emotiv Applications?
Most importantly, is there a solutions to fix my Emotiv Applications so I can continue working/training?

nguoithichkhampha, 2020-08-12:
@blameitonkarly , do you work on Wins. Pls send the all the log file in path
C:\ProgramData\Emotiv\Cortex
C:\Users\Tu\AppData\Local\Emotiv

blameitonkarly, 2020-08-12:
Yes, I work on Windows 10.
I was not able to fine either of those directories. My computers does not have a ProgramData or a AppData file. 
Could these files been deleted which caused my problem? If so how can I re-download them.

bchaperon, 2020-08-14:
Hi @blameitonkarly 

I think that by default, the folder C:\ProgramData is hidden.
So you need to ask Windows to show hidden files and folders.
This article should help you: 
https://support.microsoft.com/en-us/help/4028316/windows-view-hidden-files-and-folders-in-windows-10

> Most importantly, is there a solutions to fix my Emotiv Applications so I can continue working/training?

You can ask Windows to restart the service named "EMOTIV Cortex" 
This article show you how:
https://kb.wisc.edu/helpdesk/page.php?id=36667
If this doesn't work, then you should restart your computer.

nguoithichkhampha, 2020-09-22:
@blameitonkarly , I don't get your response, so I temporary close this issue. feel free to reopen it.

closed 2020-09-22 by nguoithichkhampha