Emotiv/cortex-example #113: Socket error: QAbstractSocket::UnsupportedSocketOperationError

gubeifang, 2020-08-12:
CPP-QT，QT5.14+VS2017，I filled in the ClientId, ClientSecret and HeadsetId according to the case prompts, and then an error occurred during the operation. Is there any way to solve this problem? thank you very much！
[image]

bchaperon, 2020-08-14:
Hi @gubeifang 

When you start EmotivPRO or EmotivBCI, do they successfully connect to Cortex?

The Qt documentation says that this error can be related to IPv6.
Do you use IPv6 on your computer?
https://doc.qt.io/qt-5/qabstractsocket.html#SocketError-enum

In the file CortexClient.cpp at line 62, can you try to replace this line:
`socket.open(QUrl("wss://localhost:6868"));`
by this:
`socket.open(QUrl("wss://127.0.0.1:6868"));`

gubeifang, 2020-08-14:
Hi
Thank you for your reply.
When i start EmotivPRO or EmotivBCI, they can successfully connect to Cortex

https://doc.qt.io/qt-5/qabstractsocket.html#SocketError-enum

gubeifang, 2020-08-14:
Hi
Thanks for  your reply.
When i start EmotivPRO or EmotivBCI,  they can successfully connect to Cortex ，But when I run the case code,EmotivPRO or EmotivBCI does not respond.
I don't use IPv6 on my computer,IPv6 on my computer is the original setting 
I used another computer to run this case code, and the same situation occurred
I replace 62 line by socket.open(QUrl("wss://127.0.0.1:6868"));and the same situation occurred

Qt keeps showing ：Socket error: QAbstractSocket::UnsupportedSocketOperationError

I have no idea to solve this problem
Looking forward to your reply again

https://doc.qt.io/qt-5/qabstractsocket.html#SocketError-enum

bchaperon, 2020-08-14:
So after you start the example code, PRO and BCI stop working?
It seems that your issue is similar to this other one https://github.com/Emotiv/cortex-v2-example/issues/112
@nguoithichkhampha is already investigating.

gubeifang, 2020-08-14:
No, when I start the example code, PRO and BCI   working normal，

It's like whether I run the code has nothing to do with  PRO and BCI，
The code usage instructions say that EmotivApp will be prompted for authorization the first time the code is run correctly, but when I run the code, EmotivApp does not respond.
Is there a problem with my computer software environment configuration?
Looking forward to your reply again

gubeifang, 2020-08-18:
hi, Is there have any ways to solve this problem?

nguoithichkhampha, 2020-08-18:
@gubeifang , Can you setup a teamview session to figure out the problem ?

gubeifang, 2020-08-18:
Yes，Please tell me when we setup a teamview session,thank you.

gubeifang, 2020-08-19:
hi , When do we start a teamviewer session？Please contact me at your convenience, I will follow the email message between 8 am and 10 pm at Beijing time

nguoithichkhampha, 2020-08-19:
@gubeifang , sorry for late response. Please email for me your team id/ pass 
cuong@emotiv.com
when you ready at 10 AM GMT+7

gubeifang, 2020-08-19:
OK，my team id/ pass is 1 573 242 908/4kt69f
Looking forward to your help

nguoithichkhampha, 2020-08-19:
@gubeifang , I mean , we will setup session tomorrow and you send these informations via my email before start.
Don't public it .

gubeifang, 2020-08-19:
haha, ok, i get it . thank you

nguoithichkhampha, 2020-08-20:
the cause is missing openssl library when run with Qt example on windows.

closed 2020-08-20 by nguoithichkhampha

gubeifang, 2022-03-04:
Hello,
I'm a Emotiv Epoc+ user, I download "Emotiv-Installer-v3.2.0-x64.exe" from my Emotiv account
There was an error when I use it, like the picture show,I can't get the data
What should I do to solve this problem？
Look forward to your reply，thank you

nguoithichkhampha, 2022-03-04:
@gubeifang this is reason for your error

> the cause is missing openssl library when run with Qt example on windows.

gubeifang, 2022-03-05:
hi，I've tried what you said, but the problem still hasn't been solved. Can you remotely control my computer to help solve it

gubeifang, 2022-03-07:
hi， I've tried what you said, install the OpenSSL library， but the problem still hasn't been solved. Can you remotely control my computer to help solve it

nguoithichkhampha, 2022-03-07:
@gubeifang , sorry, this is out of scope in our supporting.

gubeifang, 2022-03-07:
OK，Can you provide me with some other solutions to solve this problem？

nguoithichkhampha, 2022-03-07:
@gubeifang , you need to check supported version in your Qt lib and add it into project. Ex: Qt lib supports open ssl 1.1, you can not use open ssl 1.0. One more thing, openssl lib has some dependencies. You need all of them to make openssl lib work.

gubeifang, 2022-03-16:
hello , my openssl  version 1.1.1m, and it normal operation，However, the following situation still occurs，“result”:[]