Emotiv/cortex-example #117: Connection to headset lost after 1-2 minutes

guneteroglu, 2020-09-05:
Hello, Our program successfully connects to the headset and gets data for a couple of minutes in Java Android interface SDK 2.0.
But afterwards, it does not get any data from     public void onBandPowerDataRecieved(String bandPowerData) .
Is there any timout parameter? or time limit for a subscription?
This is a repeatable problem. Every subscription at most takes 1-2 mins.
Then, I have to restart the program, it connects and continues.
This is a production issue. So, I appreciate if you solve or suggest as soon as possible.
Best regards,

nguoithichkhampha, 2020-09-07:
@guneteroglu , from Emotiv app, pls send feedback. we will collect log file to figure out the problem.

guneteroglu, 2020-09-07:
Hello, I have just send feedback from emotiv app. Thank you. Best regards

şunu yazdı:

nguoithichkhampha, 2020-09-07:
@guneteroglu , only work on Emotiv app, do you see this problem ?
connect headset on Emotiv app, wait a few minutes and lost connection ?

guneteroglu, 2020-09-07:
when only working emotiv app, connection is not lost.

şunu yazdı:

nguoithichkhampha, 2020-09-07:
ok, so if you work with your app and Emotiv app. do you see Emotiv app lost headset connection ?

guneteroglu, 2020-09-07:
Yes, it gives an error message on top saying that headset connection is
lost.
I have newly run a session and send the feedback from EMOTIV APP.
Best

Günet Eroğlu

CEO

HMS Health Mobile Software A,Ş

İçerenköy Mah. Kayışdağı Cad. No:32
Acıbadem Üniversitesi Kuluçka Merkezi 34752
Ataşehir İSTANBUL
+90 532 326 55 82
gunet.eroglu@healthmobilesoftware.com

 https://www.autotrainbrain.com/

nguoithichkhampha, 2020-09-07:
> Pzt Eyl 7 12:55:45 2020: EmotivAppMobile debug  "Message Error Received:{\"jsonrpc\":\"2.0\",\"warning\":{\"code\":103,\"message\":{\"behavior\":\"Headset has been disconnected due to connection timeout.\",\"headsetId\":\"EPOCX-E202008F\"}}}"
Pzt Eyl 7 12:55:45 2020: Step: showing error: -10007 Headset has been disconnected due to connection timeout.
this is log I get from Emotiv app.
it mean, you don't connect headset successful.

guneteroglu, 2020-09-07:
We use the example Cortex API code to connect with EPOC-X.
It connects the headset successfully and gets some data, but after some
time it disconnects.
I can share the app code if you want.
Best regards,

Günet Eroğlu

CEO

HMS Health Mobile Software A,Ş

İçerenköy Mah. Kayışdağı Cad. No:32
Acıbadem Üniversitesi Kuluçka Merkezi 34752
Ataşehir İSTANBUL
+90 532 326 55 82
gunet.eroglu@healthmobilesoftware.com

 https://www.autotrainbrain.com/

guneteroglu, 2020-09-07:
In my opinion, either there is a time limit parameter for connection
duration, or an array structure is filled and overflow after getting some
data in the web socket interface.

Günet Eroğlu

CEO

HMS Health Mobile Software A,Ş

İçerenköy Mah. Kayışdağı Cad. No:32
Acıbadem Üniversitesi Kuluçka Merkezi 34752
Ataşehir İSTANBUL
+90 532 326 55 82
gunet.eroglu@healthmobilesoftware.com

 https://www.autotrainbrain.com/

cyberspace69, 2020-09-07:
@guneteroglu - 

Happy to help with your issue. If you can share the app code, I can take a look.

Also, if you could answer these:

1) Does the connection consistently cut out around the same time (~1 - 2 minutes)?
2) How are you connecting your headset to Cortex (Bluetooth or USB)?

Thanks!

guneteroglu, 2020-09-07:
Hello,
Please find enclosed the simplified code. Please compile and run the
band-power app.
Use gunprod101/123456 user-id and password to login the app.
Connect to the headset, and push the start button.
Follow the LogCat at Android IDE for 2-3 minutes to see the power band data
coming from the headset.
After 2-3 minutes, it stops working.
1) Yes, consistently the connection cuts out after 2-3 minutes.
2) Bluetooth
Please do not distribute this code, as this is part of our prod version.
CONFIDENTIALITY NOTICE.
Best,
 otizmV2 - Copy.rar
<https://drive.google.com/file/d/1qNYDEdgjqS6oVExLjNvtOVE1fYKX9hdY/view?usp=drive_web>

Günet Eroğlu

CEO

HMS Health Mobile Software A,Ş

İçerenköy Mah. Kayışdağı Cad. No:32
Acıbadem Üniversitesi Kuluçka Merkezi 34752
Ataşehir İSTANBUL
+90 532 326 55 82
gunet.eroglu@healthmobilesoftware.com

 https://www.autotrainbrain.com/

cyberspace69, 2020-09-07:
@guneteroglu -

Thanks for the follow up on the questions

Tried accessing the link, but you'll need to change the security settings so I'm able to access.

Would you be able to unlock the link momentarily so I can take a look? Or perhaps host elsewhere?

guneteroglu, 2020-09-07:
https://drive.google.com/file/d/1qNYDEdgjqS6oVExLjNvtOVE1fYKX9hdY/view?usp=drivesdk

pl3ase try

7 Eyl 2020 Pzt 22:33 tarihinde cyberspace69 <notifications@github.com> şunu
yazdı:

cyberspace69, 2020-09-07:
Hey @guneteroglu -

I have not been able to see your code, however, I do believe I understand the issue.

As you're consistently facing this issue after a few minutes while using Android over Bluetooth, and the error log posted by @nguoithichkhampha mentions the headset being disconnected, most likely the issue lies within Android's background limitations, implemented in Oreo (Android 8).

Several people have experienced issues, specifically with Bluetooth on Android, if their app goes went into the background. 

I would suggest taking a look at what others have done, such as [this stackoverflow example](https://stackoverflow.com/questions/49729452/how-to-keep-bluetooth-connection-background).

Hope it helps!

nguoithichkhampha, 2020-09-08:
@cyberspace69 , the error I mentioned above is he did not connect headset successfully. not the case, headset connected and then disconnect. Our cortex is designed as background service. it should run independent with client app. We also tested another app connect to Cortex and record data. It can record in few hours.
@guneteroglu , I will try with your example code.

guneteroglu, 2020-09-08:
Thanks very much for the information. I have given access rights to you to
review the code.
I kindly wait for your response.
Best regards,

Günet Eroğlu

CEO

HMS Health Mobile Software A,Ş

İçerenköy Mah. Kayışdağı Cad. No:32
Acıbadem Üniversitesi Kuluçka Merkezi 34752
Ataşehir İSTANBUL
+90 532 326 55 82
gunet.eroglu@healthmobilesoftware.com

 https://www.autotrainbrain.com/

nguoithichkhampha, 2020-09-08:
@guneteroglu , I tried to run your example and it works normally
[band_power_log.txt](https://github.com/Emotiv/cortex-v2-example/files/5187751/band_power_log.txt)
So, I guess have the problem with your headset . Did you try with another headset ? is headset full battery ?
if headset is disconnected while working, do you see the message like this from cortex ?
`
2020-09-08 17:35:28.640 2008-3303/com.hms2017.atbotizm I/com.emotiv.cortexv2example.controller.CortexClientController: new warning: {"jsonrpc":"2.0","warning":{"code":1,"message":{"behavior":"Cortex has closed the session 5e5924c4-f871-42da-a4e4-cbe427badf63.","sessionId":"5e5924c4-f871-42da-a4e4-cbe427badf63"}}}
2020-09-08 17:35:28.641 2008-3303/com.hms2017.atbotizm I/com.emotiv.cortexv2example.controller.CortexClientController: new warning: {"jsonrpc":"2.0","warning":{"code":0,"message":{"behavior":"Cortex has stopped all the subscriptions of session 5e5924c4-f871-42da-a4e4-cbe427badf63.","sessionId":"5e5924c4-f871-42da-a4e4-cbe427badf63"}}}

`

guneteroglu, 2020-09-08:
How many minutes you have waited.
Would it be a problem of Android or epocx? i will test with other headset
and inform you.best

şunu yazdı:

nguoithichkhampha, 2020-09-08:
you can see my log file about 5 minutes.

guneteroglu, 2020-09-08:
Hello, I have tried the program with another headset (EPOC+) and it worked
very well.
So, this is a problem of headset. Now, I charge the headset, I will try the
headset fully charged to see whether
it will work or not.
I will inform you.
Best regards,

Günet Eroğlu

CEO

HMS Health Mobile Software A,Ş

İçerenköy Mah. Kayışdağı Cad. No:32
Acıbadem Üniversitesi Kuluçka Merkezi 34752
Ataşehir İSTANBUL
+90 532 326 55 82
gunet.eroglu@healthmobilesoftware.com

 https://www.autotrainbrain.com/

guneteroglu, 2020-09-08:
I have recharged EPOC-X and retried. The result is the same, after 2
minutes, the headset connection stops.
I believe it is a problem related with EPOC-X.
How do you suggest us to move forward with this issue?
Best regards,

Günet Eroğlu

CEO

HMS Health Mobile Software A,Ş

İçerenköy Mah. Kayışdağı Cad. No:32
Acıbadem Üniversitesi Kuluçka Merkezi 34752
Ataşehir İSTANBUL
+90 532 326 55 82
gunet.eroglu@healthmobilesoftware.com

 https://www.autotrainbrain.com/

nguoithichkhampha, 2020-09-08:
do you have another epoc-x to try ? I will try with epoc-x on my side, tmr

guneteroglu, 2020-09-08:
yes. I have a band new epocx. I will charge and try. 2 of our customers who
use epoc x complained about the issue.
One customer who has epocx use without any problem.

şunu yazdı:

guneteroglu, 2020-09-08:
I have tried with the other EPOC-X. It has the same problem, after 2-3
minutes, it does not receive Band Power data.
So it must be a general problem for EOPC-X.

Günet Eroğlu

CEO

HMS Health Mobile Software A,Ş

İçerenköy Mah. Kayışdağı Cad. No:32
Acıbadem Üniversitesi Kuluçka Merkezi 34752
Ataşehir İSTANBUL
+90 532 326 55 82
gunet.eroglu@healthmobilesoftware.com

 https://www.autotrainbrain.com/

hoangphamemotiv, 2020-09-09:
hi @guneteroglu 
can you change our headset configuration to 128Hz EEG (16bit) and 0Hz (turn off) Motion? then test it again?

guneteroglu, 2020-09-09:
Where and how should I change these parameters? By using EMOTIV APP on
desktop or on mobile?

Günet Eroğlu

CEO

HMS Health Mobile Software A,Ş

İçerenköy Mah. Kayışdağı Cad. No:32
Acıbadem Üniversitesi Kuluçka Merkezi 34752
Ataşehir İSTANBUL
+90 532 326 55 82
gunet.eroglu@healthmobilesoftware.com

 https://www.autotrainbrain.com/

nguoithichkhampha, 2020-09-09:
@guneteroglu , I got same issue when test with EPOCX. I think this problem related to bluetooth throughput on Android device. For now, you should recommend user don't use EPOCX with Android . Sorry about this inconvenience.

hoangphamemotiv, 2020-09-09:
@guneteroglu 
to change headset config, you need a USB cable and it in via Emotivpro on desktop (Mac or windows)
quality of BLE data depend on BLE throughput, so i suggest you turn off Motion data on Mobile, it helps to reduce data lost.
and yes, seem turn off Motion with Epoc X on android is not enough. we are working on this and send you a feedback ASAP

guneteroglu, 2020-09-09:
Hello, I have tested by turning motion off. Unfortunately, it stopped after
a while. We have upgraded our Android version to EPOC-X as a first step.
If Android version does not work, we currently do not have any solution for
2 of our customers who newly bought EPOC-X. They don''t have any iOS
telephone.
We have also ordered 2 more EPOC-X at our office, we need to find a
solution to fix that problem, or replace the EPOC-X with another working
hardware.
Please suggest how we should proceed.
Best regards,

Günet Eroğlu

CEO

HMS Health Mobile Software A,Ş

İçerenköy Mah. Kayışdağı Cad. No:32
Acıbadem Üniversitesi Kuluçka Merkezi 34752
Ataşehir İSTANBUL
+90 532 326 55 82
gunet.eroglu@healthmobilesoftware.com

 https://www.autotrainbrain.com/

bchaperon, 2020-09-10:
Hello @guneteroglu 
We are currently looking for a solution and we are treating this issue with the highest priority.
We will keep you informed of our progress.
Thank you for your patience. We apologize for the inconvenience.

nguoithichkhampha, 2020-09-11:
@guneteroglu , Actually, we have a solution for this issue. the fix version is testing by our QA team. Probably, you can get it in next week. I will inform you when new version ready.

guneteroglu, 2020-09-11:
perfect

şunu yazdı:

guneteroglu, 2020-09-13:
Hi, one of our customers who uses new versions of emotiv app and auto train
brain with epoc+ also reported the same problem. This means it may not be
only epoc-× hardware related.
best

11 Eyl 2020 Cum 12:16 tarihinde Gunet Eroglu <
gunet.eroglu@healthmobilesoftware.com> şunu yazdı:

nguoithichkhampha, 2020-09-14:
@guneteroglu , Could you help us to contact this customer ? I'm not sure this problem is same with EPOCX.

guneteroglu, 2020-09-14:
I have told the customer to send feedback with their email address.

guneteroglu, 2020-09-14:
They told me they have sent feedback. Akgun

nguoithichkhampha, 2020-09-14:
@guneteroglu , I got the feedback. I will discuss with him/her directly. Seems there are not same root cause with EPOCX.

guneteroglu, 2020-09-14:
ok. please let me know the result and actions.

şunu yazdı:

nguoithichkhampha, 2020-09-15:
@guneteroglu , after the discussion, we are pretty sure that the problem with EPOC+ from hardware. I suggested him contact our support team to report about hardware issue.

guneteroglu, 2020-09-15:
Thank you for the information. The customer also contacted me and reported
the same result.
In this case, we only have a problem with EPOC-X. When do you plan to send
the fix/patch?
Best regards,

Günet Eroğlu

CEO

HMS Health Mobile Software A,Ş

İçerenköy Mah. Kayışdağı Cad. No:32
Acıbadem Üniversitesi Kuluçka Merkezi 34752
Ataşehir İSTANBUL
+90 532 326 55 82
gunet.eroglu@healthmobilesoftware.com

 https://www.autotrainbrain.com/

nguoithichkhampha, 2020-09-15:
Our QA team in last testing step. Probably you will get new version 2.6.0 tomorrow.

guneteroglu, 2020-09-15:
I have got an email saying that EMOTIV app 2.6.0 available.  Is this the
version that I will ask the customer to download?
Is there any change in Cortex API and app?

Günet Eroğlu

CEO

HMS Health Mobile Software A,Ş

İçerenköy Mah. Kayışdağı Cad. No:32
Acıbadem Üniversitesi Kuluçka Merkezi 34752
Ataşehir İSTANBUL
+90 532 326 55 82
gunet.eroglu@healthmobilesoftware.com

 https://www.autotrainbrain.com/

nguoithichkhampha, 2020-09-15:
No, this email announce about release version on desktop. 
we have some changes on API but should not affect your app
https://emotiv.gitbook.io/cortex-api/release-notes

nguoithichkhampha, 2020-09-16:
@guneteroglu , we have release Cortex Android 2.6.0 for partner. you and your customer should be able to download it.

guneteroglu, 2020-09-16:
Hello, I have downloaded EMOTIV APP 2.6.0 version and tested that with our
app. It works smoothly. Thank you very much.

Günet Eroğlu

CEO

HMS Health Mobile Software A,Ş

İçerenköy Mah. Kayışdağı Cad. No:32
Acıbadem Üniversitesi Kuluçka Merkezi 34752
Ataşehir İSTANBUL
+90 532 326 55 82
gunet.eroglu@healthmobilesoftware.com

 https://www.autotrainbrain.com/

nguoithichkhampha, 2020-09-16:
Great. Let's me know if you have another issues.
thanks for your feedback.

closed 2020-09-16 by nguoithichkhampha