Emotiv/cortex-example #118: "Profile name already exists" when saving profile after training

cyberspace69, 2020-09-07:
## Issue
When attempting to save a successful training session to the loaded profile, I receive the following:

> {"error":{"code":-32118,"message":"The profile name already exists."},"id":6,"jsonrpc":"2.0"}

## Steps
* Loaded my profile to my headset (EPOC+)
* Created a session
* Subscribed to a data session
* Ran a training (mental / neutral)
* Accepted the training

Next, I run the following code:

```    
async def save_profile(self, cortex_token, headset_id, profile):
        if not headset_id:
            headset_id = self.headsets[0]
        params = {'cortexToken': cortex_token,
                  'headset': headset_id,
                  'profile': profile,
                  'status': 'save'}
        resp = await self.send_command('setupProfile', **params)
        logger.debug(f"{__name__} resp:\n{resp}")
        return resp
```

## Notes
To test the issue, during the call of **save_profile**, I substituted a different profile name, which successfully created a new profile and saved my training session.

Running the code again, however, gave me the "profile name exists" error again.

Is this a bug - or is there a different status to call than **save** ?

bchaperon, 2020-09-11:
Hi @cyberspace69 

This error happens when you load a profile "a" and then try to save it with name "b", but another profile with name "b" already exists.
Please make sure that you use the same profile name when you load and when you save your profile.

cyberspace69, 2020-09-12:
@bchaperon - After the latest update, I'm able to successfully save profiles without this issue.

Thank you!

closed 2020-09-12 by cyberspace69