Emotiv/cortex-example #119: emotiv issues

Yosief208, 2020-09-08:
i have a headset with pro license but it is not subscribing to eeg data stream...here is the code below
from websocket import create_connection
import json

url = 'wss://localhost:6868'
client_id = 'bOnVA7tcEztcbr0OX0SIgAbqKAZyoI9c7EBw8dJz'
client_secret = '5p8m9WMoTME9PTEyoHXavaw9RthJ2swuExA2V4oExmWf3rXvWXWjSECzksEq318jeHfZwU0UaO9dPXwaUhHrXX1wyUKnWXzqpt6igjmXVYy2mgTHS6EWPKfBCiCyEBfE'
pro_license = "affc943a-5870-41e7-bacd-c0ca7e3f3ee8"
tok = "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhcHBJZCI6ImNvbS5oYWJlbmlkLnByb2plY3QiLCJhcHBWZXJzaW9uIjoiMS4wIiwiZXhwIjoxNTk5NzYyMTMxLCJuYmYiOjE1OTk1MDI5MzEsInVzZXJJZCI6ImZkZDA5ZTA1LTUyYTgtNGMwMC05MjlkLTk2OGY0NGVlMjMyZSIsInVzZXJuYW1lIjoiaGFiZW5pZCIsInZlcnNpb24iOiIyLjAifQ==.+XAMaLcAvFyg7jX+FfZshfP8eJJFykCuy026+X02wtw="
ws = create_connection(url)

#get user login
ws.send(json.dumps(
    {
    "id": 1,
    "jsonrpc": "2.0",
    "method": "getUserLogin"
    }
    ))
print('user information:',ws.recv())

#request access
ws.send(json.dumps(
    {
    "id": 1,
    "jsonrpc": "2.0",
    "method": "requestAccess",
    "params": {
        "clientId": client_id,
        "clientSecret": client_secret
    }}
    ))
print('request access:',ws.recv())
#authorize
{
    "id": 1,
    "jsonrpc": "2.0",
    "method": "authorize",
    "params": {
        "clientId": client_id,
        "clientSecret": client_secret,
        "licence": pro_license,
        "debit": 100
    }
}

print(ws.recv())

#generate new token
ws.send(json.dumps(
    {
        "id": 1,
    "jsonrpc": "2.0",
    "method": "generateNewToken",
    "params": {
        "cortexToken": tok,
        "clientId": client_id,
        "clientSecret": client_secret
    }
}))

val = json.loads(ws.recv())
token = val['result']['cortexToken']

#query headsets
ws.send(json.dumps(
    {
    "id": 1,
    "jsonrpc": "2.0",
    "method": "queryHeadsets"
}))
headset = json.loads(ws.recv())
headset_id = headset['result'][0]['id']
headset_status = headset['result'][0]['status']

#connect to headset
ws.send(json.dumps(
    {
    "id": 1,
    "jsonrpc": "2.0",
    "method": "controlDevice",
    "params": {
        "command": "connect",
        "headset": headset_id
    }
}))
print(ws.recv())
#create session
if headset_status == 'connected':
    print('yesss')
    ws.send(json.dumps(
        {
    "id": 1,
    "jsonrpc": "2.0",
    "method": "createSession",
    "params": {
        "cortexToken": token,
        "headset": headset_id,
        "status": "open"
    }
}))
    session_values = json.loads(ws.recv())
    session_id = session_values['result']['id']

    #activate session
    ws.send(json.dumps(
        {
    "id": 1,
    "jsonrpc": "2.0",
    "method": "updateSession",
    "params": {
        "cortexToken": token,
        "session": session_id,
        "status": "active"
    }
}))
    print(ws.recv())

    #subscribe
    ws.send(json.dumps(
        {
    "id": 1,
    "jsonrpc": "2.0",
    "method": "subscribe",
    "params": {
        "cortexToken": token,
        "session": session_id,
        "streams": ["eeg"]
    }
}))
    print(ws.recv())
    

leave your comment please......

cyberspace69, 2020-09-09:
Hey @Yosief208 -

A lot of my high school years were copying code from other people, consider this giving back :) 

Here are three files, this works on an EPOC+ **when all files are in the same directory**.

Just modify the user.json with your client id / client secret, then run **python main.py**

## main.py
```
from cortex_new import Cortex
import asyncio
import os
import json

async def cortex_steps(cortex):
    print("** USER LOGIN **")
    await cortex.get_user_login()
    print("** GET CORTEX INFO **")
    await cortex.get_cortex_info()
    print("** HAS ACCESS RIGHT **")
    await cortex.has_access_right()
    print("** REQUEST ACCESS **")
    await cortex.request_access()
    print("** AUTHORIZE **")
    auth_token = await cortex.authorize()
    print('Your token is: ' + str(auth_token))
    license_json = await cortex.get_license_info()
    license = license_json['result']['license']['licenseId']
    print('Your license is: ' + str(license))
    headsets = await cortex.query_headsets()
    if headsets != None:
        if len(headsets) > 1:
            print('multiple headsets discovered.. using first one')
            headset_id = headsets[0]
        else:
            headset_id = headsets[0]
    else:
        print('no headsets found')
    status = 'activate'
    session_json = await cortex.create_session(auth_token, status, headset_id)
    print('successfully started session')
    session = session_json['result']['id']
    print('Your session id is: ' + str(session))
    streams = ['eeg']
    await cortex.subscribe(auth_token, session, streams)
    print('Now subscribed to EEG')

def cortex_setup():
    login_data_path = 'user.json'
    if os.path.isfile(login_data_path):
        with open(login_data_path) as login_data:
            login_data = json.load(login_data)
            print(login_data)
            client_id = login_data['clientId']
            client_secret = login_data['clientSecret']
    cortex = Cortex(login_data_path)
    asyncio.run(cortex_steps(cortex))

cortex_setup()
```

## new_cortex.py
```
############
# This program is free software: you can redistribute it and/or modify
# it under the terms of the GNU General Public License as published by
# the Free Software Foundation, either version 3 of the License, or
# (at your option) any later version.
# This program is distributed in the hope that it will be useful,
[... 269 lines cut ...]
            com: mental commands data
            fac: facial expression data
            sys: system events (training data, etc)
        '''

        params = {'cortexToken': cortex_token,
                  'session': session_id,
                  'streams': stream_list}
        resp = await self.send_command('subscribe', **params)
        logger.debug(f"{__name__} resp:\n{resp}")

    async def unsubscribe(self, stream_list):
        params = {'cortexToken': self.auth_token,
                  'session': self.session_id,
                  'streams': stream_list}
        resp = await self.send_command('unsubscribe', **params)
        logger.debug(f"{__name__} resp:\n{resp}")
```

## user.json
```
{"clientId": "ENTER YOUR CLIENT ID HERE", "clientSecret": "ENTER YOUR CLIENT SECRET HERE"}
```

Lastly, if you need some of the other functions (control device, etc), you can add them, but you shouldn't need. Take a look at the example Cortex file within the python docs and you should be able to find anything else you need.

Yosief208, 2020-09-09:
thanks for your corporation...still its not working...
this is the output produced
Sending request:
{"jsonrpc": "2.0", "method": "subscribe", "params": {"cortexToken": "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhcHBJZCI6ImNvbS5oYWJlbmlkLnByb2plY3QiLCJhcHBWZXJzaW9uIjoiMS4wIiwiZXhwIjoxNTk5ODQ0MjU0LCJuYmYiOjE1OTk1ODUwNTQsInVzZXJJZCI6IjZiNWEzNTZkLTRhOWQtNDk0My1iODlkLTYxNTJlMTFhNDc4ZCIsInVzZXJuYW1lIjoiaGFiZW5pZCIsInZlcnNpb24iOiIyLjAifQ==.olBQ76XZezJnUOnq/bfbrJtBJYwuXAu2R8sBhOtAbqY=", "session": "ebcbf765-0214-4d0b-96c7-7d79a8a3885a", "streams": ["eeg"]}, "id": 9}
sent; awaiting response
cortex_new resp:
{'id': 9, 'jsonrpc': '2.0', 'result': {'failure': [{'code': -32016, 'message': 'The stream is unavailable or unsupported.', 'streamName': 'eeg'}], 'success': []}}
Now subscribed to EEG

cyberspace69, 2020-09-09:
Hey @Yosief208 -

Happy to help!

If you could comment back with the full output so I can take a look, I can get a better idea as to what's happening.

Thanks!

cyberspace69, 2020-09-09:
Also - if you haven't done so, make sure you've followed this step of the docs:

> If you have a PRO license then you must fill an application before you can use your license with the API. There is one application form for each license type: [Business](https://www.emotiv.com/commercial-sdk-application-form/), [Academic](https://www.emotiv.com/non-commercial-sdk-application-form/), [Student](https://www.emotiv.com/student-sdk-application-form/).

Each of the forms for their respective license are linked above.

bchaperon, 2020-09-10:
Hi @Yosief208 

Please call the method "getLicenseInfo" and check if Cortex is using your PRO license, or the "basic" license.
https://emotiv.gitbook.io/cortex-api/authentication/getlicenseinfo

You can set your PRO license id when you call the method "authorize".
https://emotiv.gitbook.io/cortex-api/authentication/authorize

And of course, you need to follow the steps mentioned by @cyberspace69 first.

Yosief208, 2020-09-11:
Hey guys finally it works. The problem was I didn't fill an application that's why I was unable to access it...I am really happy for your help...It's a lot to me....Thanks a lot.....

closed 2020-09-11 by bchaperon