Emotiv/cortex-example #123: Someone please help me connect EPOC+ and UNITY

ShadowsSpwn, 2020-10-12:
Greetings, I have tried for several days to connect EPOC+ to UNITY but I have not been successful, I have reviewed forums but there is no updated information or a clearer tutorial on how they do it (I would be so happy with an example video xD) I am not a programmer :(
Could you help me and guide me a little better on how to get UNITY to recognize EPOC+?
I have followed the instructions of
https://github.com/Emotiv/unity-plugin
and the example of
https://github.com/Emotiv/cortex-v2-example/tree/master/unity
But I don't understand how I enter the information in UNITY, I enter it as an asset, I import it as a package, how do I get it to be recognized by UNITY?

Help me understand please, PLEASE!

Thank you very much!

tungntEmotiv, 2020-10-14:
Hi @ShadowsSpwn ,
Did you follow the guides in docs at https://github.com/Emotiv/cortex-v2-example/tree/master/unity 
1. Change the client Id, client Secret of your application in AppConfig.cs.
2. Make sure you have logged in via EMOTIV App, and the headset has been turned on.
3. Run the example from editor. The example will connect to Cortex for authorization. You might need to grant access right for the example via EMOTIV App at the first time. After authorizing successfully, the example will list available headsets
...
Could you show me the logs from editor ?

ShadowsSpwn, 2020-10-16:
Thanks I will try with the steps that you indicated.

closed 2020-11-05 by nguoithichkhampha